esp-rs/esp-idf-svc #367: Does esp-idf-svc provide something similar to esp_hal_common::clock?

weiying-chen, 2024-02-17:
With this code (to read the weather with esp32-c3 and bm280):

```
use esp_idf_svc::hal::{
    delay,
    i2c::{I2cConfig, I2cDriver},
    peripherals::Peripherals,
    prelude::*,
};

use bme280::i2c::BME280;

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let sda = peripherals.pins.gpio2;
    let scl = peripherals.pins.gpio3;
    let config = I2cConfig::new().baudrate(400_u32.kHz().into());
    let i2c = I2cDriver::new(peripherals.i2c0, sda, scl, &config).unwrap();
    let mut bme280 = BME280::new_primary(i2c);
    let mut delay = delay::Ets;

    bme280.init(&mut delay);

    loop {
        let measurements = bme280.measure(&mut delay).unwrap();

        log::info!("Relative Humidity = {}%", measurements.humidity);
        log::info!("Temperature = {} deg C", measurements.temperature);
        log::info!("Pressure = {} pascals", measurements.pressure);
        delay::FreeRtos::delay_ms(10000u32);
    }
}
```

And I get this error:

```
thread 'main' panicked at src/main.rs:40:55:
called `Result::unwrap()` on an `Err` value: InvalidData
```

It panics here:

```
let measurements = bme280.measure(&mut delay).unwrap();
```

It's strange because this almost identical code works:

```
#![no_std]
#![no_main]

use bme280::i2c::BME280;
use esp32c3_hal::{clock::ClockControl, i2c::I2C, peripherals::Peripherals, prelude::*, Delay, IO};
use esp_backtrace as _;
use esp_println::println;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
    let mut delay = Delay::new(&clocks);
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sda = io.pins.gpio2;
    let scl = io.pins.gpio3;
    let i2c = I2C::new(peripherals.I2C0, sda, scl, 400u32.kHz(), &clocks);
    let mut bme280 = BME280::new_primary(i2c);

    bme280.init(&mut delay).unwrap();

    loop {
        let measurements = bme280.measure(&mut delay).unwrap();
        println!("Relative Humidity = {}%", measurements.humidity);
        println!("Temperature = {} deg C", measurements.temperature);
        println!("Pressure = {} pascals", measurements.pressure);
        delay.delay_ms(1000u32);
    }
}
```

I suspect I have to use something similar to this in the first code:

```
let clocks = ClockControl::boot_defaults(system.clock_control).freeze();
let mut delay = Delay::new(&clocks);
```

Does esp-idf-svc provide something similar to [ClockControl](https://github.com/esp-rs/esp-hal/blob/172d2b8777fa6d19bb5c1cd65651dda9742cede1/esp-hal/src/clock/mod.rs#L75)?

Vollbrecht, 2024-02-17:
what version of esp-idf-svc/hal are you using? It may be the ETS delay impl that is not doing its job right. We updated it on current master branch of esp-idf-hal. If you want to have a quick test you can add

```toml
[patch.crates-io]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
``
inside your Cargo.toml then run `cargo update` and `cargo b`

weiying-chen, 2024-02-17:
I added that at the end of my Cargo.toml:

```
[patch.crates-io]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
```

And I seem to be using the latest version of `esp-idf-svc`:

```
esp-idf-svc = { version = "0.48", default-features = false }
```

But I'm still getting the same error. Maybe the issue isn't ETS delay ...

weiying-chen, 2024-02-18:
@Vollbrecht Did you guys change something in `https://github.com/esp-rs/esp-idf-hal`? Because now this line:

```
[patch.crates-io]
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
```

Removes the InvalidData error, and the sensor reads the weather successfully. I don't think I did anything special apart from downgrading rust nightly to 1.77.0 (2a3e63551 2023-12-30).

Vollbrecht, 2024-02-18:
no we didn't - 99% chance you forgot `cargo update` because git dependency's need manual updating and tracking otherwise they are not pulled in

closed 2024-02-18 by ivmarkov