esp-rs/esp-idf-svc #374: The following tasks/users did not reset the watchdog in time: IDLE (CPU 0)

weiying-chen, 2024-02-25:
I'm using this code to read from my HX711/load sensor (using [this](https://github.com/DaneSlattery/hx711) crate):

```
use esp_idf_svc::hal::{delay, gpio::PinDriver, peripherals::Peripherals};
use loadcell::{hx711::HX711, LoadCell};

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let dt = PinDriver::input(peripherals.pins.gpio1).unwrap();
    let sck = PinDriver::output(peripherals.pins.gpio10).unwrap();
    let mut load_sensor = HX711::new(sck, dt, delay::Ets);

    load_sensor.tare(16);
    load_sensor.set_scale(1.0);

    loop {
        if load_sensor.is_ready() {
            let reading = load_sensor.read_scaled();
            log::info!("Last Reading = {:?}", reading);
        }

        delay::FreeRtos::delay_ms(1000u32);
    }
}
```

ESPMonitor shows me this error (the code hangs on ` load_sensor.tare(16);`):

```
E (75386) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (75386) task_wdt:  - IDLE (CPU 0)
E (75386) task_wdt: Tasks currently running:
E (75386) task_wdt: CPU 0: main
E (75386) task_wdt: Print CPU 0 (current core) registers
```

What could be the issue?

ivmarkov, 2024-02-25:
The `hx711` driver is implemented using **bit-banging**. What that means is that this driver is doing some write on a pin, *then internally using the `Ets` struct to delay, then reading some pin, and possibly all of this in a loop.

The problem is that when the driver internally calls `Ets::delay`, this function is not really suspending the current task/thread (i.e. your main one). From the point of view of FreeRTOS, this task is **busy looping**, because `Ets:delay` is not implemented in an intelligent way.

Now, there is nothing wrong with busy looping per se, except that you are *usually* not supposed to do that when you have an RTOS underneath, as that means that **other** tasks, having lower priority might never or just rarely get executed, because your task is constantly running, and it will only be preempted by higher or same priority tasks threads.

Since the FreeRTOS Idle tasks have the lowest priority possible, they don't get a chance to execute.
These "idle" tasks only exist to actually tell you that you are busy looping, which very often is a bug.

In your case, it is not a bug, so you might want to disable the idle tasks.

The real solution however would be to implement a more intelligent "delay" than `Ets`. Perhaps a one implemented with the `EspTimerService` utility plus the FreeRTOS task notification service in `esp_idf_svc::task::notification`. The idea of this more intelligent delay would be that your task would block, until awoken by a timer.

`esp-idf-sys`

ivmarkov, 2024-02-25:
[This example](https://github.com/esp-rs/esp-idf-hal/blob/master/examples/timer_notify.rs) might be a starting point to implement a better `Ets::delay` alternative. Basically you need something like a `TimerDelay` struct, where you'll instantiate and keep a Box-ed timer instance. Also when creating the struct you need to call `timer::subscribe`. Then each time your method `TimerDelay::delay` is called, you'll reload the timer with the necessary delay, and will enable the timer alarm. After that, you'll block - just like in the example - waiting to be notified via the task notification that you need to notify from within `subscribe`. Once you are notified, you disable the timer alarm and exit the `delay` method.

weiying-chen, 2024-02-26:
Thanks a lot for the advice. Now I have another [problem](https://github.com/DaneSlattery/hx711/issues/9#issuecomment-1962950154). The code runs, and I get an (empty) output if my HX711's DT is not connected to my ESP32-C3's giop (0 or 1). If I connect it, the code gets stuck in the `load_sensor.tare()` method.

Do you think this is related to the issue you're describing, it's a hardware issue (I heard some sparkling noise once, so I wonder if I damaged my HX711).

ivmarkov, 2024-02-26:
> Thanks a lot for the advice. I used FreeRTOS instead of Ets, and the error disappeared.

FreeRTOS gives you a delay resolution of 10 ms or more. Might be a bit of a too low resolution for HX711?

ivmarkov, 2024-02-26:
Look [here](https://github.com/DaneSlattery/hx711/blob/main/src/hx711.rs#L39). This driver wants a delay impl with a resolution of 1 microsecond, while `FreeRTOS` is giving a resolution of 10 milliseconds.

At the same time, it is *also* using big delays, like [these ones](https://github.com/DaneSlattery/hx711/blob/main/src/hx711.rs#L41).

You could try [this delay impl instead](https://github.com/esp-rs/esp-idf-hal/blob/master/src/delay.rs#L350).

weiying-chen, 2024-02-26:
I tried this:

```
use esp_idf_svc::hal::{delay, gpio::PinDriver, peripherals::Peripherals};
use loadcell::{hx711::HX711, LoadCell};

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let dt = PinDriver::input(peripherals.pins.gpio0).unwrap();
    let sck = PinDriver::output(peripherals.pins.gpio10).unwrap();
    let delay_provider = delay::Delay::new_default(); // Added this
    let mut load_sensor = HX711::new(sck, dt, delay_provider); // Changed this

    load_sensor.tare(16);
    load_sensor.set_scale(1.0);

    loop {
        if load_sensor.is_ready() {
            let reading = load_sensor.read_scaled();
            log::info!("Last Reading = {:?}", reading);
        }

        delay::FreeRtos::delay_ms(1000u32);
    }
}
```

But but code still gets stuck in `load_cell.tare()`. With this output:

```
I (386) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (396) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

Maybe I misunderstood your suggestion? Or my HX711 is broken?

weiying-chen, 2024-02-28:
@DaneSlattery @beeb It works now. I had to solder the load cell wires directly to the HX711 instead of soldering them to the jump wires.

```
I (8936) rust_esp32_bme280: Last Reading = Ok(-10.0)
I (9036) rust_esp32_bme280: Last Reading = Ok(-3.0)
I (9136) rust_esp32_bme280: Last Reading = Ok(28.0)
I (9236) rust_esp32_bme280: Last Reading = Ok(44.0)
I (9336) rust_esp32_bme280: Last Reading = Ok(83.0)
I (9436) rust_esp32_bme280: Last Reading = Ok(64.0)
I (9536) rust_esp32_bme280: Last Reading = Ok(67.0)
I (9636) rust_esp32_bme280: Last Reading = Ok(449.0) [I put something on the load cell here]
I (9736) rust_esp32_bme280: Last Reading = Ok(4367.0)
I (9836) rust_esp32_bme280: Last Reading = Ok(7652.0)
```

The readings are a little crazy, but I guess I have to calibrate the scale.

By the way, this is the final code:

```
use esp_idf_svc::hal::{
    delay::{Delay, FreeRtos},
    gpio::PinDriver,
    peripherals::Peripherals,
};
use loadcell::{hx711::HX711, LoadCell};

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let dt = PinDriver::input(peripherals.pins.gpio2).unwrap();
    let sck = PinDriver::output(peripherals.pins.gpio3).unwrap();
    let delay = Delay::new_default();
    let mut load_sensor = HX711::new(sck, dt, delay);

    load_sensor.tare(16);
    load_sensor.set_scale(1.0);

    loop {
        if load_sensor.is_ready() {
            let reading = load_sensor.read_scaled();
            log::info!("Last Reading = {:?}", reading);
        }

        FreeRtos::delay_ms(100u32);
    }
}
```

Vollbrecht, 2024-05-25:
closing as completed

closed 2024-05-25 by Vollbrecht