esp-rs/esp-idf-svc #378: Panic on firmware information

stabl-gjn, 2024-03-04:
Offending code: <https://github.com/esp-rs/esp-idf-svc/blob/730fa3642d7042735de924c3d6ce71eafb8dc615/src/ota.rs#L33-L35>

`esp-idf-svc` panics if the firmware version of `esp_app_desc_t.version` (and other related fields) is longer than 24 chars.

The upper limit (24 chars) is defined by [embedded_svc](https://github.com/esp-rs/embedded-svc/blob/master/src/ota.rs#L20).
`esp-idf-svc` takes the version provided by esp-idf and tries to convert it into `heapless::String::<24>` panicking if the version is any longer.

In our project the version used by `esp-idf-svc` is the output of git describe which is quite big (e.g. `v0.3.0-146-gd33fb261-dirty`) and leads to a uncontrollable panic.

I think, at least the unwrap should be replaced by an `except` to give the developer any idea of what went wrong or in the better case be handled (in what ever way, e.g. truncate, Result/Option)

esp-idf app version description: <https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#app-version>

Vollbrecht, 2024-03-04:
to set the correct firmware name i would suggest, you are using the [app_desc](https://github.com/esp-rs/esp-idf-sys/blob/master/src/app_desc.rs) makro we provide. That way you get a version controlled information and build time information, all set via your usual Cargo.toml

stabl-gjn, 2024-03-05:
@Vollbrecht thanks i will try that for now