esp-rs/esp-idf-svc #385: problem with EspMqttClient from est-idf-svc

samutigro, 2024-03-14:
I'm trying to run an MQTT client on my esp32 wroom32, this is my code, the wifi connection works and I also added dome priantln!() to see if the code was being executed, the problem comes when I call client.publish() function, it doesn't seem to be working and also doesn't print the next println!(). I'm connecting with MQTT Explorer on my Mac to subscribe to the topic I'm publishing from esp32 but nothing in being published. 
Any ideas?
`use anyhow::Result;
use embedded_svc::mqtt::client::QoS;
use esp_idf_sys as _; 
use esp_idf_svc::{wifi::EspWifi, nvs::EspDefaultNvsPartition, eventloop::EspSystemEventLoop};
use embedded_svc::wifi::{ClientConfiguration, Configuration};
use heapless::String;
use esp_idf_hal::peripherals::Peripherals;
use esp_idf_svc::{
    mqtt::client::{EspMqttClient, MqttClientConfiguration},
};
use std::{thread::sleep, time::Duration};

fn main() -> Result<()> {
  
//wifi connection
esp_idf_sys::link_patches(); // Needed for esp32-rs
esp_idf_svc::log::EspLogger::initialize_default();

    println!("Entered Main function!");

    let peripherals = Peripherals::take().unwrap();
    let sys_loop = EspSystemEventLoop::take().unwrap();
    let nvs = EspDefaultNvsPartition::take().unwrap();

    let mut wifi_driver = EspWifi::new(peripherals.modem, sys_loop, Some(nvs)).unwrap();

    
    //here wifi credentials 

    let ssid: String<32> = "OrbiNet".try_into().unwrap(); 
    let password: String<64> = "LaMiaPassword".try_into().unwrap(); 

    wifi_driver.set_configuration(&Configuration::Client(ClientConfiguration{
        ssid,
        password,
        ..Default::default()
    })).unwrap();

    wifi_driver.start().unwrap();
    wifi_driver.connect().unwrap();

    while !wifi_driver.is_connected().unwrap() {
        let config = wifi_driver.get_configuration().unwrap();
        println!("Waiting for station {:?}", config);
    }

    println!("Should be connected now");

    // Client configuration:
    println!("ok1");

    let broker_url = format!("mqtt://test.mosquitto.org"); 
    
    println!("ok2");
    
    let mqtt_config = MqttClientConfiguration::default();
    
    println!("ok3");
 
    let (mut client, _connection) = EspMqttClient::new(&broker_url, &mqtt_config).unwrap(); 
 
    println!("ok4");
 
    let payload: &str = "Hello World";
 
    println!("ok5");
 
    client.publish("Digitiamo", QoS::AtLeastOnce, true, payload.as_bytes())?;

    println!("ok6");
    loop {
        sleep(Duration::from_secs(5));
        println!("sent!!!");
    }
}
`

ivmarkov, 2024-03-14:
You have to fetch the events from the returned `_connection` or else sending won't work either. If you don't need the events (can't imagine that but oh well) use the other constructor, that does not return a connection, but expects an event callback closure, where you can supply an empty one.

closed 2024-03-14 by ivmarkov

samutigro, 2024-03-14:
how do I fetch a connection event?

samutigro, 2024-03-14:
> how do I fetch a connection event?

@ivmarkov

ivmarkov, 2024-03-14:
Would you please join the  Matrix room and ask there? Also, look at the `Connection` API. It is a single method...

samutigro, 2024-03-14:
> Would you please join the Matrix room and ask there? Also, look at the `Connection` API. It is a single method...

how do I access Matrix room?

ivmarkov, 2024-03-14:
https://matrix.to/#/#esp-rs:matrix.org

ivmarkov, 2024-03-14:
Look at [this(https://github.com/esp-rs/esp-idf-svc/blob/master/examples/mqtt_client.rs) too.