esp-rs/esp-idf-svc #386: Feature dependency error when compiling a project that was working

AloizioMacedo, 2024-03-15:
I am facing the following error when compiling a project:

```
error: failed to select a version for `embedded-svc`.
    ... required by package `esp-idf-svc v0.48.0`
    ... which satisfies dependency `esp-idf-svc = "^0.48"` of package `traffic-light v0.1.0 (/home/aloizio/Documents/Rust_Projects/traffic-light)`
versions that meet the requirements `^0.27` are: 0.27.1, 0.27.0

the package `esp-idf-svc` depends on `embedded-svc`, with features: `serde` but `embedded-svc` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

failed to select a version for `embedded-svc` which could resolve this conflict
```

I just tried to recompile a project that was working normally three days ago. Here is the [Cargo.toml](https://github.com/AloizioMacedo/traffic-controller/blob/fe5ce58e1d7df25cdb593dc8950d516afcfb6cb4/Cargo.toml) for reference.

jukrieger, 2024-04-02:
This is a bit weird, since embedded-svc's feature is called `use_serde`. Can you try upgrading to `esp-idf-svc` version 0.48.1?

Vollbrecht, 2024-06-21:
embedded-svc got some minor changes. It needs to be kept in sync with esp-idf-svc. So if you are updating esp-idf-svc and you are depending on embedded-svc directly you should also update it. If you are using esp-idf-svc git master you also need to use embedded-svc master both via [patch.crates-io} in your Cargo.toml.

Closing as if the above is true it should not be a problem. Feel free to reopen it if you still cant get it to work.

closed 2024-06-21 by Vollbrecht

muffix, 2024-11-11:
This is happening to me, too. In a freshly generated project from the `esp-rs/esp-idf-template` template (version 5.2).

```
❯ cargo build
    Updating crates.io index
error: failed to select a version for `embedded-svc`.
    ... required by package `esp-idf-svc v0.49.0`
    ... which satisfies dependency `esp-idf-svc = "^0.49.0"` of package `hello-world v0.1.0 (/Users/muffix/src/hello-world)`
versions that meet the requirements `^0.28` are: 0.28.0

the package `esp-idf-svc` depends on `embedded-svc`, with features: `serde` but `embedded-svc` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

failed to select a version for `embedded-svc` which could resolve this conflict
```

I've tried pinning `esp-idf-svc` to `0.49.1` and `0.49.0`, with the same results.

ivmarkov, 2024-11-11:
> This is happening to me, too. In a freshly generated project from the `esp-rs/esp-idf-template` template (version 5.2).
>
> ```
> ❯ cargo build
>     Updating crates.io index
> error: failed to select a [...]

Can you paste your `Cargo.toml` file here? I find this error a bit hard to believe, as the CI of `esp-idf-template` does pass.

muffix, 2024-11-11:
```
[package]
name = "hello-world"
version = "0.1.0"
authors = ["..."]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "hello-world"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = []

experimental = ["esp-idf-svc/experimental"]

[dependencies]
log = "0.4"
esp-idf-svc = { version = "0.49", features = ["critical-section", "embassy-time-driver", "embassy-sync"] }

[build-dependencies]
embuild = "0.32.0"
cc = "=1.1.30" # Necessary until a new version of `esp-idf-sys` is released
```

This is really just the current template with the project name filled in, the rest are defaults.

ivmarkov, 2024-11-11:
Do you have any environment variables? RUSTFLAGS? `.cargo/config.toml` in a parent dir?

muffix, 2024-11-11:
No Rust-related env vars, no parent project.

ivmarkov, 2024-11-11:
There must be some configuration specific for your PC. Otherwise [this](https://github.com/esp-rs/esp-idf-template/actions/runs/11774344885) would've failed as well.

ivmarkov, 2024-11-11:
Can you just try on another PC? Or in a VM? I know this could all be annoying, but I'm not sure what else to suggest. You could also try wiping out the `git` and `registry` folders inside your `~/.cargo`.