esp-rs/esp-idf-svc #389: No server verification option set in esp_tls_cfg_t structure

weiying-chen, 2024-03-17:
1. I generated a project with [esp-idf-template](https://github.com/esp-rs/esp-idf-template).
2. I copied the code from [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs).
3. And tried to make a simple HTTP GET request:

```
fn get_request(client: &mut HttpClient<EspHttpConnection>) -> anyhow::Result<()> {
    let url = "https://httpbin.org/get";
    let request = client.get(url)?;
    info!("-> GET {}", url);
    let response = request.submit()?;
    info!("<- {}", response.status());
    std::thread::sleep(core::time::Duration::from_secs(5));
    Ok(())
}
```

I got this error:

```
E (4918) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (4918) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (4928) esp-tls: create_ssl_handle failed
E (4928) esp-tls: Failed to open new connection
E (4938) transport_base: Failed to open a new connection
E (4948) HTTP_CLIENT: Connection failed, sock < 0
```

In this repository (esp-idf-svc), there is mention of "esp_tls" in the following files: src/tls.rs, ws.rs, src/mqtt/client.rs

What am I supposed to add to my project to stop getting that error?

ivmarkov, 2024-03-17:
[This](https://github.com/ivmarkov/rust-esp32-std-demo/blob/58b9ea6680d282dcb26c0a77b8108f4ae2494b48/src/main.rs#L673). The example is for mqtt, but https client wqorks the same.

closed 2024-03-17 by ivmarkov

weiying-chen, 2024-03-19:
I guess it should be something like this, but what should I put in `??`? 

```
    let config = ?? {
        crt_bundle_attach: Some(esp_idf_svc::sys::esp_crt_bundle_attach),
        ..Default::default()
    };

    let mut client = HttpClient::wrap(EspHttpConnection::new(&config)?);
```

weiying-chen, 2024-03-19:
Thanks a lot! This worked.

```
use esp_idf_svc::http::client::{Configuration as HttpConfiguration, EspHttpConnection};

let config = &HttpConfiguration {
    crt_bundle_attach: Some(esp_idf_svc::sys::esp_crt_bundle_attach),
    ..Default::default()
};

let mut client = HttpClient::wrap(EspHttpConnection::new(&config)?);

get_request(&mut client)?;
```

I think this should be in the official examples? I was stuck for days.

ivmarkov, 2024-03-19:
> I think this should be in the official examples? I was stuck for days.

Feel free to open a PR for a new / improved HTTP client example. I can only dedicate so much of my free time to this project in general, and writing examples for folks in particular. :)

weiying-chen, 2024-03-19:
You're right. You're already doing a great job. Okay, I'll open a PR soon.

Before I open the PR, one question: how come the [http_client.rs](https://github.com/esp-rs/esp-idf-svc/blob/master/examples/http_client.rs) example runs without `crt_bundle_attach: Some(esp_idf_svc::sys::esp_crt_bundle_attach)`? Or it hasn't been tested yet?

ivmarkov, 2024-03-19:
Because it is using HTTP, not HTTPS.

weiying-chen, 2024-03-19:
Oh, got it. Thanks for the clarification.