esp-rs/esp-idf-svc #395: websocket causes the stack overflow

Hydrostic, 2024-03-21:
I'm using the examples from `examples/http_ws_client.rs`, with `CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096`,
 but the firmware kept exiting due to stack overflow. I tried to increase the size of stack, but it does not work. 
here is the log:
```
I (4556) esp_netif_handlers: sta ip: 192.168.3.67, mask: 255.255.255.0, gw: 192.168.3.1
I (4556) doorlock::wifi: Wifi netif up
W (4566) websocket_client: `reconnect_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
W (4566) websocket_client: `network_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
I (4596) doorlock: Websocket before connect

***ERROR*** A stack overflow in task websocket_task has been detected.

Backtrace: 0x40082fe2:0x3ffd1610 0x400891d1:0x3ffd1630 0x4008bf62:0x3ffd1650 0x4008a91f:0x3ffd16d0 0x4008c0b0:0x3ffd16f0 0x4008c062:0x3f40035c |<-CORRUPTED
0x40082fe2 - panic_abort
    at /home/insolublehco3/.espressif/esp-idf/v5.1.2/components/esp_system/panic.c:452
0x400891d1 - esp_system_abort
    at /home/insolublehco3/.espressif/esp-idf/v5.1.2/components/esp_system/port/esp_system_chip.c:84
0x4008bf62 - vApplicationStackOverflowHook
    at /home/insolublehco3/.espressif/esp-idf/v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:581
0x4008a91f - vTaskSwitchContext
    at /home/insolublehco3/.espressif/esp-idf/v5.1.2/components/freertos/FreeRTOS-Kernel/tasks.c:3729
0x4008c0b0 - _frxt_dispatch
    at /home/insolublehco3/.espressif/esp-idf/v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450
```
And the code:
```rust
fn main() -> Result<()>{
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();
    let sysloop = EspSystemEventLoop::take()?;

    // The constant `CONFIG` is auto-generated by `toml_config`.
    let app_config = CONFIG;
    // todo: loop
    let nvs = EspDefaultNvsPartition::take()?;
        let mut wifi_t = BlockingWifi::wrap(
            EspWifi::new(peripherals.modem, sysloop.clone(), Some(nvs))?,
            sysloop,
        )?;
        wifi::connect_wifi(&mut wifi_t, 
            app_config.wifi_ssid,
            app_config.wifi_psk
        )?;

    let config = EspWebSocketClientConfig {
        ..Default::default()
    };
    let timeout = Duration::from_secs(10);
    let (tx, rx) = mpsc::channel::<ExampleEvent>();
    let mut client =
        EspWebSocketClient::new("ws://192.168.3.7:7771", &config, timeout, move |event| {
            handle_event(&tx, event)
        })?;
    assert_eq!(rx.recv(), Ok(ExampleEvent::Connected));
    assert!(client.is_connected());
    let message = "Hello, World!";
        info!("Websocket send, text: {}", message);
        client.send(FrameType::Text(false), message.as_bytes())?;
    drop(client);
    Ok(())
}

```

ivmarkov, 2024-03-21:
`CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE` does not affect the stack of the main task where you are running this stuff. You want `CONFIG_MAIN_TASK_STACK_SIZE`

closed 2024-03-21 by ivmarkov

Hydrostic, 2024-03-21:
Dont work either

ivmarkov, 2024-03-21:
I can't debug your program for you. It is not even the complete program, because the event handler is missing.

 You _have to_ read the ESP IDF documentation in terms of what parameters you need to configure to extend the various tasks stack sizes. It could be, that the websocket client is running its own task too, which might be running out of stack. Play a bit with the program by simplifying it until you figure out which stack blows up.