esp-rs/esp-idf-svc #399: Only a few post requests are sent before program panics

weiying-chen, 2024-03-25:
I'm sending post requests every 5 seconds inside a loop that will break when the iterations reach 10:

```
let mut iterations = 0;

loop {
    if scale.is_ready() {
        log::info!("Iteration number {}", iterations);
        
        // logic to set payload
     
        post_request(&mut client, payload_bytes)?;

        log::info!("Post request sent!");
    }

    FreeRtos::delay_ms(5000u32);

    iterations += 1;

    if iterations >= 10 {
        break;
    }
}
```

This is the `post_request` function:

```
fn post_request(client: &mut HttpClient<EspHttpConnection>, payload: &[u8]) -> anyhow::Result<()> {
    // set the key, url, and headers 

    let mut request = client.post(supabase_url, &headers)?;

    request.write_all(payload)?;
    request.flush()?;
    info!("-> POST {}", supabase_url);

    match request.submit() {
        Ok(response) => {
            let status = response.status();
            info!("<- HTTP Status: {}", status);
        }
        Err(e) => {
            info!("Error sending POST request: {:?}", e);
        }
    }

    Ok(())
}
```

The first POST requests will be sent. But then some warning and errors will show up and the program will panic and exit:

```
I (21162) rust_esp32c3_hx711: Iteration: 1
I (21162) rust_esp32c3_hx711: Iteration 1
I (21502) esp-x509-crt-bundle: Certificate validated
I (22222) rust_esp32c3_hx711: -> POST https://pratqgdulutgohggfwfo.supabase.co/rest/v1/messages
I (22812) rust_esp32c3_hx711: <- HTTP Status: 201
I (22812) rust_esp32c3_hx711: Post request sent!
I (27812) rust_esp32c3_hx711: Iteration: 2
I (27812) rust_esp32c3_hx711: Iteration 2
I (33152) rust_esp32c3_hx711: -> POST https://pratqgdulutgohggfwfo.supabase.co/rest/v1/messages
W (38472) HTTP_CLIENT: Connection timed out before data was ready!
I (38472) rust_esp32c3_hx711: Error sending POST request: EspIOError(EspError(-28679))
I (38472) rust_esp32c3_hx711: Post request sent!
I (43482) rust_esp32c3_hx711: Iteration: 4
I (43482) rust_esp32c3_hx711: Iteration 4
thread 'main' panicked at /home/weiying-chen/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.48.1/src/http/client.rs:477:13:
connection is not in initial phase
````

What could be the issue here?

weiying-chen, 2024-03-27:
I've made two discoveries:

1. The same issue happens with a GET request: `get_request(&mut client)?;`
2. To fix the issue you have to create a new HTTP client after the timeout `*client = HttpClient::wrap(EspHttpConnection::new(&config)?);` 

But I still don't know what's causing the timeout.

weiying-chen, 2024-03-27:
I made a discovery. The timeout happens when you send a requests (more than one) to an HTTPS server:

    let url = "https://httpbin.org/get";

The timeout won't happen if you send requests to an HTTP server:

     let url = "http://ifconfig.net/";

weiying-chen, 2024-03-30:
I finally found the solution ... I had to drain the remaining response bytes:

```
 // Drain the remaining response bytes
 while response.read(&mut buf)? > 0 {}
```

ivmarkov, 2024-03-31:
Yes, we don't auto-drain by default, and I'm not sure that's actually possible. If you find a way to auto-drain and open a PR, I think we can accept it.

weiying-chen, 2024-03-31:
Okay, I'll open a PR if I find a way.

PR #404 (2024-04-04): Flush responses to avoid repeated request failures
Fixes #399.

I tested by using a loop that sends many HTTPS requests and by removing the part that reads and drains the data in the `http_client.rs` example. The pattern was the same: `ESP_FAIL` without `flush_response()`. No `ESP_FAIL` with `flush_response()`.

Some considerations:

1. Will flushing the response data in `initiate_request()` cause issues that cannot be observed in the `http_client.rs` example?
2. If the response data is [flushed](https://github.com/weiying-chen/esp-idf-svc/blob/feature/flush-response/src/http/client.rs#L213) i [... cut]
closed 2024-04-04 by Vollbrecht