esp-rs/esp-idf-svc #402: cargo-espflash not picking up partitions.csv?

weiying-chen, 2024-03-30:
I cloned the repo and ran:

```
MCU=esp32c3 cargo espflash flash --target riscv32imc-esp-espidf --example wifi --monitor
```

And I got this error:

```
Error: espflash::image_too_big (link)

  × Supplied ELF image of 1799664B is too big, and doesn't fit configured app partition of 1048576B
  help: Reduce the size of the binary or increase the size of the app partition.
```

This is the current `partitions.csv`:

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        3M,
```

The app partition seems to be set to 3M. And no matter how I change `partitions.csv`, I keep getting the same error.

weiying-chen, 2024-03-30:
I solved the issue by generating my own partitions-table.bin:

```
espflash partition-table --to-binary --output partition-table.bin partitions.csv
```

Then:

```
MCU=esp32c3 cargo espflash flash --target riscv32imc-esp-espidf --example wifi --monitor --partition-table="partition-table.bin"
```

Vollbrecht, 2024-03-30:
If you are using `espflash` than this is not a problem, but `cargo espflash` works a bit different. Because they made changes in the respected versions 3 of `espflash` and `cargo espflash` we now provide the partitions.csv in the root. Though with the extra espflash.toml file it should automatically pick it up. 

What version of `espflash` are you using? It should working with a csv file and if not it may indicate a bug in espflash

weiying-chen, 2024-03-30:
Ah, I updated espflash and cargo-espflash. The error is gone. Sorry, I should have done that before opening this issue.

closed 2024-03-31 by ivmarkov

vult0306, 2024-10-13:
[image]
My espflash is 3.1.1 but the issue is still there

Vollbrecht, 2024-10-13:
> ![image](https://private-user-images.githubusercontent.com/19576509/376049536-81db2323-c4b6-4341-b008-196ba0505dd3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

No it is not a problem. Please read the comments carefully. You are using `espflash` where you have to provide the partiton-table as an argument. Only `cargo espflash` has the capability to pick it up for you.