esp-rs/esp-idf-svc #407: Sporadic crash when running async TLS example

jothan, 2024-04-09:
I have reported the issue here initially, I thought it might be appropriate to also raise the issue here.

https://github.com/smol-rs/async-io/issues/189

Vollbrecht, 2024-04-09:
It may be a xtensa rust specific problem. Can you please also add information on what xtensa compiler version you are currently using?

jothan, 2024-04-09:
I am using the version distributed via espup, updated not long ago.

```
cargo --version --verbose
cargo 1.77.0-nightly (3fe68eabf 2024-02-29)
release: 1.77.0-nightly
commit-hash: 3fe68eabf93cbf3772bbcad09a9206c783e2de3f
commit-date: 2024-02-29
host: x86_64-unknown-linux-gnu
libgit2: 1.7.2 (sys:0.18.2 vendored)
libcurl: 8.5.0-DEV (sys:0.4.70+curl-8.5.0 vendored ssl:OpenSSL/1.1.1w)
ssl: OpenSSL 1.1.1w  11 Sep 2023
os: Debian 12 (bookworm) [64-bit]
```

jothan, 2024-05-14:
Closing in favor of https://github.com/smol-rs/async-lock/issues/84

In all likelihood, it is an issue with the Xtensa LLVM backend.

closed 2024-05-14 by jothan