esp-rs/esp-idf-svc #408: WPA2 Enterprise not connecting

gustavowd, 2024-04-10:
Hi, anyone was able to connect with a WPA2 enterprise network using Rust?

I'm trying with this code, but not luck by now:
`    unsafe {
        let ret1 = esp_idf_sys::esp_eap_client_set_ca_cert(CA_CERT.as_ptr(), CA_CERT.len() as i32);
        let ret2 = esp_idf_sys::esp_eap_client_set_identity(EAP_ID.as_ptr(), EAP_ID.len() as i32);
        let ret3 = esp_idf_sys::esp_eap_client_set_username(EAP_USERNAME.as_ptr(), EAP_USERNAME.len() as i32);
        let ret4 = esp_idf_sys::esp_eap_client_set_password(EAP_PASSWORD.as_ptr(), EAP_PASSWORD.len() as i32);
        let ret5 = esp_idf_sys::esp_eap_client_set_certificate_and_key(CLIENT_CERT.as_ptr(), CLIENT_CERT.len() as i32, CLIENT_KEY.as_ptr(), CLIENT_KEY.len() as i32, std::ptr::null(), 0);
        info!("Wifi Enterprise: {}:{}:{}:{}:{}", ret1, ret2, ret3, ret4, ret5);
        esp_idf_sys::esp_wifi_sta_enterprise_enable();
    }

    wifi.set_configuration(&Configuration::Client(
        ClientConfiguration {
            ssid: SSID.try_into().unwrap(),
            password: PASS.try_into().unwrap(),
            auth_method: AuthMethod::WPA2Enterprise,
            channel,
            ..Default::default()
        }
    ))?;

    info!("Connecting wifi...");

    wifi.connect()?;`

And the full log is:
Full log:

I (691) rust_esp32_c3_demo::wifi_wpa2: Starting wifi...
I (693) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (736) wifi:mode : sta (7c:df:a1:b1:58:0c)
I (737) wifi:enable tsf
I (741) rust_esp32_c3_demo::wifi_wpa2: Scanning...
I (5845) rust_esp32_c3_demo::wifi_wpa2: Found configured access point UTFPR-SERVIDOR on channel 1
I (5846) rust_esp32_c3_demo::wifi_wpa2: Wifi Enterprise: 0:0:0:0:0
I (5854) rust_esp32_c3_demo::wifi_wpa2: Connecting wifi...
I (8270) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (8538) wifi:state: init -> auth (b0)
I (9539) wifi:state: auth -> init (200)
I (9541) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (9542) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (9546) wifi:state: init -> auth (b0)
I (9557) wifi:state: auth -> assoc (0)
I (9573) wifi:state: assoc -> run (10)
I (20862) wifi:state: run -> init (0)
I (20864) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (20865) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (20868) wifi:state: init -> auth (b0)
I (20943) wifi:flush txq
I (20943) wifi:stop sw txq
I (20944) wifi:lmac stop hw txq
I (20946) esp_idf_svc::wifi: EspWifi dropped
I (20947) esp_idf_svc::netif: Dropped
I (20949) esp_idf_svc::netif: Dropped
I (20955) wifi:Deinit lldesc rx mblock:10
I (20959) esp_idf_svc::nvs: NvsDefault dropped
I (20963) esp_idf_svc::eventloop: System event loop dropped
Error: ESP_ERR_TIMEOUT
I (20971) main_task: Returned from app_main()
E (21871) wifi:timeout when WiFi un-init, type=11

Any ideas on this issue?

Paumanok, 2024-04-14:
I'm seeing similar behavior. I didn't see it right away, as my main development machine is kept on rustc 1.75.0-nightly (8edb9b87e 2024-01-02), and my secondary machine is at 1.77.0-nightly(424037dcb 2024-03-18). 

On the older version, wifi is able to connect without much issue.

On this newer version of rustc, I get the same behavior: 

```
I (628) kasa_remote::wifi: Starting wifi...
I (628) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (668) wifi:mode : sta (48:27:e2:7f:c3:64)
I (678) wifi:enable tsf
I (678) kasa_remote::wifi: Connecting wifi...
I (18098) wifi:flush txq
I (18098) wifi:stop sw txq
I (18098) wifi:lmac stop hw txq
I (18098) esp_idf_svc::wifi: EspWifi dropped
I (18098) esp_idf_svc::netif: Dropped
I (18098) esp_idf_svc::netif: Dropped
I (18108) wifi:Deinit lldesc rx mblock:10
I (18118) esp_idf_svc::eventloop: System event loop dropped
Error: Could not connect to Wi-Fi network: ESP_ERR_TIMEOUT
I (18118) main_task: Returned from app_main()
```

gustavowd, 2024-04-15:
@Paumanok, I don't think that's the case. I can connect to a personal network. My problem is with the WPA2 Enterprise network.
I tried the Rust version 1.75 and I have the same issue.

Paumanok, 2024-04-15:
> @Paumanok, I don't think that's the case. I can connect to a personal network. My problem is with the WPA2 Enterprise network. I tried the Rust version 1.75 and I have the same issue.

So I did some more testing after my original response, I was able to connect to my home router's WPA2 network. My issue was in a raspberry pi I've been using as a development network, which it can no longer connect to, which is WPA-PSK. 

I was unable to downgrade to 1.75 manually as well, as the 1.75-nightly is no longer available. Standard 1.75 didn't work. 

I'm wondering if there was some regression that may have a common cause.

Paumanok, 2024-04-16:
Coming back to say, beginner mistake and I didn't downgrade the actual esp toolchain. 

going back to 1.75 via 
```
espup uninstall
espup install --toolchain-version 1.75.0.0
```
Fixed my wifi connection for now, but the issue is still there with 1.77

Vollbrecht, 2024-06-21:
did you do your test always with a debug build or did you build with `--release`. Its unlikely that there is something in the compiler itself that is causing it, probably more of a site effect.

ivmarkov, 2024-09-21:
ALL: Read this: https://github.com/esp-rs/esp-idf-svc/issues/304#issuecomment-2365053299