esp-rs/esp-idf-svc #426: Weird behaviour when both Eth and Wifi are started with static IP config

thetek42, 2024-05-03:
In our application, we are starting both Ethernet and Wi-Fi. When a static IP is configured for both interfaces and an Ethernet cable is connected to the ESP32, connections to a server cannot be made under certain circumstances.

Notes:
- There are no Wi-Fi credentials configured, but the driver is started anyways. `.connect()` is not called.
- The specific place this seems to happen at is the `.to_socket_addrs()` method, which does not terminate. However, I am unsure if DNS is actually the cause of the issue; it is just where I noticed it.
- When DHCP is enabled for both Ethernet and Wi-Fi, everything is fine. Only static IP config seems to be affected.
- When Ethernet is configured to use a static IP and Wi-Fi is set to use DHCP (or vice-versa), the connection cannot be established at all.
- When connecting to Wi-Fi with credentials, then disconnecting once an Ethernet cable has been plugged in, the connection *can* be established all the time. I am unsure how connecting to Wi-Fi in advance and then disconnecting (without stopping the driver) causes Ethernet to work all of a sudden.

These are the cases that I observed:

- **Case 1:**
  - start Ethernet
  - start Wi-Fi
  - try to connect to server, connection fails
- **Case 2:**
  - start Wi-Fi
  - start Ethernet
  - try to connect to server, connection succeeds
- **Case 3:**
  - start Ethernet (without waiting for it to finish)
  - start Wi-Fi
  - wait for Ethernet to finish (`.wait_netif_up()`)
  - try to connect to server. Here, my observations varied. With BlockingEth, I could never manage to get a connection to a server. With AsyncEth, I could get a connnection in approximately 1 of 5 of cases. I tried the same thing with AsyncEth again some time later, but for some reason I could not get a connection there, not even in the rare 20%-ish chance that I observed earlier.

Below is a piece of sample code that can be used to (hopefully) reproduce the issue. Feel free to play around a bit by uncommenting stuff and moving stuff around (e.g. putting Wi-Fi after Ethernet or putting the `eth.wait_*()` block after the Wi-Fi code).

<details>
<summary>Code</summary>

```rust
#![allow(unused_imports)]

use std::io::{Read, Write};
use std::net::TcpStream;

use esp_idf_svc::eth::{EspEth, EthDriver, RmiiEthChipset, RmiiClockConfig, BlockingEth, AsyncEth};
use esp_idf_svc::eventloop::EspSystemEventLoop;
use esp_idf_svc::hal::gpio::{self, PinDriver};
use esp_idf_svc::hal::prelude::Peripherals;
use esp_idf_svc::ipv4::{ClientConfiguration, Configuration, ClientSettings, Subnet, Mask, DHCPClientSettings};
use esp_idf_svc::log::EspLogger;
use esp_idf_svc::netif::{NetifConfiguration, EspNetif};
use esp_idf_svc::timer::EspTaskTimerService;
use esp_idf_svc::wifi::{self, WifiDriver, EspWifi, BlockingWifi, AuthMethod, AsyncWifi};

fn main() -> anyhow::Result<()> {
    esp_idf_svc::sys::link_patches();
    EspLogger::initialize_default();

    let peripherals = Peripherals::take()?;
    let pins = peripherals.pins;
    let sys_loop = EspSystemEventLoop::take()?;
    let timer_service_wifi = EspTaskTimerService::new()?;
    let timer_service_eth = EspTaskTimerService::new()?;

    let mut eth_pwr = PinDriver::output(pins.gpio5)?;
    let mut clk_en = PinDriver::output(pins.gpio4)?;
    eth_pwr.set_low()?;
    clk_en.set_low()?;
    std::thread::sleep(std::time::Duration::from_millis(100));
    eth_pwr.set_high()?;
    std::thread::sleep(std::time::Duration::from_millis(10));
    clk_en.set_high()?;
    std::thread::sleep(std::time::Duration::from_millis(10));

    log::info!("--- EthDriver");
    let eth_driver = EthDriver::new(
        peripherals.mac,
        pins.gpio25,
        pins.gpio26,
        pins.gpio27,
        pins.gpio23,
        pins.gpio22,
        pins.gpio21,
        pins.gpio19,
        pins.gpio18,
        RmiiClockConfig::<gpio::Gpio0, gpio::Gpio16, gpio::Gpio17>::Input(pins.gpio0),
        Some(pins.gpio33),
        RmiiEthChipset::LAN87XX,
        None,
        sys_loop.clone(),
    )?;

    log::info!("--- EspNetif");
    let netif = EspNetif::new_with_conf(&NetifConfiguration {
        //ip_configuration: Configuration::Client(ClientConfiguration::DHCP(DHCPClientSettings {
        //    hostname: Some("ethwifitest".try_into().unwrap()),
        //})),
        ip_configuration: Configuration::Client(ClientConfiguration::Fixed(ClientSettings {
            ip: "192.168.178.234".parse()?,
            subnet: Subnet {
                gateway: "192.168.178.1".parse()?,
                mask: Mask(24),
            },
            dns: Some("192.168.178.1".parse()?),
            secondary_dns: None,
        })),
        ..NetifConfiguration::eth_default_client()
    })?;

    log::info!("--- EspEth");
    let eth = EspEth::wrap_all(eth_driver, netif)?;

    //log::info!("--- BlockingEth");
    //let mut eth = BlockingEth::wrap(eth, sys_loop.clone())?;
    //log::info!("--- eth.start()");
    //eth.start()?;
    //log::info!("--- eth.wait_connected()");
    //eth.wait_connected()?;
    //log::info!("--- eth.wait_netif_up()");
    //eth.wait_netif_up()?;

    log::info!("--- AsyncEth");
    let mut eth = AsyncEth::wrap(eth, sys_loop.clone(), timer_service_eth)?;
    esp_idf_svc::hal::task::block_on(async {
        log::info!("--- eth.start()");
        eth.start().await
    })?;
    esp_idf_svc::hal::task::block_on(async {
        log::info!("--- eth.wait_connected()");
        eth.wait_connected().await?;
        log::info!("--- eth.wait_netif_up()");
        eth.wait_netif_up().await
    })?;

    log::info!("--- WifiDriver");
    let wifi_driver = WifiDriver::new(peripherals.modem, sys_loop.clone(), None)?;

    let netif_sta = EspNetif::new_with_conf(&NetifConfiguration {
        //ip_configuration: Configuration::Client(ClientConfiguration::DHCP(DHCPClientSettings {
        //    hostname: Some("ethwifitest".try_into().unwrap()),
        //})),
        ip_configuration: Configuration::Client(ClientConfiguration::Fixed(ClientSettings {
            ip: "192.168.178.234".parse()?,
            subnet: Subnet {
                gateway: "192.168.178.1".parse()?,
                mask: Mask(24),
            },
            dns: Some("192.168.178.1".parse()?),
            secondary_dns: None,
        })),
        ..NetifConfiguration::wifi_default_client()
    })?;

    let netif_ap = EspNetif::new_with_conf(&NetifConfiguration {
        ip_configuration: Configuration::Client(ClientConfiguration::DHCP(DHCPClientSettings {
            hostname: None,
        })),
        ..NetifConfiguration::wifi_default_router()
    })?;

    log::info!("--- EspWifi");
    let wifi = EspWifi::wrap_all(wifi_driver, netif_sta, netif_ap)?;

    //log::info!("--- BlockingWifi");
    //let mut wifi = BlockingWifi::wrap(wifi, sys_loop.clone())?;
    log::info!("--- AsyncWifi");
    let mut wifi = AsyncWifi::wrap(wifi, sys_loop.clone(), timer_service_wifi)?;

    log::info!("--- wifi.set_configuration()");
    wifi.set_configuration(&wifi::Configuration::Client(wifi::ClientConfiguration {
        ssid: "".try_into().unwrap(),
        bssid: None,
        auth_method: AuthMethod::WPA2Personal,
        password: "".try_into().unwrap(),
        channel: None,
    }))?;

    //log::info!("--- wifi.start()");
    //wifi.start()?;
    log::info!("--- wifi.start()");
    esp_idf_svc::hal::task::block_on(wifi.start())?;

    log::info!("--- TcpStream");
    let mut stream = TcpStream::connect("example.com:80")?;
    stream.write_all(b"GET / HTTP/1.1\r\nHost:example.com\r\n\r\n")?;

    log::info!("--- response");
    let mut buf = [0; 512];
    loop {
        let read = stream.read(&mut buf)?;
        print!("{}", core::str::from_utf8(&buf[..read])?);
        if read == 0 {
            break;
        }
    }
    log::info!("--- done");

    Ok(())
}
```
</details>

ivmarkov, 2024-05-03:
Do any of your problems happen, if you replace "example.com" with an IP address?

thetek42, 2024-05-03:
Yes. It does not matter if I use example.com or the corresponding IP address.

ivmarkov, 2024-05-03:
IYou are sure 100% about that (as in you have used `IpAddr` and not a string that might still trigger DNS resolution somehow)?

ivmarkov, 2024-05-03:
Your example is way too big and I have difficulties understanding what exactly you are trying to achieve...

This seems very odd though:
```
let netif_ap = EspNetif::new_with_conf(&NetifConfiguration {
        ip_configuration: Configuration::Client(ClientConfiguration::DHCP(DHCPClientSettings {
            hostname: None,
        })),
        ..NetifConfiguration::wifi_default_router()
    })?;
```

How is this supposed to even work?

thetek42, 2024-05-03:
> How is this supposed to even work?

That must have snuck in there somehow. I had it set to a regular ClientConfiguration::Fixed previously, but it seems like I somehow managed to clone the config into the wrong place 🤷. It does not affect the result though.

ivmarkov, 2024-05-03:
Even `Fixed` is wrong. You can't set a client config on an access point and expect not to have issues w.r.t. routing on IP level. Why are you configuring the access point in the first place?

 Can you:
a) try to simplify this example to the bare minimum. Remove everything not necessary, including dns resolutions and whatnot
b) try to report _one_ problem only at first and an example about this single one problem.

thetek42, 2024-05-03:
> You are sure 100% about that (as in you have used `IpAddr` and not a string that might still trigger DNS resolution somehow)?

I specified it as `(Ipv4Addr::new(93, 184, 215, 14), 80)`.

thetek42, 2024-05-03:
I updated the code in the original post.

thetek42, 2024-05-03:
Also, this issue has nothing to do with the Access Point. EspWifi::wrap_all simply requires a netif for the AP to be present.

ivmarkov, 2024-05-03:
Maybe. But it has a default configuration and you are changing it. Maybe it does not matter, as the Wifi is not in Mixed mode, but maybe there is.

My point is the following: if you leave the example as-is, and with all the cases you have enumerate, it might take me days, if not weeks to get there. 

And if/when I get there, I'll start by anyway with eliminating every single line of code in the example which should not be there, which is additional unknown variable that only confuses and obfuscates the issue. Like the DNS issue. Or configuring the AP. Etc. And then try to tackle just one case of the ones you have enumerated and which should work, but isn't.

... or you could try to do it, and then I can help with thinking / brainstorming. I think it might be faster that way. :)

ivmarkov, 2024-05-03:
> I updated the code in the original post.

In your "simplified example"... why are you setting the eth and wifi-sta to the same IP, namely "192.168.178.234"?

ivmarkov, 2024-05-03:
Also, for the `netif_ap` (I know it should not affect anything, but still) - can you please use the **default** network configuration, which would then be different from your 192.168.178.1/24, that you are setting everywhere.

ivmarkov, 2024-05-06:
@thetek42 What happens once you assign different static IPs to Eth and Wifi?

thetek42, 2024-05-07:
Sorry for the delay, I was busy the last couple of days.

Using the default values for the AP netif does not change anything.

Using a different IP address for Wi-Fi and Ethernet does in fact change the behaviour, except that it makes it worse. When both have a different static IP address configured, no matter in which order Wi-Fi and Ethernet are initialized, the connection to the server is not possible.

Most of the time, the device does not even show up in my router's user interface if that happens. However, I was able to observe it show up in the list of connections once, and then never again. In that incident, it showed the device name as "PC-192-168-178-235", even though the IP address was listed as 192.168.178.**234** -- while the static IP address configuration for Ethernet was set to 192.168.178.**235** and the IP for Wi-Fi was set to 192.168.178.**234**. I am unsure if this was just a bug by the router or if this is the actual behaviour. The fact that the device does not show up in the list makes sense considering that running it through Wireshark yielded no transmitted packets, but the fact that it turned up in the list exactly once does not make sense to me.

The more I investigate this issue, the more confused I am.

Again, I will update the code in the original post in order to use of default values for AP config.

ivmarkov, 2024-05-07:
One little detail that is interesting: what list do you mean? The router does not keep any "list" of devices with static IP addresses (let alone knowing their host names) as it is just not aware of these devices.

More advanced routers might produce a "list" based on tx/rx packet statistics and potentially, based on nat statistics.

Btw: have you tried pinging the eth and wifi interfaces from another pc?

Another experiment would be to assign a static ip to the wifi interface which is from a completely different subnet (even if the wifi won't be reachable this way).

thetek42, 2024-05-07:
I was talking about the list that the router provides in the user interface of its website.

thetek42, 2024-05-07:
A different IP in a different subnet for the Wi-Fi interface does not have any effect.

Pinging the Ethernet interface *does* work.

thetek42, 2024-05-07:
Also, when setting the Ethernet IP address to .235 and the Wi-Fi IP address to .234, I can ping *both*, even though Wi-Fi is not connected.

ivmarkov, 2024-05-07:
> I was talking about the list that the router provides in the user interface of its website.

See above. :) I would not trust this list for static IPs **at all**. :)

ivmarkov, 2024-05-07:
> Also, when setting the Ethernet IP address to .235 and the Wi-Fi IP address to .234, I can ping _both_, even though Wi-Fi is not connected.

OK, _hold on_. So just to summarize. What you are saying:
- Both interfaces are pingable from outside the Esp
- The Wifi is **not** connected

Are you sure the Wifi is not connected? Otherwise, how pinging _that_ IP would work?! Are you sure you don't have an IP conflict in your net, and you are not pinging something else, which is not the Esp?

thetek42, 2024-05-07:
Ok, I just tried it again and cannot reproduce it.

thetek42, 2024-05-07:
I used the `ping` command for pinging the esp32.

It seems that it only happens for approximately 15-20 secs after flashing a new firmware on the device with a *different* static IP than the previous firmware. After that timespan, it does not occur again. The only way I can explain that is that there is some nasty caching going on somewhere, not neccesarily on esp-idf's side.

Pinging the gateway from the esp32 provides the exact same behaviour as opening a socket to somewhere. For this, I used the `EspPing`. However, I was a bit unsure as to what to set the interface to, so I left it at 0. From what I gathered, you are supposed to put the netif index in there, but obtaining the netif index via `.get_index()` and plugging that into `EspPing` yielded to no ping succeeding, even when only Ethernet was enabled in the code. Is there something I missed or misunderstood?

ivmarkov, 2024-05-07:
> Pinging the gateway from the esp32 provides the exact same behaviour as opening a socket to somewhere. For this, I used the `EspPing`. However, I was a bit unsure as to what to set the interface to, [...]

By the way, what are you trying to ping, and did you check that it is pingable in the first place?

thetek42, 2024-05-07:
I am pinging the gateway (192.168.178.1).

ivmarkov, 2024-05-07:
Ok assuming it is pingable from inside (99.9% routers are, but you never know and you should check it from a pc), can you try with just the ethernet connection, without even creating the wifi driver? Until we get a reliable ping, we cannot progress any further. Try with get_index + 1, or with index = 1 and/or = 2 until it works.

ivmarkov, 2024-05-07:
And just to confirm again: None of these issues happen if you don't create the wifi driver in the first place? But once you create it, even if you dont start id, issues start?

thetek42, 2024-05-10:
- Pinging the router from a PC works as expected.
- Pinging the router with `EspPing::default()`:
  - Only works when the Ethernet driver is created after the Wi-Fi driver.
  - When Ethernet and Wi-Fi have differing static IP configs, the TCP connection to a server is not possible. `EspPing` seems unaffected by this and continues working even with two different static IPs.
- Pinging the router directly with the Ethernet interface *(requires **`get_index() + 1`**)* works - even when creating the Wi-Fi interface after the Ethernet interface. I guess that is to be expected since we are directly using the Ethernet interface. I guess the `get_index() + 1` behaviour could be fixed or at least documented somewhere, that way people won't stumble over it in the future.
- Not starting the Wi-Fi driver in the first place causes everything to work as expected. The problems only occur when calling `wifi.start()`. Just creating but *not* starting is fine.

thetek42, 2024-05-17:
So, I did some more investigation and implemented this in C. There, it works perfectly and as expected. See below.

<details>
<summary>C Code</summary>

```c
#include <driver/gpio.h>
#include <esp_check.h>
#include <esp_eth.h>
#include <esp_eth_driver.h>
#include <esp_event.h>
#include <esp_http_client.h>
#include <esp_log.h>
#include <esp_mac.h>
#include <esp_netif.h>
#include <esp_wifi.h>
#include <freertos/FreeRTOS.h>
#include <nvs_flash.h>

#define WIFI_SSID     ""
#define WIFI_PASS     ""
#define ETH_PHY_ADDR  -1
#define ETH_PIN_RESET 33
#define ETH_PIN_MDC   23
#define ETH_PIN_MDIO  18
#define ETH_PIN_PWR   5
#define ETH_PIN_CLK   4
#define GATEWAY       "192.168.178.1"
#define SUBNET        "255.255.255.0"
#define ETH_IP        "192.168.178.234"
#define WIFI_IP       "192.168.178.234"
#define DNS_1         "192.168.178.1"
#define DNS_2         "192.168.178.1"
#define HTTP_HOST     "example.com"
#define HTTP_PORT     80

#define WIFI_MAX_RETRY     3
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static void init_netif (void);
static void nvs_init (void);
static void set_static_ip (esp_netif_t *netif, const char *ip);
static void wifi_init (void);
static void wifi_event_handler (void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static void eth_init (void);
static void eth_event_handler (void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static void send_http_request (void);
static esp_err_t http_event_handler (esp_http_client_event_t *ev);

static size_t s_wifi_retry_num = 0;
static EventGroupHandle_t s_wifi_event_group = NULL;

const char *TAG = "main";

void
app_main (void)
{
    nvs_init ();
    init_netif ();
    eth_init ();
    wifi_init ();
    send_http_request ();
}

static void
nvs_init (void)
{
    esp_err_t ret;

[... 196 lines cut ...]
    cli = esp_http_client_init (&cfg);

    ESP_ERROR_CHECK (esp_http_client_perform (cli));
    ESP_LOGI (TAG, "-- http ok, status %d, len %" PRId64,
                esp_http_client_get_status_code (cli),
                esp_http_client_get_content_length (cli));

    esp_http_client_cleanup (cli);
}

static esp_err_t
http_event_handler (esp_http_client_event_t *ev)
{
    return ESP_OK;
}
```
</details>

<details>
<summary>sdkconfig.defaults</summary>

```
ETH_USE_ESP32_EMAC=y
```
</details>

ivmarkov, 2024-05-17:
@thetek42 Weird. Did you just delete your last comment from 5 minutes ago?

thetek42, 2024-05-17:
Yes, I still had to fix one small part of the code.

I investigated the issue a bit further by trying it in C. It works perfectly there.

<summary>
<details>C Code</details>

```c
#include <driver/gpio.h>
#include <esp_check.h>
#include <esp_eth.h>
#include <esp_eth_driver.h>
#include <esp_event.h>
#include <esp_http_client.h>
#include <esp_log.h>
#include <esp_mac.h>
#include <esp_netif.h>
#include <esp_wifi.h>
#include <freertos/FreeRTOS.h>
#include <netdb.h>
#include <nvs_flash.h>

#define WIFI_SSID     ""
#define WIFI_PASS     ""
#define ETH_PHY_ADDR  -1
#define ETH_PIN_RESET 33
#define ETH_PIN_MDC   23
#define ETH_PIN_MDIO  18
#define ETH_PIN_PWR   5
#define ETH_PIN_CLK   4
#define GATEWAY       "192.168.178.1"
#define SUBNET        "255.255.255.0"
#define ETH_IP        "192.168.178.234"
#define WIFI_IP       "192.168.178.234"
#define DNS_1         "192.168.178.1"
#define DNS_2         "192.168.178.1"
#define HTTP_HOST     "example.com"
#define HTTP_PORT     80

#define WIFI_MAX_RETRY     3
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static void init_netif (void);
static void nvs_init (void);
static void set_static_ip (esp_netif_t *netif, const char *ip);
static void wifi_init (void);
static void wifi_event_handler (void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static void eth_init (void);
static void eth_event_handler (void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data);
static void resolve_hostname (void);
static void send_http_request (void);
static esp_err_t http_event_handler (esp_http_client_event_t *ev);

static size_t s_wifi_retry_num = 0;
static EventGroupHandle_t s_wifi_event_group = NULL;

const char *TAG = "main";

void
app_main (void)
{
    nvs_init ();
    init_netif ();
    eth_init ();
    wifi_init ();
    resolve_hostname ();
    send_http_request ();
}

[... 226 lines cut ...]
    cli = esp_http_client_init (&cfg);

    ESP_ERROR_CHECK (esp_http_client_perform (cli));
    ESP_LOGI (TAG, "-- http ok, status %d, len %" PRId64,
                esp_http_client_get_status_code (cli),
                esp_http_client_get_content_length (cli));

    esp_http_client_cleanup (cli);
}

static esp_err_t
http_event_handler (esp_http_client_event_t *ev)
{
    return ESP_OK;
}
```
</summary>

<summary>
<details>sdkconfig.defaults</details>

```
ETH_USE_ESP32_EMAC=y
```
</summary>

ivmarkov, 2024-05-17:
OK, that's great news. The question is now, what is different in the Rust code, if you compare it with C...?

thetek42, 2024-05-17:
I don't see any differences. It starts Ethernet, then it starts Wi-Fi (without credentails and without waiting for it connect). Both Ethernet and Wi-Fi have a static IP assigned to them (the same static IP, in fact). Then, I try to resolve a hostname via DNS and try to send a HTTP request to it. Both of those work.

ivmarkov, 2024-05-17:
No I mean the rust code in `esp-idf-svc` which is driving the netif stack, the wifi and the ethernet drivers.
There _must_ be a difference. And sorry for persisting that you look at it instead of me doing a deep dive. :(
I simply don't have the physical time to do it these days, as I'm chasing other problems.

thetek42, 2024-05-21:
Ok. I figured out what the issue is. If you initialize the netif like it is currently done in esp-idf-svc (with different flags and ip_info set), it does not work. However, if you do it just like the official C example [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/static_ip/main/static_ip_example_main.c) (that is, initializing the netif with dhcp enabled, and then disabling it and setting the static IP config afterwards), it works perfectly fine. Below is a `git diff` of what I had to get it to work:

```diff
diff --git a/src/netif.rs b/src/netif.rs
index 0bc1b784d..72a0daf1e 100644
--- a/src/netif.rs
+++ b/src/netif.rs
@@ -235,38 +235,13 @@ impl EspNetif {
         {
             ipv4::Configuration::Client(ref ip_conf) => (
                 esp_netif_inherent_config_t {
-                    flags: match ip_conf {
-                        ipv4::ClientConfiguration::DHCP(_) => {
-                            esp_netif_flags_ESP_NETIF_DHCP_CLIENT
-                                | esp_netif_flags_ESP_NETIF_FLAG_GARP
-                                | esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED
-                        }
-                        ipv4::ClientConfiguration::Fixed(_) => {
-                            esp_netif_flags_ESP_NETIF_FLAG_AUTOUP
-                        }
-                    },
+                    flags: esp_netif_flags_ESP_NETIF_DHCP_CLIENT
+                        | esp_netif_flags_ESP_NETIF_FLAG_GARP
+                        | esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED,
                     mac: initial_mac,
                     ip_info: ptr::null(),
-                    get_ip_event: match ip_conf {
-                        ipv4::ClientConfiguration::DHCP(_) => {
-                            if conf.stack == NetifStack::Sta {
-                                ip_event_t_IP_EVENT_STA_GOT_IP
-                            } else {
-                                0
-                            }
-                        }
-                        ipv4::ClientConfiguration::Fixed(_) => 0,
-                    },
-                    lost_ip_event: match ip_conf {
-                        ipv4::ClientConfiguration::DHCP(_) => {
-                            if conf.stack == NetifStack::Sta {
-                                ip_event_t_IP_EVENT_STA_LOST_IP
-                            } else {
-                                0
-                            }
-                        }
-                        ipv4::ClientConfiguration::Fixed(_) => 0,
-                    },
+                    get_ip_event: ip_event_t_IP_EVENT_STA_GOT_IP,
+                    lost_ip_event: ip_event_t_IP_EVENT_STA_LOST_IP,
                     if_key: c_if_key.as_c_str().as_ptr() as _,
                     if_desc: c_if_description.as_c_str().as_ptr() as _,
                     route_prio: conf.route_priority as _,
@@ -297,11 +272,8 @@ impl EspNetif {
             ),
             ipv4::Configuration::Router(ref ip_conf) => (
                 esp_netif_inherent_config_t {
-                    flags: (if ip_conf.dhcp_enabled {
-                        esp_netif_flags_ESP_NETIF_DHCP_SERVER
-                    } else {
-                        0
-                    }) | esp_netif_flags_ESP_NETIF_FLAG_AUTOUP,
+                    flags: esp_netif_flags_ESP_NETIF_DHCP_SERVER
+                        | esp_netif_flags_ESP_NETIF_FLAG_AUTOUP,
                     mac: initial_mac,
```

ivmarkov, 2024-05-21:
This is great news!...
... but a bit weird that we have to do it in such a strange way. What if there is simply no DHCP server on the network and we are just sitting there waiting for a DHCP address? Also the change where we run the AP DHCP server and then stop it is especially annoying. For a short while we would be running a DHCP server, and that might be unexpected for the user. Also, that last trick is not part of the C example anyway (the example is for STA only of course, but still)

Would you do another variation of your changes:
* Set `get_ip_event` and `lost_ip_event` to always be equal to `ip_event_t_IP_EVENT_STA_GOT_IP` and `ip_event_t_IP_EVENT_STA_LOST_IP` respectively? Not sure if that matters, but why not?
* For the fixed client configuration (STA) we are currently not setting the `esp_netif_flags_ESP_NETIF_FLAG_GARP` and `esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED`. Not sure about 
`esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED`, but my gut feeling is `esp_netif_flags_ESP_NETIF_FLAG_GARP` might be important and the fact that we are not setting it for the fixed configuration *might actually be a bug*.  So how about - for the STA fixed case - if we set the `flags` to 
```
esp_netif_flags_ESP_NETIF_FLAG_AUTOUP
| esp_netif_flags_ESP_NETIF_FLAG_GARP
| esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED
```
* Remove the 
```
esp!(unsafe { esp_netif_dhcpc_stop(handle.0) })?;
esp!(unsafe { esp_netif_dhcps_stop(handle.0) })?;
```
... as we would not be doing the DHCP trick this way.

thetek42, 2024-05-21:
So, I did some more digging around. Apparently the only thing you actually need to do in order to make it work is to remove the `esp_netif_flags_ESP_NETIF_FLAG_AUTOUP`! Simply replacing it with 0 makes it behave just as expected. (As already mentioned, the diff above was simply me messing around in an attempt to get it to work)

ivmarkov, 2024-05-21:
> So, I did some more digging around. Apparently the only thing you actually need to do in order to make it work is to remove the `esp_netif_flags_ESP_NETIF_FLAG_AUTOUP`! Simply replacing it with 0 [...]

You mean from the client (STA) configuration only? I guess it should stay in the server (AP) configuration?

ivmarkov, 2024-05-21:
Can you try with latest `master`? I just removed the `AUTOUP` flag from the client conf, and also set the `GARP` and `IP_MODFIFIED` flags even for fixed client configuration.

thetek42, 2024-05-21:
> You mean from the client (STA) configuration only? I guess it should stay in the server (AP) configuration?

I only tried STA. I don't know about AP, but I'll check that as well when I have the time to do so. But I *think* that it should be fine with AP since I never had any issues with it.

> Can you try with latest master? I just removed the AUTOUP flag from the client conf, and also set the GARP and IP_MODFIFIED flags even for fixed client configuration.

Seems to be working now. Thanks!!

ivmarkov, 2024-05-21:
Thank you for persisting through this journey and figuring out the root cause!

closed 2024-05-21 by ivmarkov