esp-rs/esp-idf-svc #436: Error when running "cargo  build" after running "cargo generate esp-rs/esp-idf-template generate" on Ubuntu

Spyker2000, 2024-06-21:
[image]

Vollbrecht, 2024-06-21:
what ESP_IDF_VERSION are you using? 

Also can you please check if the issue persist with our close to release esp-idf-svc version from git?

You can test them out by adding the following into your Cargo.toml
```toml
[patch.crates-io]
embuild ={ git = "https://github.com/esp-rs/embuild"}
esp-idf-svc ={ git = "https://github.com/esp-rs/esp-idf-svc"}
esp-idf-hal ={ git = "https://github.com/esp-rs/esp-idf-hal"}
esp-idf-sys ={ git = "https://github.com/esp-rs/esp-idf-sys"}
embedded-svc ={ git = "https://github.com/esp-rs/embedded-svc"}
```
and then run `cargo clean` + `cargo update` and then run `cargo build`

Spyker2000, 2024-06-22:
I did everything requested and then got this when building, 

[image]

Vollbrecht, 2024-06-22:
a) do you have any ESP_IDF specifc env variables in your environement? For example does `env | grep ESP` give any output?
b) what ESP_IDF_VERSION is written inside the `.cargo/config.toml` in your project root?

Spyker2000, 2024-06-22:
a)
 `gavin@gavin-ubuntu:~$ env | grep ESP`

` gavin@gavin-ubuntu:~$ env | grep esp`

``` IDF_PATH=/home/gavin/esp/esp-idf
PATH=/home/gavin/.cargo/bin:/home/gavin/esp/esp-idf/tools:/home/gavin/.local/bin:/home/gavin/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
```

b)

gavin@gavin-ubuntu:~/gavin_rust_1$ ls -al
total 124

``````````````
drwxrwxr-x   9 gavin gavin  4096 Jun 21 11:31 .
drwxr----- 151 gavin gavin 20480 Jun 21 14:35 ..
drwxrwxr-x   2 gavin gavin  4096 Jun 17 11:16 build
-rw-r--r--   1 gavin gavin    53 Jun 16 16:54 build.rs
drwxrwxr-x   2 gavin gavin  4096 Jun 16 16:54 .cargo
-rw-rw-r--   1 gavin gavin 49669 Jun 21 11:21 Cargo.lock
-rw-r--r--   1 gavin gavin  1268 Jun 21 11:19 Cargo.toml
drwxrwxr-x   3 gavin gavin  4096 Jun 16 16:55 .embuild
drwxrwxr-x   6 gavin gavin  4096 Jun 21 11:32 .git
-rw-r--r--   1 gavin gavin    39 Jun 16 16:54 .gitignore
-rw-r--r--   1 gavin gavin    28 Jun 16 16:54 rust-toolchain.toml -rw-r--r--   1 gavin gavin   465 Jun 16 16:54 sdkconfig.defaults
drwxrwxr-x   2 gavin gavin  4096 Jun 16 16:54 src drwxrwxr-x   4 gavin gavin  4096 Jun 21 11:32 target> drwxrwxr-x   2 gavin gavin  4096 Jun 16 16:54 .vscode
`````````````
c)
![cargo options](https://github.com/esp-rs/esp-idf-svc/assets/7892368/a25b33bc-a965-48b1-8523-5c3bf10de630)

Spyker2000, 2024-06-22:
[image]

Spyker2000, 2024-06-22:
I originally started (about a year ago) with  Espressif cli when I purchased some ESP32's and then shifted to Platformio and then started with Rust about a month ago. Programmed for many years with Arduino IDE.

Spyker2000, 2024-06-22:
I have also tried the recommended alias **get_esprs** when I open a NEW terminal and I get the same error message.

````````
gavin@gavin-ubuntu:~/gavin_rust_1$ get_esprs
gavin@gavin-ubuntu:~/gavin_rust_1$ env | grep esp
IDF_PATH=/home/gavin/esp/esp-idf
LIBCLANG_PATH=/home/gavin/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.4-20231113/esp-clang/lib
PATH=/home/gavin/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/gavin/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/gavin/.cargo/bin:/home/gavin/esp/esp-idf/tools:/home/gavin/.local/bin:/home/gavin/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
````````

Vollbrecht, 2024-06-22:
What you are seeing should only happen when for one reason or another the used esp-idf-version does not correspond with the one we want to build for. 

By default with the settings you used from the cargo generate project, we pull the corresponding version of esp-idf ourself into the .embuild directory and also put the necessary tools there to build it. We don't set any additional env variables, they are only directly put forward by `cargo`, though we pick them up if a user brings it's own esp-idf installation into play. 

If you provide your own IDF_PATH that is not corresponding to the ESP_IDF_VERSION set inside the .cargo/config.toml file than there will be confusion in the build system. Simmilar problems arise with incompatible xtensa/riscv gcc tool version installed. 

The simplest way is that you use an environement where you dont provide an IDF_PATH so its all automatically handeld for you, otherwise you would manually need to match your system installed esp-idf version with the one you are setting to use in the template.

Spyker2000, 2024-06-24:
I will take a look at tonight and see what I can do. I will let you know the outcome.

Spyker2000, 2024-06-24:
There is however a PATH setting **/home/gavin/esp/esp-idf/tools** which is not a Enviromental variable as such. I don"t think this is an issue.

`PATH=/home/gavin/.cargo/bin:/home/gavin/esp/esp-idf/tools:/home/gavin/.local/bin:/home/gavin/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
`

Spyker2000, 2024-06-24:
I created a new project /home/gavin/gavin_amended_1/
I had the same  errors with my original project.

Vollbrecht, 2024-06-24:
The error around the `git clone` seams like a spurios network error, that sounds a bit unfortunate. To make sure i would delete the `.embuild` dir inside your project and when a new build is run it should be creating a new instance off .embuild dir.

Regarding other stuff set in your PATH, that should not be a problem, as yes we look into the path, but the tools we are using are prepending to the PATH, e.g first reading from the tools inside .embuild. That was only a problem if you provide the specific IDF_PATH env var.

Spyker2000, 2024-06-25:
I tried again this morning with the errors. 

I deleted **.embuild**

I did a clean, update and build.

```
gavin@gavin-ubuntu:~/gavin_amended_1$ cargo clean
     Removed 1406 files, 1.1GiB total
```gavin@gavin-ubuntu:~/gavin_amended_1$ ls .embuild/
espressif
gavin@gavin-ubuntu:~/gavin_amended_1$ ls .embuild/ -al
total 12
drwxrwxr-x 3 gavin gavin 4096 Jun 25 13:49 .
drwxrwxr-x 8 gavin gavin 4096 Jun 25 13:49 ..
drwxrwxr-x 3 gavin gavin 4096 Jun 25 13:49 espressif
gavin@gavin-ubuntu:~/gavin_amended_1$ cd .embuild/
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild$ ls
espressif
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild$ cd espressif/
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild/espressif$ ls
esp-idf
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild/espressif$ cd esp-idf/
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild/espressif/esp-idf$ ls
gavin@gavin-ubuntu:~/gavin_amended_1/.embuild/espressif/esp-idf$ 

 and a cargo update

```
gavin@gavin-ubuntu:~/gavin_amended_1$ cargo update
    Updating git repository `https://github.com/esp-rs/embedded-svc`
    Updating git repository `https://github.com/esp-rs/embuild`
    Updating git repository `https://github.com/esp-rs/esp-idf-hal`
    Updating git repository `https://github.com/esp-rs/esp-idf-svc`
    Updating git repository `https://github.com/esp-rs/esp-idf-sys`
    Updating crates.io index
    Removing embedded-svc v0.28.0
      Adding embedded-svc v0.28.0 (https://github.com/esp-rs/embedded-svc#30133e01)
    Removing embuild v0.32.0
      Adding embuild v0.32.0 (https://github.com/esp-rs/embuild#5c30249c)
    Removing esp-idf-hal v0.44.0
      Adding esp-idf-hal v0.44.0 (https://github.com/esp-rs/esp-idf-hal#d849cac7)
    Removing esp-idf-svc v0.49.0
      Adding esp-idf-svc v0.49.0 (https://github.com/esp-rs/esp-idf-svc#128607c5)
    Removing esp-idf-sys v0.35.0
      Adding esp-idf-sys v0.35.0 (https://github.com/esp-rs/esp-idf-sys#ff086e7d)

```
and then a cargo build ,
              
and waited about 10-15 mins  and got the following ,                                           

```
gavin@gavin-ubuntu:~/gavin_amended_1$ cargo build
   Compiling bindgen v0.69.4
   Compiling embuild v0.32.0 (https://github.com/esp-rs/embuild#5c30249c)
   Compiling esp-idf-sys v0.35.0 (https://github.com/esp-rs/esp-idf-sys#ff086e7d)
   Compiling esp-idf-hal v0.44.0 (https://github.com/esp-rs/esp-idf-hal#d849cac7)
   Compiling esp-idf-svc v0.49.0 (https://github.com/esp-rs/esp-idf-svc#128607c5)
   Compiling gavin_amended_1 v0.1.0 (/home/gavin/gavin_amended_1)
error: failed to run custom build command for `esp-idf-sys v0.35.0 (https://github.com/esp-rs/esp-idf-sys#ff086e7d)`

Caused by:
  process didn't exit successfully: `/home/gavin/gavin_amended_1/target/debug/build/esp-idf-sys-3cbdaafc966db396/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Cloning into '/home/gavin/gavin_amended_1/.embuild/espressif/esp-idf/v5.2.2'...
  error: 7438 bytes of body are still expected
  fetch-pack: unexpected disconnect while reading sideband packet
  fatal: early EOF
  fatal: fetch-pack: invalid index-pack output
  Error: Could not install esp-idf

  Caused by:
      command '"git" "clone" "--jobs=12" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.2.2" "https://github.com/espressif/esp-idf.git" "/home/gavin/gavin_amended_1/.embuild/espressif/esp-idf/v5.2.2"' exited with non-zero status code 128

```

The .embuild directory is re-created but the esp-idf directory is empty

```
[image]

Vollbrecht, 2024-06-25:
hmm can you test if you can manually successfully run the git clone in an empty directory and see if it errors out with the same message? 
E.g run the exact command like this `git clone --jobs=12 --recursive --depth 1 --shallow-submodules --branch v5.2.2 https://github.com/espressif/esp-idf.git`  and it should create an directory called esp-idf that contains the idf plus all submodules

Vollbrecht, 2024-06-25:
you may also want to read this on [stackoverflow](https://stackoverflow.com/a/67390274)

Spyker2000, 2024-06-25:
gavin@gavin-ubuntu:~$ git clone --jobs=12 --recursive --depth 1 --shallow-submodules --branch v5.2.2 https://github.com/espressif/esp-idf.git

Cloning into 'esp-idf'...
remote: Enumerating objects: 14951, done.
remote: Counting objects: 100% (14951/14951), done.
remote: Compressing objects: 100% (11693/11693), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 4121 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

gavin@gavin-ubuntu:~$

Spyker2000, 2024-06-25:
I am trying out  these commands
 
export GIT_TRACE_PACKET=1
export GIT_TRACE=1
export GIT_CURL_VERBOSE=1

Spyker2000, 2024-06-25:
gavin@gavin-ubuntu:~$ export GIT_TRACE_PACKET=1

gavin@gavin-ubuntu:~$ export GIT_TRACE=1

gavin@gavin-ubuntu:~$ export GIT_CURL_VERBOSE=1

gavin@gavin-ubuntu:~$ git clone --jobs=12 --recursive --depth 1 --shallow-submodules --branch v5.2.2 https://github.com/espressif/esp-idf.git
15:47:55.284898 git.c:463               trace: built-in: git clone --jobs=12 --recursive --depth 1 --shallow-submodules --branch v5.2.2 https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
15:47:55.335109 run-command.c:659       trace: run_command: git remote-https origin https://github.com/espressif/esp-idf.git
15:47:55.336497 git.c:749               trace: exec: git-remote-https origin https://github.com/espressif/esp-idf.git
15:47:55.336531 run-command.c:659       trace: run_command: git-remote-https origin https://github.com/espressif/esp-idf.git
15:47:55.340818 http.c:845              == Info: Couldn't find host github.com in the .netrc file; using defaults
15:47:55.376234 http.c:845              == Info: Host github.com:443 was resolved.
15:47:55.376256 http.c:845              == Info: IPv6: (none)
15:47:55.376259 http.c:845              == Info: IPv4: 20.87.245.0
15:47:55.376302 http.c:845              == Info:   Trying 20.87.245.0:443...
15:47:55.409629 http.c:845              == Info: Connected to github.com (20.87.245.0) port 443
15:47:55.430230 http.c:845              == Info: found 440 certificates in /etc/ssl/certs
15:47:55.430277 http.c:845              == Info: GnuTLS ciphers: NORMAL:-ARCFOUR-128:-CTYPE-ALL:+CTYPE-X509:-VERS-SSL3.0
15:47:55.430294 http.c:845              == Info: ALPN: curl offers h2,http/1.1
15:47:55.461082 http.c:845              == Info: SSL connection using TLS1.3 / ECDHE_RSA_AES_128_GCM_SHA256
15:47:55.462527 http.c:845              == Info:   server certificate verification OK
15:47:55.462534 http.c:845              == Info:   server certificate status verification SKIPPED
15:47:55.462581 http.c:845              == Info:   common name: github.com (matched)
15:47:55.462586 http.c:845              == Info:   server certificate expiration date OK
15:47:55.462589 http.c:845              == Info:   server certificate activation date OK
15:47:55.462595 http.c:845              == Info:   certificate public key: EC/ECDSA
15:47:55.462600 http.c:845              == Info:   certificate version: #3
15:47:55.462606 http.c:845              == Info:   subject: CN=github.com
15:47:55.462612 http.c:845              == Info:   start date: Thu, 07 Mar 2024 00:00:00 GMT
15:47:55.462615 http.c:845              == Info:   expire date: Fri, 07 Mar 2025 23:59:59 GMT
15:47:55.462627 http.c:845              == Info:   issuer: C=GB,ST=Greater Manchester,L=Salford,O=Sectigo Limited,CN=Sectigo ECC Domain Validation Secure Server CA
15:47:55.462635 http.c:845              == Info: ALPN: server accepted h2
15:47:55.462691 http.c:845              == Info: using HTTP/2
15:47:55.462722 http.c:845              == Info: [HTTP/2] [1] OPENED stream for https://github.com/espressif/esp-idf.git/info/refs?service=git-upload-pack
15:47:55.462727 http.c:845              == Info: [HTTP/2] [1] [:method: GET]
15:47:55.462731 http.c:845              == Info: [HTTP/2] [1] [:scheme: https]
15:47:55.462735 http.c:845              == Info: [HTTP/2] [1] [:authority: github.com]
15:47:55.462741 http.c:845              == Info: [HTTP/2] [1] [:path: /espressif/esp-idf.git/info/refs?service=git-upload-pack]
15:47:55.462744 http.c:845              == Info: [HTTP/2] [1] [user-agent: git/2.43.0]
15:47:55.462747 http.c:845              == Info: [HTTP/2] [1] [accept: */*]
15:47:55.462749 http.c:845              == Info: [HTTP/2] [1] [accept-encoding: deflate, gzip, br, zstd]
15:47:55.462753 http.c:845              == Info: [HTTP/2] [1] [accept-language: en-ZA, en;q=0.9, *;q=0.8]
15:47:55.462757 http.c:845              == Info: [HTTP/2] [1] [pragma: no-cache]
15:47:55.462761 http.c:845              == Info: [HTTP/2] [1] [git-protocol: version=2]
15:47:55.462800 http.c:792              => Send header, 0000000254 bytes (0x000000fe)
15:47:55.462806 http.c:804              => Send header: GET /espressif/esp-idf.git/info/refs?service=git-upload-pack HTTP/2
15:47:55.462809 http.c:804              => Send header: Host: github.com
15:47:55.462812 http.c:804              => Send header: User-Agent: git/2.43.0
15:47:55.462814 http.c:804              => Send header: Accept: */*
15:47:55.462818 http.c:804              => Send header: Accept-Encoding: deflate, gzip, br, zstd
15:47:55.462822 http.c:804              => Send header: Accept-Language: en-ZA, en;q=0.9, *;q=0.8
15:47:55.462826 http.c:804              => Send header: Pragma: no-cache
15:47:55.462828 http.c:804              => Send header: Git-Protocol: version=2
15:47:55.462831 http.c:804              => Send header:
15:47:55.753466 http.c:792              <= Recv header, 0000000013 bytes (0x0000000d)
15:47:55.753486 http.c:804              <= Recv header: HTTP/2 200
15:47:55.753493 http.c:792              <= Recv header, 0000000026 bytes (0x0000001a)
15:47:55.753497 http.c:804              <= Recv header: server: GitHub-Babel/3.0
15:47:55.753502 http.c:792              <= Recv header, 0000000059 bytes (0x0000003b)
15:47:55.753506 http.c:804              <= Recv header: content-type: application/x-git-upload-pack-advertisement
15:47:55.753510 http.c:792              <= Recv header, 0000000054 bytes (0x00000036)
[... cut ...]
15:50:51.122584 http.c:845              == Info: HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
15:50:51.122615 http.c:845              == Info: Connection #0 to host github.com left intact
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 3924 bytes of body are still expected
15:50:51.122688 pkt-line.c:85           packet:          git> 0002
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

Spyker2000, 2024-06-25:
This is the first time I have had a git cloning issue. Can I clone individual parts? It looks like the tcp session times out before the files are fully compressed. But looking at the time stamps it is less than 10 seconds from start to finish. 15:47:55 to 15:47:59

Spyker2000, 2024-06-25:
I am in a rural area on a farm. I don't have cell phone reception on the farm but I have a 2.5Mbps WIFI connection. So my Wifi Router connects to a dish via a Cat 5 cable and then go to a WIFI tower. I can speak to them and see if they can shape the traffic.

In the mean while, can I not wget or ftp the files from github? I have also had a look at your previous comment and will persue that option in the mean. I will keep you informed.

Vollbrecht, 2024-06-25:
You can use a custom provided esp-idf installation, you have to than yourself match the version you provide in that path with the version you set in the ESP_IDF_VERSION env variable set in .cargo/config.toml. For more information please read [this](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#idf_path-idf_path-native-builder-only). 

Please keep in mind that we also install the tools for you like the riscv_gcc compiler / cmake / ninja and setup the python env. So essentially what esp-idf needs if you don't provide a fully activated install. Though i think that is not an issue, since our problem is pretty specific to the git clone here. 

So try to get the esp-idf version 5.2.2 cloned some other way, and make sure all submodules are in that clonse. Than use the mention IDF_PATH env to point to your provided dir.

Spyker2000, 2024-06-25:
Thanks Frederick.  I will read up and see what transpires

Spyker2000, 2024-06-26:
Good news Frederick. I did my first RUST ESP-IDF clone. I spoke to my ISP and temporarily upgraded my WIFI throughput from 3Mbps to 4Mbps with a burst of 4.9Mpbs and successfully ran the `cargo build` with this message
`Finished dev [optimized + debuginfo] target(s) in 32m 30s`

The .embuild directory is fully loaded and I did a successful **cargo run** with "Hello World"

I also setup a  **wifi hot spot** on my cell phone and connecting my PC through it. How ever, the throughput was terrible. I go through a cellphone booster with boosts the cell phone signal but not the data signal.

The last thing I would like to try is the custom config to see if that do a` git clone` without issues. If this does not work I will have to upgrade my ISP to the 4Mpbs package.

Vollbrecht, 2024-06-26:
Good to hear that its working for you now. One thing that you defiantly should do, considering your internet connection, is to switch from a per project based install to a global install that is auto managed, in the case where you don't want to provide your own IDF_PATH. 

That way multiple projects can share an install. The only thing you have to do is to set the ESP_IDF_TOOLS_INSTALL_DIR="global" described [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir) in your project .cargo/config,toml under the [env] section. 

That way you don't pollute your env variables but can use a global managed install.

Closing this issue for now, if you encounter another issue feel free to create a new one. Also note that you can always come into the matrix chat, there you also can find help.

closed 2024-06-26 by Vollbrecht