esp-rs/esp-idf-svc #439: sd mmc init faild

niuhuan, 2024-06-24:
I am truly certain that my hardware is not faulty, and I can use the SD card normally with other firmware.

I have checked the correctness of the PINs, but still cannot read the tfcard

[image]

```rust

    use esp_idf_svc::{
        fs::{Fat, FatConfiguration},
        hal::{
            gpio::{self, AnyIOPin},
            prelude::*,
        },
        sd::{host::SdHost, mmc::SlotConfiguration, SdConfiguration},
    };
    use std::{fs::File, io::Write};

    let slot = SlotConfiguration::new(
        pins.gpio16,
        pins.gpio12,
        pins.gpio14,
        Some(pins.gpio17),
        Some(pins.gpio21),
        Some(pins.gpio18),
        Option::<gpio::Gpio16>::None,
        Option::<gpio::Gpio17>::None,
        Option::<gpio::Gpio15>::None,
        Option::<gpio::Gpio18>::None,
        Option::<AnyIOPin>::None,
        Option::<AnyIOPin>::None,
    );

    log::info!("INIT MMC");

    let host_config = SdConfiguration::new();

    let host = SdHost::new_with_mmc(&host_config, slot);

    let fat_config = FatConfiguration::new();

    let _fat = Fat::mount(fat_config, host, "/")?;

    let mut  read = fs::read_dir("/")?;
    while let Some(entry) = read.next() {
        let entry = entry?;
        let path = entry.path();
        log::info!("File: {:?}", path);
    }

    let mut file = File::create("/test.txt")?;

    file.write_all(b"Hello, world!")?;

```

```
I (434) cpu_start: cpu freq: 160000000 Hz
I (435) cpu_start: Application information:
I (437) cpu_start: Project name:     libespidf
I (443) cpu_start: App version:      1
I (447) cpu_start: Compile time:     Jun 23 2024 21:51:15
I (453) cpu_start: ELF file SHA256:  0000000000000000...
I (459) cpu_start: ESP-IDF:          v5.1.3
I (464) cpu_start: Min chip rev:     v0.0
I (469) cpu_start: Max chip rev:     v0.99
I (473) cpu_start: Chip rev:         v0.2
I (478) heap_init: Initializing. RAM available for dynamic allocation:
I (485) heap_init: At 3FC9F3F8 len 0004A318 (296 KiB): DRAM
I (492) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (498) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (504) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (512) spi_flash: detected chip: winbond
I (515) spi_flash: flash io: dio
W (520) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (528) sleep: Configure to isolate all GPIO pins in sleep state
I (535) sleep: Enable automatic switching of GPIO sleep configuration
I (542) app_start: Starting scheduler on CPU0
I (547) app_start: Starting scheduler on CPU1
I (547) main_task: Started on CPU0
I (557) main_task: Calling app_main()
I (567) myesp: MCU STARTED!
I (587) myesp::mmc_controller: INIT MMC
I (587) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (587) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (597) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (607) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (617) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (627) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (657) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (657) esp_idf_svc::nvs: NvsDefault dropped
I (657) esp_idf_svc::eventloop: System event loop dropped
Error: ESP_ERR_TIMEOUT
I (657) main_task: Returned from app_main()

```

niuhuan, 2024-06-24:
If I want to use SPI to read tfcard, what are the relationships between these pins?

   `CMD CLK DATA0 DATA1 DATA2 DATA3`  x  `sclk sdo sdi cs  cd int`

niuhuan, 2024-06-24:
this code has been panic at `let sd_device =`
```

    let device = esp_idf_svc::hal::spi::SpiDriver::new(
            spi3,
            pins.gpio12,
            pins.gpio14,
            Some(pins.gpio16),
            &esp_idf_svc::hal::spi::config::DriverConfig::default(),
    )?;

    let sd_device = esp_idf_svc::sd::spi::SpiDevice::new(
            device,
            pins.gpio18,
            Option::<esp_idf_svc::hal::gpio::AnyInputPin>::None,
            Option::<esp_idf_svc::hal::gpio::AnyInputPin>::None,
            Option::<esp_idf_svc::hal::gpio::AnyInputPin>::None,
    );

```

```
I (570) myesp::mmc_controller: INIT MMC
I (570) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (570) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (580) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
thread 'main' panicked at D:\Developments\Runtimes\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-idf-hal-0.44.0\src\spi.rs:482:52:
called `Result::unwrap()` on an `Err` value: ESP_ERR_INVALID_STATE (error code 259)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x4202e986 on core 0
0x4202e986 - panic_abort::__rust_start_panic::abort
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:48

Backtrace: 0x40376cb2:0x3fca2ba0 0x4037b96d:0x3fca2bc0 0x40382182:0x3fca2be0 0x4202e986:0x3fca2c50 0x4202e97a:0x3fca2c70 0x4201df16:0x3fca2c90 0x4201def2:0x3fca2ce0 0x4201ab10:0x3fca2d50 0x4201aa04:0x3fca2d80 0x4201dc9f:0x3fca2d
a0 0x4203ae9c:0x3fca2dd0 0x4203abf7:0x3fca2e10 0x4200e58e:0x3fca2e70 0x420079c0:0x3fca2ea0 0x42006833:0x3fca2ec0 0x42006e62:0x3fca2f50 0x420ae9bf:0x3fca31a0 0x42007074:0x3fca31c0 0x42010026:0x3fca31e0 0x42007064:0x3fca3210 0x42007047:0x3fca3240 0x4200eb4b:0x3fca3260 0x420b44fb:0x3fca3280
0x40376cb2 - panic_abort
    at D:\Developments\Projects\myesp\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-38574e7328fc2ed0\out\build\D:/Developments/Projects/myesp/.embuild/espressif/esp-idf/v5.1.3/components/esp_system\panic.c:472
0x4037b96d - esp_system_abort
    at D:\Developments\Projects\myesp\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-38574e7328fc2ed0\out\build\D:/Developments/Projects/myesp/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/port\esp_system_chip.c:84
0x40382182 - abort
    at D:\Developments\Projects\myesp\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-38574e7328fc2ed0\out\build\D:/Developments/Projects/myesp/.embuild/espressif/esp-idf/v5.1.3/components/newlib\abort.c:38
0x4202e986 - panic_abort::__rust_start_panic::abort
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:48
0x4202e97a - __rust_start_panic
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:43
0x4201df16 - rust_panic
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:833
0x4201def2 - std::panicking::rust_panic_with_hook
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:803
0x4201ab10 - std::panicking::begin_panic_handler::{{closure}}
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:659
0x4201aa04 - std::sys_common::backtrace::__rust_end_short_backtrace
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\sys_common\backtrace.rs:171
0x4201dc9f - rust_begin_unwind
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:647
0x4203ae9c - core::panicking::panic_fmt
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\panicking.rs:72
0x4203abf7 - core::result::unwrap_failed
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1649
0x4200e58e - core::result::Result<T,E>::unwrap
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1073
0x420079c0 - core::ptr::drop_in_place<esp_idf_hal::spi::SpiDriver>
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\ptr\mod.rs:507
0x42006833 - myesp::mmc_controller::mmc_main
    at D:\Developments\Projects\myesp\src\mmc_controller.rs:130
0x42006e62 - myesp::main
    at D:\Developments\Projects\myesp\src\main.rs:34
0x420ae9bf - core::ops::function::FnOnce::call_once
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\ops\function.rs:250
0x42007074 - std::rt::lang_start::{{closure}}
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\rt.rs:166
0x42010026 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\ops\function.rs:284
0x42007064 - std::rt::lang_start
    at D:\Developments\Runtimes\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\rt.rs:165
0x42007047 - main
    at ??:??
0x4200eb4b - app_main
    at D:\Developments\Runtimes\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-idf-sys-0.35.0\src\start.rs:46
0x420b44fb - main_task
[... cut ...] if/esp-idf/v5.1.3/components/freertos\app_startup.c:208        

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403765d6
0x403765d6 - esp_restart_noos_dig
    at D:\Developments\Projects\myesp\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-38574e7328fc2ed0\out\build\D:/Developments/Projects/myesp/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/port\esp_system_chip.c:57
```

niuhuan, 2024-06-24:
panic on this unwrap
```
impl<'d> Drop for SpiDriver<'d> {
    fn drop(&mut self) {
        esp!(unsafe { spi_bus_free(self.host()) }).unwrap();
    }
}
```

AlixANNERAUD, 2024-07-09:
Hello,
Sorry for the delay, I am currently traveling.
I just made a fix for the SD SPI part, I invite you to try with [my branch](https://github.com/AlixANNERAUD/esp-idf-svc/tree/sd-card-fix) that fixes the encountered bugs. Here is [the hardware configuration I used](https://wokwi.com/projects/323656763409695316).

matt-agius, 2024-07-10:
I was having the same issue. Your branch fixed it.

Using:
- esp32s3
- SD SPI Device: ATOM TF-CARD

Vollbrecht, 2024-07-28:
closing since esp-rs/esp-idf-hal#457 landed. If you still got problems please report back.

closed 2024-07-28 by Vollbrecht