esp-rs/esp-idf-svc #441: mqtt demo can't connect

muei, 2024-06-27:
Attempted multiple mqtt servers, but were unable to connect，logs below

```
I (4712) s3: Wifi netif up
I (4712) s3: Wifi created
I (4712) s3: MQTT client created
I (4722) s3: About to start the MQTT client
I (4722) s3: MQTT Listening for messages
I (4732) s3: [Queue] Event: BeforeConnect
E (4732) mqtt_client: Client has not connected
```

kgv, 2024-07-04:
The same problem.
After update to `esp-idf-svc` version 0.49.0, it's been working fine with 0.48.

<details>

```sh
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1714
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (32) boot: ESP-IDF v5.1.2-342-gbcf1645e44 2nd stage bootloader
I (32) boot: compile time Dec 12 2023 10:50:58
I (32) boot: chip revision: v0.3
I (36) boot.esp32c3: SPI Speed      : 40MHz
I (41) boot.esp32c3: SPI Mode       : DIO
I (46) boot.esp32c3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 003f0000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=3ed38h (257336) map
I (159) esp_image: segment 1: paddr=0004ed60 vaddr=3fc90c00 size=012b8h (  4792) load
I (160) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=de9dch (911836) map
I (366) esp_image: segment 3: paddr=0012ea04 vaddr=3fc91eb8 size=01758h (  5976) load
I (368) esp_image: segment 4: paddr=00130164 vaddr=40380000 size=10be4h ( 68580) load
I (395) boot: Loaded app from partition at offset 0x10000
I (395) boot: Disabling RNG early entropy source...
I (406) cpu_start: Unicore app
I (415) cpu_start: Pro cpu start user code
I (415) cpu_start: cpu freq: 160000000 Hz
I (415) cpu_start: Application information:
I (418) cpu_start: Project name:     libespidf
I (423) cpu_start: App version:      e3bf697-dirty
I (429) cpu_start: Compile time:     Jul  4 2024 18:22:41
I (435) cpu_start: ELF file SHA256:  000000000...
I (440) cpu_start: ESP-IDF:          v5.2.2
I (445) cpu_start: Min chip rev:     v0.3
I (450) cpu_start: Max chip rev:     v1.99 
I (455) cpu_start: Chip rev:         v0.3
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (467) heap_init: At 3FC97D50 len 000282B0 (160 KiB): RAM
I (473) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (480) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (487) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (494) spi_flash: detected chip: generic
I (498) spi_flash: flash io: dio
W (502) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (510) sleep: Configure to isolate all GPIO pins in sleep state
I (517) sleep: Enable automatic switching of GPIO sleep configuration
I (524) main_task: Started on CPU0
I (524) main_task: Calling app_main()
I (534) pp: pp rom version: 9387209
I (534) net80211: net80211 rom version: 9387209
I (544) wifi:wifi driver task: 3fca0770, prio:23, stack:6656, core=0
I (544) wifi:wifi firmware version: 3e0076f
I (554) wifi:wifi certification version: v7.0
I (554) wifi:config NVS flash: enabled
I (554) wifi:config nano formating: disabled
I (554) wifi:Init data frame dynamic rx buffer num: 32
I (564) wifi:Init static rx mgmt buffer num: 10
I (564) wifi:Init management short buffer num: 32
I (574) wifi:Init dynamic tx buffer num: 32
[... 226 lines cut ...]
    at ??:??
0x3c0e1244 - $d.3
    at ??:??
0x42020130 - core::result::Result<T,E>::map_err
    at /home/.../.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/result.rs:852
3fc9bd90: 0x3fc9bdc4 0x00000000 0x00000000 0x3c0e13d0 0x00000000 0x00000000 0x3c0fe000 0x420022ec
0x3c0e13d0 - $d.345
    at ??:??
0x420022ec - std::rt::lang_start
    at ??:??
3fc9bdb0: 0x3c0fe5c4 0x0000020c 0x4200382c 0x420ddc66 0x00000000 0x00001388 0x00000001 0x00000000
0x4200382c - exec::main
    at /home/.../temp/src/main.rs:26
0x420ddc66 - main_task
    at /home/.../temp/.embuild/espressif/esp-idf/v5.2.2/components/freertos/app_startup.c:209
3fc9bdd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bdf0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x3fc9b880

ELF file SHA256: 000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
```

</details>

ivmarkov, 2024-07-08:
This has fortunately nothing to do with new vs old `esp-idf-svc` version and is actually caused by the speed (and number of re-connect attempts) with which the MQTT **examples** you are trying happen to work in your concrete network.

In the MQTT blocking and async examples, there was a race condition between the client trying to connect, and our code trying to call `client.subscribe(...)?`. 

The thing is, if the client fails to connect fast enough (or tries to connect, fails, and then re-tries after a while), we may end up calling `client.subscribe(...)?` when the client is not (yet) connected, and as a result, the `subscribe` call would fail, and the whole example would fail.

I updated the examples to just call `subscribe` in a loop and re-try on unsuccessful subscribe every 0.5s, so it should work for you now (I was able to reproduce your problem with a c3 and ESP IDF 5.2.2; couldn't reproduce it with an xtensa esp32 and ESP-IDF 5.1.2). 

Here's my log, and how it should hopefully look the same for you with the updated examples:
```
I (3323) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (3333) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3343) mqtt_client: Wifi connected
I (3343) wifi:<ba-add>idx:0 (ifx:0, c4:ad:34:30:cf:a9), tid:0, ssn:3, winSize:64
I (4343) esp_netif_handlers: sta ip: 192.168.30.150, mask: 255.255.255.0, gw: 192.168.30.1
I (4343) mqtt_client: Wifi netif up
I (4353) mqtt_client: About to start the MQTT client
I (4363) mqtt_client: MQTT Listening for messages
I (4363) mqtt_client: [Queue] Event: BeforeConnect
E (4373) mqtt_client: Client has not connected
E (4373) mqtt_client: Failed to subscribe to topic "esp-mqtt-demo": ESP_FAIL, retrying...
E (4873) mqtt_client: Client has not connected
E (4873) mqtt_client: Failed to subscribe to topic "esp-mqtt-demo": ESP_FAIL, retrying...
I (4963) mqtt_client: [Queue] Event: Connected(session: false)
I (5373) mqtt_client: Subscribed to topic "esp-mqtt-demo"
I (5573) mqtt_client: [Queue] Event: Subscribed(12331)
I (5873) mqtt_client: Published "Hello from esp-mqtt-demo!" to topic "esp-mqtt-demo"
```

The `E (4373) mqtt_client: Failed to subscribe to topic "esp-mqtt-demo": ESP_FAIL, retrying...` is key. It used to fail the whole example before. Now it would just re-try until it (hopefully) succeeds.

closed 2024-07-08 by ivmarkov