esp-rs/esp-idf-svc #442: Missing BT related fields when trying to compile this against esp-idf v5.2.2

oddlama, 2024-07-03:
I'm trying to compile this project (https://github.com/ivmarkov/esp-idf-matter) as-is,
but - no matter what I do - it fails to compile esp-idf-svc complaining about some bluetooth-related defines that are apparently missing.

```
   Compiling esp-idf-svc v0.49.0
error[E0425]: cannot find value `ESP_BT_CTRL_CONFIG_MAGIC_VAL` in module `crate::sys`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:497:32
    |
497 |             magic: crate::sys::ESP_BT_CTRL_CONFIG_MAGIC_VAL,
    |                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `crate::sys`
```

(full error log below)

Is there anything obvious I am missing? I am compiling for `MCU=esp32c6` by using `cargo build --target riscv32imac-esp-espidf` and made sure to use a fresh clone of esp-idf v5.2.2 where esp-idf-sys compiles just fine. The sdkconfig.defaults of the project contains the following:

```
# Increase a bit these stack sizes as they are too low by default
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096
# For async-io
#CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=8192

# Go figure...
CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=4096

# BT
CONFIG_BT_BTC_TASK_STACK_SIZE=15000
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
CONFIG_BT_CLASSIC_ENABLED=n
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BT_A2DP_ENABLE=n
CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y
CONFIG_BT_BLE_50_FEATURES_SUPPORTED=n
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
```

<details><summary>cargo build output</summary>

```
   Compiling esp-idf-svc v0.49.0
error[E0425]: cannot find value `ESP_BT_CTRL_CONFIG_MAGIC_VAL` in module `crate::sys`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:497:32
    |
497 |             magic: crate::sys::ESP_BT_CTRL_CONFIG_MAGIC_VAL,
    |                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `crate::sys`

error[E0425]: cannot find value `ESP_BT_CTRL_CONFIG_VERSION` in module `crate::sys`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:498:34
    |
498 |             version: crate::sys::ESP_BT_CTRL_CONFIG_VERSION,
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `crate::sys`

error[E0425]: cannot find value `CONFIG_BT_CTRL_PINNED_TO_CORE` in module `crate::sys`
    --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:501:50
     |
501  |             controller_task_run_cpu: crate::sys::CONFIG_BT_CTRL_PINNED_TO_CORE as _,
     |                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a constant with a similar name exists: `CONFIG_BLUEDROID_PINNED_TO_CORE`
     |
    ::: /home/nixuser/projects/embedded/esp-idf-matter/target/riscv32imac-esp-espidf/debug/build/esp-idf-sys-c8d14348dff4d490/out/bindings.rs:1167:1
     |
1167 | pub const CONFIG_BLUEDROID_PINNED_TO_CORE: u32 = 0;
     | ---------------------------------------------- similarly named constant `CONFIG_BLUEDROID_PINNED_TO_CORE` defined here

error[E0425]: cannot find value `CONFIG_BT_CTRL_MODE_EFF` in module `crate::sys`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:502:41
    |
502 |             bluetooth_mode: crate::sys::CONFIG_BT_CTRL_MODE_EFF as _,
[... 380 lines cut ...]
    |
    = note: available fields are: `config_version`, `ble_ll_resolv_list_size`, `ble_hci_evt_hi_buf_count`, `ble_hci_evt_lo_buf_count`, `ble_ll_sync_list_cnt` ... and 46 others

error[E0560]: struct `esp_idf_hal::sys::esp_bt_controller_config_t` has no field named `dup_list_refresh_period`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:529:13
    |
529 |             dup_list_refresh_period: crate::sys::DUPL_SCAN_CACHE_REFRESH_PERIOD as _,
    |             ^^^^^^^^^^^^^^^^^^^^^^^ `esp_idf_hal::sys::esp_bt_controller_config_t` does not have this field
    |
    = note: available fields are: `config_version`, `ble_ll_resolv_list_size`, `ble_hci_evt_hi_buf_count`, `ble_hci_evt_lo_buf_count`, `ble_ll_sync_list_cnt` ... and 46 others

error[E0560]: struct `esp_idf_hal::sys::esp_bt_controller_config_t` has no field named `scan_backoff_upperlimitmax`
   --> /home/nixuser/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.0/src/bt.rs:530:13
    |
530 |             scan_backoff_upperlimitmax: crate::sys::BT_CTRL_SCAN_BACKOFF_UPPERLIMITMAX as _,
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^ `esp_idf_hal::sys::esp_bt_controller_config_t` does not have this field
    |
    = note: available fields are: `config_version`, `ble_ll_resolv_list_size`, `ble_hci_evt_hi_buf_count`, `ble_hci_evt_lo_buf_count`, `ble_ll_sync_list_cnt` ... and 46 others

Some errors have detailed explanations: E0425, E0560.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-idf-svc` (lib) due to 58 previous errors
```

</details>

ivmarkov, 2024-07-03:
1. Does the compile work when using esp idf `v5.2` and just changing the target to `riscv32imac-esp-espidf`?
2. Can you re-open the issue in the `esp-idf-matter` repository please?

Vollbrecht, 2024-07-03:
@ivmarkov you can transfer the issue with the transfer-issue button on the right

oddlama, 2024-07-03:
> 1. Does the compile work when using esp idf `v5.2` and just changing the target to `riscv32imac-esp-espidf`?

Unfortunately not, it leads to exactly the same error. 

>    2. Can you re-open the issue in the `esp-idf-matter` repository please?

Of course, it was hard to tell what crate is the true cause of the issue, so since esp-idf-svc failed first I reported it here initially. It may also very well be that I still have a wrong environment setup.

oddlama, 2024-07-03:
Closing this and re-opening here: https://github.com/ivmarkov/esp-idf-matter/issues/5

closed 2024-07-03 by oddlama