esp-rs/esp-idf-svc #443: How do I get  the examples to compile?

henkoch, 2024-07-05:
Hi

How do I get  the examples to compile?

runing:
`MCU=esp32c3 cargo espflash flash --target riscv32imc-esp-espidf --example wifi --monitor`

fails with error[E0463]: can't find crate for `core`

I can see `examples/wifi.rs`has `use core::convert::TryInto;`

But I have no idea of where I can get the core crate from?
Any idea of where I can find a description of what I need to do to get this to work.

I can get the example to compile that is described in the youtube video [Rust on ESP32 - Adding WiFi](https://www.youtube.com/watch?v=tx_ynPOQF4w)

- source: git cloned main of the repo
- rust: rustc 1.78.0 (9b00956e5 2024-04-29)
- did call: . $HOME/export-esp.sh
- OS: Ubuntu 22.04.4 LTS
- I had previously run the Prerequisites of [Rust on ESP-IDF "Hello, World" template](https://github.com/esp-rs/esp-idf-template#prerequisites)

```text
[2024-07-05T13:33:18Z INFO ] Serial port: '/dev/ttyACM0'
[2024-07-05T13:33:18Z INFO ] Connecting...
[2024-07-05T13:33:19Z INFO ] Using flash stub
   Compiling pin-project-lite v0.2.14
   Compiling semver v1.0.23
   Compiling serde_json v1.0.120
   Compiling which v4.4.2
   Compiling bindgen v0.69.4
   Compiling prettyplease v0.2.20
   Compiling regex v1.10.5
   Compiling anyhow v1.0.86
   Compiling rustc-hash v1.1.0
   Compiling cc v1.0.104
   Compiling shlex v1.3.0
   Compiling fastrand v2.1.0
   Compiling normpath v1.2.0
error: could not compile `pin-project-lite` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error[E0463]: can't find crate for `core`
  |
  = note: the `riscv32imc-esp-espidf` target may not be installed
  = help: consider downloading the target with `rustup target add riscv32imc-esp-espidf`

error: aborting due to 1 previous error

For more information about this error, try `rustc --explain E0463`.
```

ivmarkov, 2024-07-05:
1. Install `rustup` and de-install any pre-packaged Rust compiler that might be coming with your distro
2. `rustup default nightly`

ivmarkov, 2024-07-05:
... or if you have `espup` installed: `rustup default esp`.

closed 2024-07-05 by henkoch