esp-rs/esp-idf-svc #446: NVS clear on reset

jnthbdn, 2024-07-06:
Hello everyone!

I have a really strange problem with the NVS. I have created, via a csv file, a partition to store user configuration data.
On the whole, everything works fine, and I'm able to store my data without any problem (I'm checking this with the `espflash read-flash 0xe000 0x1000 dump_nvs` command). On the other hand, I have a very strange behavior. As soon as I reset the ESP, the NVS is empty...

This is the result from `cargo run`
```
 $ cargo run
   Compiling esp-wifi-soil-moisture v0.1.0 (/home/jonathan/Workspace-Perso/rust/ESP_Garden/Wifi/soil_moisture)
    Finished `dev` profile [optimized + debuginfo] target(s) in 1.04s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/esp-wifi-soil-moisture`
[2024-07-06T13:33:09Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-07-06T13:33:09Z INFO ] Connecting...
[2024-07-06T13:33:09Z INFO ] Using flash stub
[2024-07-06T13:33:10Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:b4:72:4e:ee:e4
Partition table:   ./custom_partitions.csv
App/part. size:    1,304,816/3,145,728 bytes, 41.48%
[2024-07-06T13:33:10Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-07-06T13:33:10Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:18] [========================================]     711/711     0x10000                                                                                                                                                                                                                                                                                                                                                                                              [2024-07-06T13:33:31Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1714
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (32) boot: ESP-IDF v5.1.2-342-gbcf1645e44 2nd stage bootloader
I (32) boot: compile time Dec 12 2023 10:50:58
I (32) boot: chip revision: v0.3
I (36) boot.esp32c3: SPI Speed      : 40MHz
I (41) boot.esp32c3: SPI Mode       : DIO
I (46) boot.esp32c3: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (74) boot:  1 config           WiFi data        01 02 0000e000 00001000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 factory          factory app      00 00 00010000 00300000
I (96) boot: End of partition table
[...]
```

Here is some links:
* The repository project: https://github.com/jnthbdn/esp32-garden
* The custom partitions file: https://github.com/jnthbdn/esp32-garden/blob/master/custom_partitions.csv
* The file used to save & read from NVS: https://github.com/jnthbdn/esp32-garden/blob/master/src/main_configuration.rs

I don't understand what's going on, and if I'm missing or misunderstanding something.

Thanks for your help,

P.S: Screen shot before/after

Before reset (after settings some value though a web page):
[image]

After reset:
[image]

jnthbdn, 2024-07-06:
I forgot to mention that I'm using an ESP32-C3FN4 (more precisely, AI-Thinker's ESP32-C3-13U: https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-13u-kit-v1.0_specification.pdf).

Vollbrecht, 2024-07-06:
That indeed sounds a bit strange ;D Just to clarify by "reseting" what do you exactly do here? Do you unplug and replug the esp? Or when you came back from deep_sleep ? Or when you just hit Ctrl + R when connected within the monitor instance?

Vollbrecht, 2024-07-06:
Can you check if you hit [this](https://github.com/esp-rs/esp-idf-svc/blob/master/src/nvs.rs#L102) codepath. If you got one of the two errors returned on init than it will earse the flash.

jnthbdn, 2024-07-06:
> That indeed sounds a bit strange ;D Just to clarify by "reseting" what do you exactly do here? Do you unplug and replug the esp? Or when you came back from deep_sleep ? Or when you just hit Ctrl + R when connected within the monitor instance?

By "resetting" I mean both unplug/replug or press the reset button.

> Can you check if you hit [this](https://github.com/esp-rs/esp-idf-svc/blob/master/src/nvs.rs#L102) codepath. If you got one of the two errors returned on init than it will earse the flash.

Try to make the debug works and and I check ;)

jnthbdn, 2024-07-06:
Well... I added an `info!` line before line 102, and I can confirm I hit this part of code... But why ?

jnthbdn, 2024-07-06:
OOOOKKKkkkkk....
I found... 0x1000 is the minimal size... but the recommemded size is 0x3000...
see: https://esp32.com/viewtopic.php?p=71743&sid=710ede16c6b82109e6fab8ad544532cc#p71743

> Hi,
>
> For data type partitions, the minimum size is 0x1000 (4kB), but we recommend to size the NVS partition with at least  0x3000 (12kB).
> 
> This is due the [Structure of a page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#structure-of-a-page) ([https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#structure-of-a-page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#structure-of-a-page)).

Thanks for your help...

closed 2024-07-06 by jnthbdn