esp-rs/esp-idf-svc #450: Ethernet: Add option to set poll_period_ms intead of interrupt pin

stabl-gjn, 2024-07-12:
Currently the `src/eth.rs` implementation forces the setting of a interupt pin for the spi ethernet drivers (e.g. <https://github.com/esp-rs/esp-idf-svc/blob/fb390905fe1a368ea9d150d6ea983137a17ee644/src/eth.rs#L399>).

Starting with version v5.1.4 of esp-idf the option to instead define a `poll_period_ms` was added: <https://github.com/espressif/esp-idf/blob/d7b0a45ddbddbac53afb4fc28168f9f9259dbb79/components/esp_eth/include/esp_eth_mac.h#L614-L620>.

With the current implementation we force the setting of an interrupt pin (the argument is not optional) and give no option to instead define a polling interval.

I have a working implementation of this and would be happy to contribute it, if this is a desired feature.

Vollbrecht, 2024-07-12:
Feel free to send a PR ! 

Since the module lags a bit of docstrings, if you also could  add a brief description to what the advantages/disadvanteges of the polling impl versus the interrupt pin driven are, that you observerd would be helpful for others. 

It would be nice if you can implement it with a second new constructor that doesn't need the pin such we don't break existing use of the Driver, and change the internal implementation accordingly.

stabl-gjn, 2024-07-12:
I have created this pr: #452

closed 2024-07-15 by ivmarkov