esp-rs/esp-idf-svc #467: Follow up on  sd mmc init faild #439

fabracht, 2024-08-04:
Hi,
I've been eagerly following the status of the SPI issue for SD card here: https://github.com/esp-rs/esp-idf-svc/issues/439

I see that it has been merged, but there has bee no release for the crate. Using it from the master branch instead of the crate repo causes an error because esp-idf-hal doesn't have an "experimental" feature flag (yet). My guess is that this will all be released soon, so, I just thought of kindly asking if there's any date for that yet.

Great work with the crates though, programming for embedded has never been this easy.
Thanks.

ivmarkov, 2024-08-04:
You just need to:
- use the `[patch.crates-io]` cargo functionality
- patch BOTH `esp-idf-svc` _and_ `esp-idf-hal` with their git repository, even if you don't depend on `edp-idf-hal`
... and then report if the new sd support works for you. I would be particularly interested in the sd-spi code path (ad opposed to sd-mmc) as I did not test it carefully.

ivmarkov, 2024-08-05:
> Although, if I try to return just the filesystem, (i assume) the spi device drivers are being dropped, and I cannot call `std::io::File` commands anymore. I cannot return `SdSpiHostDriver` along with `MountedFatFs` easily, because it is already passed when you call `mount()`.

No they are not dropped, and this ^^^ is a bug. Read on.

> The example implies I just need to keep `MountedFatfs::mount()` struct in memory to keep the filesystem mounted.

Yes, exactly.

> 2. Return `Ok(SdCard { fs: _mounted_fatfs })` from [sd_spi.rs](https://github.com/esp-rs/esp-idf-svc/blob/bf47d2bcee6e1bc71530a23264eae63a7af84aae/examples/sd_spi.rs#L49) and call `File::create()` in main or some other place once more with `SdCard` still in memory, and I get this error:

This is a bug. Fortunately, I think I just fixed it. Can you try the bugfix? You only need to do `cargo update` in the root of your binary crate, rebuild and then re-flash and re-run.

In general you are free to move and place the driver, the mounted fatfs etc. however you want, and the code should not crash at all. There is **no** such requirement that you need to "manually" keep the `SpiDeviceDriver` around, because you actually do:
- If you pass it by moving it into `FatFs` (as you actually do by just following the example) then `FatFs` "owns" it, and recursively, `MountedFatFs` owns `FatFs` and thus the driver too if you pass `FatFs` by moving it
- You can also use refcounting or raw Rust references, and it should also work, and Rust should warn you if you somehow have wrong lifetimes
-

ivmarkov, 2024-08-05:
Fix here: https://github.com/esp-rs/esp-idf-hal/commit/b186a3be4c68c634089260d472d59c9b70c2b061

ivmarkov, 2024-08-05:
> ```
> pub struct SdCard<'d> {
>     #[allow(dead_code)]
>     spi_device_driver: SpiDeviceDriver<'d, SpiDriver<'d>>,
>     fs: Fat,
> }
> ```

This ^^^ in the "original" PR was wrong and very dangerous, hence I removed the support for this pattern, in that the `fs` (`Fat`) variable actually depends on the lifetime of `spi_device_driver` - yet - this was not captured **at all** in the driver code. So if you drop `spi_device_driver` before `fs`, the program might crash, which is UB and not OK in Rust.

fabracht, 2024-08-05:
I've created a blank project using the template, and modified the Cargo.toml for:
```toml
[package]
name = "esps3-sd-spi"
version = "0.1.0"
authors = ["Fabrício Bracht"]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "esps3-sd-spi"
harness = false       # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = [
    "esp-idf-svc/embassy-sync",
    "esp-idf-svc/critical-section",
    "esp-idf-svc/embassy-time-driver",
]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git", features = [
    "experimental",
    "alloc",
] }

[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal.git" }

[build-dependencies]
embuild = "0.32.0"
```

But I'm getting an error building it:
```
...
   Compiling embassy-futures v0.1.1
   Compiling embedded-svc v0.28.0
error[E0308]: mismatched types
   --> /home/esp/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/b186a3b/src/sd.rs:223:38
    |
223 |         let mut card: sdmmc_card_t = alloc::boxed::Box::new(Default::default());
    |                       ------------   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `sdmmc_card_t`, found `Box<_>`
    |                       |
    |                       expected due to this
    |
    = note: expected struct `esp_idf_sys::sdmmc_card_t`
               found struct `Box<_>`
help: consider unboxing the value
    |
223 |         let mut card: sdmmc_card_t = *alloc::boxed::Box::new(Default::default());
    |                                      +

error[E0614]: type `esp_idf_sys::sdmmc_card_t` cannot be dereferenced
   --> /home/esp/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/b186a3b/src/sd.rs:225:60
    |
225 |         esp!(unsafe { sdmmc_card_init(&configuration, &mut *card) })?;
    |                                                            ^^^^^

error[E0308]: mismatched types
   --> /home/esp/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/b186a3b/src/sd.rs:227:32
    |
227 |         Ok(Self { _host: host, card })
    |                                ^^^^ expected `Box<sdmmc_card_t>`, found `sdmmc_card_t`
    |
    = note: expected struct `Box<esp_idf_sys::sdmmc_card_t>`
               found struct `esp_idf_sys::sdmmc_card_t`
    = note: for more on the distinction between the stack and the heap, read https://doc.rust-lang.org/book/ch15-01-box.html, https://doc.rust-lang.org/rust-by-example/std/box.html, and https://doc.rust-lang.org/std/boxed/index.html
help: store this in the heap by calling `Box::new`
    |
227 |         Ok(Self { _host: host, Box::new(card) })
    |                                +++++++++    +

error[E0308]: mismatched types
   --> /home/esp/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/b186a3b/src/sd.rs:301:32
    |
301 |         Ok(Self { _host: host, card })
    |                                ^^^^ expected `Box<sdmmc_card_t>`, found `sdmmc_card_t`
    |
    = note: expected struct `Box<esp_idf_sys::sdmmc_card_t>`
               found struct `esp_idf_sys::sdmmc_card_t`
    = note: for more on the distinction between the stack and the heap, read https://doc.rust-lang.org/book/ch15-01-box.html, https://doc.rust-lang.org/rust-by-example/std/box.html, and https://doc.rust-lang.org/std/boxed/index.html
help: store this in the heap by calling `Box::new`
    |
301 |         Ok(Self { _host: host, Box::new(card) })
    |                                +++++++++    +

Some errors have detailed explanations: E0308, E0614.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `esp-idf-hal` (lib) due to 4 previous errors
```

Am I missing something still?

ivmarkov, 2024-08-05:
> I've created a blank project using the template, and modified the Cargo.toml for:

Ah yes, sorry - my bad. I thought my changes build fine, but forgot that the CI of `esp-idf-hal` is not yet enabled for experimental features. Now fixed. Do a `cargo update` in the crate root of the blank project.

> [dependencies]
> log = { version = "0.4", default-features = false }
> esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git", features = [
>     "experimental",
>     "alloc",
> ] }

Why are you doing this? If you use the `[patch.crates-io]` feature, you are **not** supposed to touch the `[dependencies]` section at all. It goes with regular "crates.io" dependencies like `esp-idf-svc = { version = "..", features = [...] }`. That's the whole point of it.

> [patch.crates-io]
> esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git" }
> esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal.git" }

This is _all_ you need, as I explained.

fabracht, 2024-08-05:
You are correct. Those are remains of multiple tries to fix the issue. They got left behind.

fabracht, 2024-08-05:
You said:

```
If you use the [patch.crates-io] feature, you are not supposed to touch the [dependencies] section at all. It goes with regular "crates.io" dependencies like esp-idf-svc = { version = "..", features = [...] }. That's the whole point of it.
```

Not sure I understand what you meant by "not supposed to touch". Can you explain please?

ivmarkov, 2024-08-05:
> You said:
>
> ```
> If you use the [patch.crates-io] feature, you are not supposed to touch the [dependencies] section at all. It goes with regular "crates.io" dependencies like esp-idf-svc = { version = [...]

By not "touching" I meant not _blindly_ changing regular crates.io dependencies with GIT ones. Use cargo dependencies as per my example from above: `esp-idf-svc = { version = "..", features = [...] }`.

The `[patch.crates-io]` section will _automatically_ turn them / "patch" them / "replace" them with the GIT ones. Which is why it exists as a feature **and why it is named like that** - so that you don't manually change your crates.io dependencies to GIT ones, _because manual changing of crates.io deps to GIT deps does not always work_. 

_And even in your case it does not really work_, because **inside** `esp-idf-svc` there is a dependency to the crates.io version of `esp-idf-hal`. You can't change this manually to a GIT dependency yourself, as the `Cargo.toml` of `esp-idf-svc` is not part of your source code. But a `[patch.crates-io]` section that "patches" `esp-idf-hal` WILL patch it also for the Cargo.toml of `esp-idf-svc`, which is the whole point. I think it might be useful if you read the `[patch.crates-io]` section in the Cargo documentation, and also if you read a bit on `cargo tree` so that you understand how dependencies actually work.

fabracht, 2024-08-05:
Will do. Thank you.

fabracht, 2024-08-05:
Just to leave it documented. This should work right off the bat, correct?
```
[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.49", default-features = false }

[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal.git" }
```

ivmarkov, 2024-08-05:
> Just to leave it documented. This should work right off the bat, correct?
> 
> ```
> [package]
> name = "esps3-sd-spi"
> version = "0.1.0"
> authors = ["Fabrício Bracht"]
> edition = "2021"
> resolver = "2"
> rust-version = "1.77"
> 
> [[bin]]
> name = "esps3-sd-spi"
> harness = false       # do not use the built in cargo test harness -> resolve rust-analyzer errors
> 
> [profile.release]
> opt-level = "s"
> 
> [profile.dev]
> debug = true    # Symbols are nice and they don't increase the size on Flash
> opt-level = "z"
> 
> [features]
> default = ["std", "embassy", "esp-idf-svc/native"]
> 
> pio = ["esp-idf-svc/pio"]
> std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
> alloc = ["esp-idf-svc/alloc"]
> nightly = ["esp-idf-svc/nightly"]
> experimental = ["esp-idf-svc/experimental"]
> embassy = [
>     "esp-idf-svc/embassy-sync",
>     "esp-idf-svc/critical-section",
>     "esp-idf-svc/embassy-time-driver",
> ]
> 
> [dependencies]
> log = { version = "0.4", default-features = false }
> esp-idf-svc = { version = "0.49", default-features = false }
> 
> [patch.crates-io]
> esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc.git" }
> esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal.git" }
> 
> [build-dependencies]
> embuild = "0.32.0"
> ```

Yes it should.

fabracht, 2024-08-05:
Hmm, obviously, I'm still missing something. A cargo build resulted in:
```
   Compiling esp-idf-hal v0.44.1 (https://github.com/esp-rs/esp-idf-hal.git#72f6448c)
   Compiling esp-idf-svc v0.49.1 (https://github.com/esp-rs/esp-idf-svc.git#bf47d2bc)
error[E0432]: unresolved import `esp_idf_svc::hal::sd`
 --> src/main.rs:7:9
  |
7 |         sd::{SdCardConfiguration, SdCardDriver},
  |         ^^ could not find `sd` in `hal`

error[E0433]: failed to resolve: use of undeclared type `SdSpiHostDriver`
  --> src/main.rs:33:9
   |
33 |         SdSpiHostDriver::new(
   |         ^^^^^^^^^^^^^^^ use of undeclared type `SdSpiHostDriver`

error[E0433]: failed to resolve: use of undeclared type `MountedFatfs`
  --> src/main.rs:45:26
   |
45 |     let _mounted_fatfs = MountedFatfs::mount(Fatfs::new_sdcard(0, sd_card_driver)?, "/sdcard", 4)?;
   |                          ^^^^^^^^^^^^ use of undeclared type `MountedFatfs`

error[E0433]: failed to resolve: use of undeclared type `Fatfs`
  --> src/main.rs:45:46
   |
45 |     let _mounted_fatfs = MountedFatfs::mount(Fatfs::new_sdcard(0, sd_card_driver)?, "/sdcard", 4)?;
   |                                              ^^^^^ use of undeclared type `Fatfs`

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `esps3-sd-spi` (bin "esps3-sd-spi") due to 4 previous errors
```

ivmarkov, 2024-08-05:
Put the `experimental` feature in the `defaults`...

fabracht, 2024-08-05:
Ah, thank you very much.

fabracht, 2024-08-06:
Yes. I'll let you know.
Right now I'm still struggling to make it work. But I'm sure it has more to do with me not setting things up properly and not the code.
Now I'm struggling with:
```
I (388) sleep: Configure to isolate all GPIO pins in sleep state
I (395) sleep: Enable automatic switching of GPIO sleep configuration
I (402) main_task: Started on CPU0
I (406) main_task: Calling app_main()
I (412) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (441) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (442) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (449) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (458) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Error: ESP_ERR_TIMEOUT (error code 263)
I (471) main_task: Returned from app_main()
```

From the example, I changed the pins to:
```
    let spi_driver = SpiDriver::new(
        peripherals.spi2,
        pins.gpio12,       // SCLK
        pins.gpio13,       // MISO
        Some(pins.gpio11), // MOSI
        &DriverConfig::default().dma(Dma::Auto(4096)),
    )?;

    let sd_card_driver = SdCardDriver::new_spi(
        SdSpiHostDriver::new(
            spi_driver,
            Some(pins.gpio10),
            AnyIOPin::none(),
            AnyIOPin::none(),
            AnyIOPin::none(),
            None,
        )?,
        &SdCardConfiguration::new(),
    )?;
    info!("Mounting File");
    // Keep it around or else it will be dropped and unmounted
    let _mounted_fatfs = MountedFatfs::mount(Fatfs::new_sdcard(0, sd_card_driver)?, "/sdcard", 4)?;
```

To work with my esp32-s3.
I tried on 2 different sd-card devices (one is a Waveshare (https://www.waveshare.com/micro-sd-storage-board.htm) and the other is an Amazon bought one.
I read in some obscure posts that the problem might be with my sd card, so I'll have to wait for the new ones to arrive. I am using this sd card formatter (https://www.sdcard.org/downloads/formatter/).

This may take a while, and since there was another person who confirmed that it works (here: https://github.com/esp-rs/esp-idf-svc/issues/467#issuecomment-2268271039), I think you can close it (if you feel that is the case, of course).

Thank you

ivmarkov, 2024-08-06:
It is very likely you might just not be connecting it properly.

Are you sure you are connecting mosi of the board to miso of the card and miso of the board to mosi of the card?

fabracht, 2024-08-06:
Ah, that might have been it. I swapped them, and now I get a different problem:
```
I (1019) main_task: Started on CPU0
I (1023) main_task: Calling app_main()
I (1029) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1038) sdspi_transaction: cmd=52, R1 response: command not supported
I (1086) sdspi_transaction: cmd=5, R1 response: command not supported
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060930  A0      : 0x820345a5  A1      : 0x3fc9a7d0  
A2      : 0x3fc9c97c  A3      : 0x3fc9ca0c  A4      : 0x0000012b  A5      : 0x3fc9a7d8  
A6      : 0x3fc9a804  A7      : 0x00000008  A8      : 0x820352ac  A9      : 0x3fc9a7b0  
A10     : 0x00000001  A11     : 0x3fc9a804  A12     : 0x00000028  A13     : 0x3fc9b590  
A14     : 0x5b590001  A15     : 0x00000009  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3fc9a7d0 0x420345a2:0x3fc9a840 0x42033a2c:0x3fc9a870 0x420036af:0x3fc9a8a0 0x42003e0e:0x3fc9a9d0 0x4204f9a7:0x3fc9ae90 0x420038d0:0x3fc9aeb0 0x4201aa07:0x3fc9af00 0x42004667:0x3fc9af30 0x42005977:0x3fc9af60 0x4205253f:0x3fc9af80

ELF file SHA256: 000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376675
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1710
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb4
SHA-256 comparison failed:
Calculated: 1549cc352de9d9bcb10175fcd43380ff2e0209e9a6b10dd38e454bc060f04976
Expected: ab492d70a98095a7841c7bfbf1d6d0b6161e082227abfb2028c51ce7cbe6144a
Attempting to boot anyway...
entry 0x403b61c0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bccf4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1710
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb4
SHA-256 comparison failed:
Calculated: 1549cc352de9d9bcb10175fcd43380ff2e0209e9a6b10dd38e454bc060f04976
Expected: ab492d70a98095a7841c7bfbf1d6d0b6161e082227abfb2028c51ce7cbe6144a
Attempting to boot anyway...
entry 0x403b61c0
```

fabracht, 2024-08-06:
I finally got it to work. I had to change some things around and I made sure to double check everything along the way, so I'm pretty confident there were no "mess-ups" this time. 
The working setup:
- ESP32 Devkit V1
- Waveshare mini sd card reader
- 16Gb Kingston MicroSD HC

Using the other 64GB Sandisk SDXC gave me a 
"Error: Os { code: 19, kind: Uncategorized, message: "No such device" }"

With the ESP32-S3-WROOM-1-N16R8, I get the panic message. No matter which card, pins, or sd reader I use.
I'm probably missing something there, but I just wanted to leave it documented.

So, in summary. I can confirm it works for the esp32.

ivmarkov, 2024-08-07:
> Ah, that might have been it. I swapped them, and now I get a different problem:
>
> ```
> I (1019) main_task: Started on CPU0
> I (1023) main_task: Calling app_main()
> I (1029) gpio: GPIO[10]| InputEn: 0| [...]

If you run this with `espflash monitor --elf <your-elf-file>` we'll get a decoded stacktrace on the esp32s3 so we'll see where exactly it is crashing. Though the previous errors before the crash:
```
I (1038) sdspi_transaction: cmd=52, R1 response: command not supported
I (1086) sdspi_transaction: cmd=5, R1 response: command not supported
```

... might mean things are already messed up. Do you get these two on the esp32 with SPI as well?

BTW: You ONLY try to do SD-SPI? Right?

fabracht, 2024-08-07:
Hello,

> BTW: You ONLY try to do SD-SPI? Right?

Yes. Only SD-SPI so far.

I ran with the espflash monitor command, and got this:
```
I (1043) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1052) sdspi_transaction: cmd=52, R1 response: command not supported
I (1100) sdspi_transaction: cmd=5, R1 response: command not supported
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x82042409  A1      : 0x3fc9a760
0x00000000 - KEXC_pc
    at ??:??
0x00060f30 - UserFrameTotalSize
    at ??:??
0x82042409 - _rtc_reserved_end
    at ??:??
0x3fc9a760 - _heap_low_start
    at ??:??
A2      : 0x3fc9c98c  A3      : 0x3fc9ca1c  A4      : 0x0000012b  A5      : 0x3fc9a768
0x3fc9c98c - _heap_low_start
    at ??:??
0x3fc9ca1c - _heap_low_start
    at ??:??
0x0000012b - UserFrameTotalSize
    at ??:??
0x3fc9a768 - _heap_low_start
    at ??:??
A6      : 0x3fc9a794  A7      : 0x00000008  A8      : 0x82043110  A9      : 0x3fc9a740
0x3fc9a794 - _heap_low_start
    at ??:??
0x00000008 - UEXC_sar
    at ??:??
0x82043110 - _rtc_reserved_end
    at ??:??
0x3fc9a740 - _heap_low_start
    at ??:??
A10     : 0x00000001  A11     : 0x3fc9a794  A12     : 0x00000028  A13     : 0x3fc9b5a0
0x00000001 - UEXC_pc
    at ??:??
0x3fc9a794 - _heap_low_start
    at ??:??
0x00000028 - UEXC_lbeg
    at ??:??
0x3fc9b5a0 - _heap_low_start
    at ??:??
A14     : 0x5b590001  A15     : 0x00000009  SAR     : 0x00000000  EXCCAUSE: 0x00000014
0x5b590001 - _rtc_data_end
    at ??:??
0x00000009 - KEXC_areg
    at ??:??
0x00000000 - KEXC_pc
    at ??:??
0x00000014 - UEXC_a3
    at ??:??
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000
0x00000000 - KEXC_pc
    at ??:??
0x40056f5c - rom_rx_gain_force
    at ??:??
0x40056f72 - rom_rx_gain_force
    at ??:??
0x00000000 - KEXC_pc
    at ??:??

Backtrace: 0xfffffffd:0x3fc9a760 0x42042406:0x3fc9a7d0 0x42041890:0x3fc9a800 0x42005d58:0x3fc9a830 0x42005072:0x3fc9a960 0x4205e023:0x3fc9ae50 0x42005868:0x3fc9ae70 0x4201863c:0x3fc9ae90 0x42018664:0x3fc9aec0 0x42018079:0x3fc9aef0 0x42005858:0x3fc9af20 0x4200583b:0x3fc9af50 0x42008917:0x3fc9af70 0x42060407:0x3fc9af90
0xfffffffd - _rtc_reserved_end
    at ??:??
0x3fc9a760 - _heap_low_start
    at ??:??
0x42042406 - sdmmc_init_sd_scr
    at /home/esp/esps3_sd_spi/.embuild/espressif/esp-idf/master/components/sdmmc/sdmmc_sd.c:71
0x3fc9a7d0 - _heap_low_start
    at ??:??
0x42041890 - sdmmc_card_init
    at /home/esp/esps3_sd_spi/.embuild/espressif/esp-idf/master/components/sdmmc/sdmmc_init.c:122
0x3fc9a800 - _heap_low_start
    at ??:??
0x42005d58 - esp_idf_hal::sd::sdcard::SdCardDriver<esp_idf_hal::sd::spi::SdSpiHostDriver<T>>::new_spi
    at /home/esp/.cargo/git/checkouts/esp-idf-hal-923d08f725716f5e/72f6448/src/sd.rs:230
0x3fc9a830 - _heap_low_start
    at ??:??
0x42005072 - esps3_sd_spi::main
    at /home/esp/esps3_sd_spi/src/main.rs:64
0x3fc9a960 - _heap_low_start
    at ??:??
0x4205e023 - core::ops::function::FnOnce::call_once
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:250
0x3fc9ae50 - _heap_low_start
    at ??:??
0x42005868 - std::rt::lang_start::{{closure}}
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:159
0x3fc9ae70 - _heap_low_start
    at ??:??
0x4201863c - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ops/function.rs:284
0x3fc9ae90 - _heap_low_start
    at ??:??
0x42018664 - std::panic::catch_unwind
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panic.rs:149
0x3fc9aec0 - _heap_low_start
    at ??:??
0x42018079 - std::panic::catch_unwind
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panic.rs:149
0x3fc9aef0 - _heap_low_start
    at ??:??
0x42005858 - std::rt::lang_start
    at /home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/rt.rs:158
0x3fc9af20 - _heap_low_start
    at ??:??
0x4200583b - main
    at ??:??
0x3fc9af50 - _heap_low_start
    at ??:??
0x42008917 - app_main
    at /home/esp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.35.0/src/start.rs:46
0x3fc9af70 - _heap_low_start
    at ??:??
0x42060407 - main_task
    at /home/esp/esps3_sd_spi/.embuild/espressif/esp-idf/master/components/freertos/app_startup.c:208
0x3fc9af90 - _heap_low_start
    at ??:??

ELF file SHA256: 000000000
```

ivmarkov, 2024-08-08:
Why are you using ESP IDF `master`? You should not be doing that as it is explicitly marked as UNSTABLE.

Please use a released ESP IDF (v5.1, v5.2 or v5.3 or just whatever the template generates by default!).

If you could do that and then grab the stacktrace again and paste here.

fabracht, 2024-08-08:
My fellow programmer. Why are you so snappy and quick to blame on your responses?
You ask, "Why are you using the esp idf master"? Because it is a choice there in the template. If I should not use it, then, don't put the choice there. Also, I don't think that's the problem here.
I've been trying to keep it cool, but I didn't like the "pointing of fingers" on the "crates patch". 
I think you should really rethink what you are doing here. If you don't like people pointing out problems with the code, don't take out your frustrations on them and try to focus solely on the problem at hand.
If you don't think the user is not doing enough of a job, do as I so, solve it yourself.
You need to stop assuming that this is my main concern. Mistakes and left over code from previous modifications and learning experiments will go through. I have plenty of experience to recognize that when a help request comes my way. Patience and cordiality are key to having quality and proper feedback.
If you don't want people to choose unstable, don't put it there.

I think I'm done with you. 
Again, you guys are doing a terrific job and I wish you all the luck in the world.
I wouldn't worry too much about my errors. As you pointed out time and again, I probably messed up again, so your code is likely solid.

I'm closing the issue.
Thank you again for the patience.

closed 2024-08-08 by fabracht

ivmarkov, 2024-08-08:
> You ask, "Why are you using the esp idf master"? Because it is a choice there in the template. If I should not use it, then, don't put the choice there. 

_I never said this is the root cause_ - or else why am I asking that you grab another stack trace with a stable/released ESP IDF? 

However, putting one unstable thing (`esp-idf-hal`-`master` with the `experimental` flag) on top of another unstable thing (ESP IDF `master`) means _even more variables to an already complex equation_. Hence why my request to simplify your setup.

> Because it is a choice there in the template. If I should not use it, then, don't put the choice there.

There is a reason why it is marked with "UNSTABLE" (all upper-case if I remember correctly) _and_ - in the "advanced" settings - this is to signal folks who are new to the echo-system that "there are dragons", so they should not go there unless they know really well what they are doing (and try not to get offense with this in particular again - everybody was a rookie at some point, including me and I still am - in a lot of things.)

> Also, I don't think that's the problem here. I've been trying to keep it cool, but I didn't like the "pointing of fingers" on the "crates patch". I think you should really rethink what you are doing [...]

What finger-pointing? Either you are doing it correctly, or you are doing it wrongly. Shall I just stay silent when it is just wrong and is - again - another variable in the equation we are trying to solve? Also, you explicitly asked for a detailed explanation w.r.t. how `[patch.crates-io]` works. In retrospective, I should have probably just pasted a link to the Cargo Book...

=====

With that said:

**If I sound harsh sometimes - I do apologize!**

1. I don't mean to, but it is indeed a bit my character in that in the lack of regular verbal communication I have found I do sound a bit aggressive indeed! Hence the apology.

2. But also - and **unfortunately** - it is a bit of being between a rock and a hard place in my situation in that if I'm massaging my messages too much (and not using e.g. bold and italic and a bit more direct language), folks often seem not to pay attention to the detail, and then I have to repeat and repeat :(

fabracht, 2024-08-08:
Thank you. And I apologize if I caused any inconvenience for you. But it is my personal preference to simply step away from these misunderstanding and small possible conflicts. Cutting it by the root is the best option, before I get emotionally involved. It's a me thing.
I just want to emphasize that this is not about you. Like I said, you are doing a great job. Much better than what I could have done, honestly. So kudos on it.