esp-rs/esp-idf-svc #486: Serial commands received and stored

djhunter67, 2024-09-19:
# Serial Commands
- I am unable to receive serial commands and store them into a simple variable, let alone a struct.

I have been working on this for several days.  Solely trying to receive serial text from Arduino IDE serial monitor and/or putty.  I cannot find any examples in any of the Rust books to accomplish this. 

## Help
- How do I receive text from the Serial Pins and the usb port?

ivmarkov, 2024-09-19:
https://github.com/esp-rs/esp-idf-hal/blob/master/examples/uart_loopback.rs

closed 2024-09-19 by ivmarkov

djhunter67, 2024-09-22:
Thank you so much.  In the many hours of looking, I could not find anything that looks similar to this.  Looks like I need to refactor my research skills.