esp-rs/esp-idf-svc #489: esp32c2 use example http_sw_server report err ESP_ERR_HTTPD_TASK

xiaguangbo, 2024-09-21:
changed code:
```rust
const SSID: &str = "hhh";
const PASSWORD: &str = "12345678";
```

.cargo/config.toml
```toml
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor --baud 2000000" # Select this runner for espflash v3.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c2"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.3.1"

# Workaround for https://github.com/esp-rs/esp-idf-template/issues/174 
CRATE_CC_NO_DEFAULTS = "1"

ESP_IDF_TOOLS_INSTALL_DIR = "custom:../../../../data/esp/espressif"

```

sdkconfig.defaults:
```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granularity for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n

CONFIG_XTAL_FREQ_26=y
CONFIG_FREERTOS_HZ=1000

CONFIG_HTTPD_WS_SUPPORT=y

```

Cargo.toml:
```toml
[package]
name = "project"
version = "0.1.0"
authors = ["xiaguangbo"]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "project"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
lto=true
incremental=true
codegen-units = 1
panic = "abort"
strip = "symbols"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "*", default-features = false }
esp-idf-svc = { version = "*", default-features = false }

embedded-svc = "*"
anyhow = "*"

[build-dependencies]
embuild = "*"

[patch.crates-io]
embuild = { git = "https://github.com/esp-rs/embuild" }
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
embedded-svc = { git = "https://github.com/esp-rs/embedded-svc" }

```

log:
```
$ cargo run --release
   Compiling project v0.1.0 (/var/home/xiaguangbo/xiaguangbo/project/espwifi/program_esp32c2/project)
    Finished `release` profile [optimized] target(s) in 3.89s
     Running `espflash flash --monitor --baud 2000000 target/riscv32imc-esp-espidf/release/project`
[2024-09-21T12:26:41Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-09-21T12:26:41Z INFO ] Connecting...
[2024-09-21T12:26:41Z INFO ] Using flash stub
[2024-09-21T12:26:43Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0e:c1:1c
App/part. size:    890,336/2,031,616 bytes, 43.82%
[2024-09-21T12:26:43Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-09-21T12:26:43Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:09] [========================================]     541/541     0x10000                                                                                                 [2024-09-21T12:26:54Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 001f0000
I (126) boot: End of partition table
I (132) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=278fch (162044) map
I (201) esp_image: segment 1: paddr=00037924 vaddr=3fcaf940 size=028b4h ( 10420) load
I (205) esp_image: segment 2: paddr=0003a1e0 vaddr=40380000 size=05e38h ( 24120) load
I (219) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9fa98h (653976) map
I (447) esp_image: segment 4: paddr=000dfac0 vaddr=40385e38 size=09afch ( 39676) load
I (467) boot: Loaded app from partition at offset 0x10000
I (467) boot: Disabling RNG early entropy source...
I (486) cpu_start: Unicore app
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 120000000 Hz
I (494) app_init: Application information:
I (499) app_init: Project name:     libespidf
I (506) app_init: App version:      1
I (513) app_init: Compile time:     Sep 21 2024 19:41:25
I (522) app_init: ELF file SHA256:  000000000...
I (530) app_init: ESP-IDF:          v5.3.1
I (538) efuse_init: Min chip rev:     v1.0
I (545) efuse_init: Max chip rev:     v1.99
I (553) efuse_init: Chip rev:         v1.0
I (560) heap_init: Initializing. RAM available for dynamic allocation:
I (571) heap_init: At 3FCB6220 len 00026950 (154 KiB): RAM
I (581) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
I (591) spi_flash: detected chip: gd
I (597) spi_flash: flash io: dio
I (603) sleep: Configure to isolate all GPIO pins in sleep state
I (613) sleep: Enable automatic switching of GPIO sleep configuration
I (624) main_task: Started on CPU0
I (630) main_task: Calling app_main()
I (651) pp: pp rom version: de8c800
I (652) net80211: net80211 rom version: de8c800
I (654) wifi:wifi driver task: 3fcbe8a0, prio:23, stack:6144, core=0
I (669) wifi:wifi firmware version: ccaebfa
I (669) wifi:wifi certification version: v7.0
I (672) wifi:config NVS flash: enabled
I (678) wifi:config nano formating: enabled
I (684) wifi:Init data frame dynamic rx buffer num: 32
I (691) wifi:Init static rx mgmt buffer num: 10
I (698) wifi:Init management short buffer num: 32
I (704) wifi:Init dynamic tx buffer num: 32
I (710) wifi:Init static tx FG buffer num: 2
I (717) wifi:Init static rx buffer size: 1600
I (723) wifi:Init static rx buffer num: 10
I (729) wifi:Init dynamic rx buffer num: 32
I (736) wifi_init: rx ba win: 6
I (741) wifi_init: accept mbox: 6
I (747) wifi_init: tcpip mbox: 32
I (753) wifi_init: udp mbox: 6
I (759) wifi_init: tcp mbox: 6
I (764) wifi_init: tcp tx win: 5760
I (771) wifi_init: tcp rx win: 5760
I (777) wifi_init: tcp mss: 1440
I (783) wifi_init: WiFi IRAM OP enabled
I (790) wifi_init: WiFi RX IRAM OP enabled
W (803) wifi:Affected by the ESP-NOW encrypt num, set the max connection num to 2
I (810) phy_init: phy_version 340,e255ce0,Jun  4 2024,16:44:11
I (848) wifi:mode : softAP (34:98:7a:0e:c1:1d)
I (851) wifi:Total power save buffer number: 16
I (852) wifi:Init max length of beacon: 752/752
I (852) wifi:Init max length of beacon: 752/752
I (860) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.71.1
I (872) project: Created Wi-Fi with WIFI_SSID `hhh` and WIFI_PASS `12345678`
Error: ESP_ERR_HTTPD_TASK
I (887) main_task: Returned from app_main()
```

cgudrian, 2024-09-21:
The reason for this behavior is a missing  reasonable default value for the `task_caps` field of the `httpd_config_t` struct. The Rust side initializes this field with 0 while the correct value would be `(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT)` which translates to `(1 << 11) | (1 << 2)`.

ivmarkov, 2024-09-22:
> The reason for this behavior is a missing reasonable default value for the `task_caps` field of the `httpd_config_t` struct. The Rust side initializes this field with 0 while the correct value would [...]

Indeed. I've pushed a quick fix which defines this field.

The more general solution is that we need to find a way to expose to Rust ESP IDF functional macros like [this one](https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/esp_http_server/include/esp_http_server.h#L53).

Given [this extension to bindgen](https://github.com/rust-lang/rust-bindgen/pull/2779) this task might seem finally possible.

ivmarkov, 2024-09-22:
Unfortunately the above extension does not work for me. Not sure what is wrong. Perhaps what ojega mentioned - it only works for "int"-like types.

closed 2024-09-22 by ivmarkov

xiaguangbo, 2024-09-22:
@cgudrian thanks

cgudrian, 2024-09-22:
> The more general solution is that we need to find a way to expose to Rust ESP IDF functional macros like [this one](https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/esp_http_server/include/esp_http_server.h#L53).

I doubt there can be a general solution. You would have to translate an arbitrary snippet of C code into the equivalent Rust code. In this special case my naive approach would be to write a small glue function in C that receives a `httpd_config_t*` and assigns it the default value:

```c
void httpd_default_config(httpd_config_t *cfg)
{
    static const httpd_config_t DEFAULT = HTTPD_DEFAULT_CONFIG();
    *cfg = DEFAULT;
}
```

One might even write a C macro for that. :)

ivmarkov, 2024-09-22:
> I doubt there can be a general solution. You would have to translate an arbitrary snippet of C code into the equivalent Rust code. In this special case my naive approach would be to write a small [...]

Sure, but how to compile the resulting C file and link it to the final executable? It is not so simple.

cgudrian, 2024-09-22:
> but how to compile the resulting C file and link it to the final executable?

I was hoping there's a way to inject custom code in the _embuild_ process.

ivmarkov, 2024-09-23:
There is no "call this single function to compile and link this/these C files" way. You have to do it yourself.
It is definitely possible, but a lot of work. As in:
- You have to first build the ESP-IDF libs (which is already happening as part of building `esp-idf-sys` of course)
- Post ESP-IDF build, you need to extract the path to CC from the BuildOptions returned to you
- You have to call CC on the generated C file and remember the location of the resulting `.o` file
- You have to output the `.o` file position as a cargo linker arg so that the final elf is linked with that file too

cgudrian, 2024-09-23:
Can we apply patches to the IDF source code before compiling?

ivmarkov, 2024-09-23:
We can, and we used to do it for older ESP-IDFs, but I really want to stop doing this.
It is just not nice at all.