esp-rs/esp-idf-svc #493: eth_phy_802_3: esp_eth_phy_802_3_reset(140): reset timeout

Graceful-Massana, 2024-09-27:
I'm trying to connect an [ESP32-PoE-ISO-WROVER](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) using ethernet with a fixed IP following the examples.

I'm getting this error:
`E (624) eth_phy_802_3: esp_eth_phy_802_3_reset(140): reset timeout
E (624) eth_phy_802_3: esp_eth_phy_802_3_basic_phy_init(435): reset failed
E (624) lan87xx: lan87xx_init(341): failed to init PHY
E (624) esp_eth: esp_eth_driver_install(229): init phy failed`

Esp-idf version: v5.1-beta1-378-gea5e0ff298-dirt

esp-idf-svc = { version = "0.49", default-features = false }

sdkconfig:
`# Ethernet
#
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
# CONFIG_ETH_RMII_CLK_INPUT is not set
CONFIG_ETH_RMII_CLK_OUTPUT=y
# CONFIG_ETH_RMII_CLK_OUTPUT_GPIO0 is not set
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
CONFIG_ETH_USE_SPI_ETHERNET=y
# CONFIG_ETH_SPI_ETHERNET_DM9051 is not set
CONFIG_ETH_SPI_ETHERNET_W5500=y
# CONFIG_ETH_SPI_ETHERNET_KSZ8851SNL is not set
# CONFIG_ETH_USE_OPENETH is not set
# end of Ethernet`

ivmarkov, 2024-09-27:
Can't say much without also looking at your program.
Some things to check is whether you are providing the _correct_ pins. Those which are hard-coded you cannot get wrong, but for example `rmii_mdio` and `rst` are NOT hard-coded. Are you configuring those correctly, according to this [schematics](file:///home/ivan/Downloads/ESP32-PoE-ISO_Rev_K.pdf)?

ivmarkov, 2024-09-27:
Copy-paste from [here](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv5.3/components/ethernet_init/ethernet_init.c#L65) as to what you need to use.

Graceful-Massana, 2024-10-07:
Just uploaded my code [here](https://github.com/Graceful-Massana/eth) .

I sent an email to Olimex and they advised to not use gpio12 as reset because it's meant to power PHY, but I'm still getting similar errors:

[image]

ivmarkov, 2024-10-07:
Did you read my two previous comments? Did you check- in their own C example - what pin they use for "reset"? Did you also check this in the schematics?

I hope you understand, that you can't just pick a free pin for "reset"? You have to use as a reset pin the exact pin, which - on their schematics - is connected to the resetnpin of the phy.

ivmarkov, 2024-10-07:
Ditto for rmii_mdio.

ivmarkov, 2024-10-21:
No feedback, closing.

closed 2024-10-21 by ivmarkov