esp-rs/esp-idf-svc #503: WiFi switching causes panic

hhy50, 2024-10-18:
I want to switch to different Wi-Fi every time the `connect_wifi`  is called.
but, it only succeeds the first time it is called, and second time it will cause the process to panic.

code:
```rust
static STARTED: AtomicBool = AtomicBool::new(false);
static ESP_WIFI_INSTANCE: OnceLock<Mutex<EspWifi<'_>>> = OnceLock::new();

#[derive(Debug, Default)]
pub struct Config<'a> {
    pub ssid: &'a str,
    pub passwd: &'a str,
}

/// init wifi instance
pub fn init<'a>(
    modem: modem::Modem,
    sys_loop: EspEventLoop<System>,
    nvs: Option<EspDefaultNvsPartition>,
) -> anyhow::Result<()> {
    sys_loop.subscribe::<WifiEvent, _>(|event| match event {
        WifiEvent::StaConnected => {
            log::info!("Esp Wifi connected");
        }
        WifiEvent::StaDisconnected => {
            log::warn!("Esp Wifi Disconnect");
        }
        other => {
            log::warn!("Wifi event: {:?}", other);
        }
    })?;
    let mut esp_wifi = EspWifi::new(modem, sys_loop, nvs)?;
    if let Err(_) = ESP_WIFI_INSTANCE.set(Mutex::new(esp_wifi)) {
        return Err(anyhow!("set ESP_WIFI_INSTANCE failed"));
    }
    Ok(())
}

pub fn connect_wifi(config: &Config) -> anyhow::Result<()> {
    log::info!("connect_wifi....");
    let connected = ESP_WIFI_INSTANCE.get().unwrap().lock().is_connected();
    let _ = match connected.ok() {
        Some(true) => {
            ESP_WIFI_INSTANCE.get().unwrap().lock().disconnect()?;
            wait_wifi_conn_status(false, Duration::from_secs(10));
        }
        _ => (),
    };
    let wifi_conf: Configuration = Configuration::Client(ClientConfiguration {
        ssid: config.ssid.try_into().unwrap(),
        password: config.passwd.try_into().unwrap(),
        auth_method: AuthMethod::WPA2Personal,
        ..Default::default()
    });
    ESP_WIFI_INSTANCE
        .get()
        .unwrap()
        .lock()
        .set_configuration(&wifi_conf)?;
    if !STARTED.load(std::sync::atomic::Ordering::Acquire) {
        ESP_WIFI_INSTANCE.get().unwrap().lock().start()?;
    }
    ESP_WIFI_INSTANCE.get().unwrap().lock().connect()?;
    wait_wifi_conn_status(true, Duration::from_secs(10));
    Ok(())
}

fn wait_wifi_conn_status(conn: bool, timeout: Duration) {
    let timeout = SystemTime::now().checked_add(timeout).unwrap();
    log::info!("wait_wifi_conn_status status [{}] start...", conn);
[... 353 lines cut ...]
I (640) net80211: net80211 rom version: 9387209
I (660) wifi:wifi driver task: 3fcbfe08, prio:23, stack:6656, core=0
I (660) wifi:wifi firmware version: cc1dd81
I (660) wifi:wifi certification version: v7.0
I (660) wifi:config NVS flash: disabled
I (670) wifi:config nano formating: disabled
I (670) wifi:Init data frame dynamic rx buffer num: 32
I (670) wifi:Init static rx mgmt buffer num: 10
I (680) wifi:Init management short buffer num: 32
I (680) wifi:Init dynamic tx buffer num: 32
I (690) wifi:Init static tx FG buffer num: 2
I (690) wifi:Init static rx buffer size: 1600
I (700) wifi:Init static rx buffer num: 10
I (700) wifi:Init dynamic rx buffer num: 32
I (700) wifi_init: rx ba win: 6
I (710) wifi_init: tcpip mbox: 32
I (710) wifi_init: udp mbox: 6
I (710) wifi_init: tcp mbox: 6
I (720) wifi_init: tcp tx win: 5760
I (720) wifi_init: tcp rx win: 5760
I (730) wifi_init: tcp mss: 1440
I (730) wifi_init: WiFi IRAM OP enabled
I (730) wifi_init: WiFi RX IRAM OP enabled
I (740) esp_idf_svc::nvs: NvsDefault dropped
```

ivmarkov, 2024-10-18:
Can you try with the `esp-idf-svc` from `master` instead? I think there is a problem with a recent event (home channel change) that was added via a contribution.

I just work-arounded the problem on `master` so you might want to try it out. To do it, just add these lines to your `Cargo.toml` and don't change anything else. Then rebuild:
```toml
[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
```

hhy50, 2024-10-18:
> Can you try with the `esp-idf-svc` from `master` instead? I think there is a problem with a recent event (home channel change) that was added via a contribution.
>
> I just work-arounded the problem on [...]

I'll try it

hhy50, 2024-10-18:
@ivmarkov ok, It can run normally now

hhy50, 2024-10-18:
@ivmarkov I have another question, which is that the subscribed event in `init()` has not been triggered. This issue is not a recent occurrence, but has been existing before

ivmarkov, 2024-10-18:
> @ivmarkov I still have a problem, that is, the wifi event subscribed in init() has never been triggered. This problem has always existed before

That's because you are not keeping around the `Subscription` object returned to you by `subscribe`. As a result, the subscription you are doing gets dropped immediately, and you end up in an unsubscribed state.

By the way, for these questions, it might be much better if you just join the `ESP-RS` Matrix room (linked in the readme of this crate).