esp-rs/esp-idf-svc #508: Ethernet to WiFi bridge: where to start?

FrigoEU, 2024-10-23:
Hi,

This isn't an issue but rather a request for a nudge in the right direction. I've made a fairly large application in Rust running on the ESP32. So far it's been fairly smooth sailing. But the next feature I'd like to build is to me quite difficult to get started on. 

The situation: my ESP32 will be plugged into Ethernet. When connection is successful (and it's configured as such) it should start a WiFi access point (with pre-configured ssid/password) and allow another device to connect to the WiFi and connect to the internet via the ESP32's Ethernet connection. I feel in theory this should be possible but I've found only very limited information/examples online (especially in the Rust language). 

I'd greatly appreciate it if somebody could point me in the right direction to get started on this.

Simon

FrigoEU, 2024-10-25:
Hi @ivmarkov ,

This helps so much! I plan to get started on this next week and I'll report my findings. Thank you so much for this, I don't think I would have been able to figure this out in a reasonable time myself. Many thanks!

Simon

ivmarkov, 2024-10-25:
Np. 
Actually, since you have one and only one Eth instance, your work would be simpler. Where I said:
> Create N (N > 0) EspEth instances (these are a combination of phy (EthDriver) + netif (EspNetif). For their netifs, use a custom NetworkConfiguration, so that you can assign a different netif key and different distance to each ethernet netif. In each ethernet EspEth instance, you need to set the same Ethernet Mac address, as per the example. There is a public API for that, I think, otherwise unsafely call into esp_idf_svc::sys::esp_eth_ioctl as per the example

... the custom network configuration is only necessary if you have multiple eth interfaces, as then each needs to have its own unique key and distance. Since you'll have only one, none of this is necessary.

Only the step where you assign the same MAC address to both your eth and the netif bridge is necessary.

FrigoEU, 2024-10-30:
Hey @ivmarkov,

I've been getting some work on and this and have been slowly porting the C code from the example to Rust. However, I got stuck on the [last part](https://github.com/espressif/esp-idf/blob/release/v5.2/examples/network/bridge/main/bridge_example_main.c#L181C48-L181C69) that uses the esp_netif_br_glue_* types and structs and functions. I can't seem to find these anywhere on the rust side. Not in esp-idf-svc, nor in esp-idf-sys. Is this just missing for now or am I looking in the wrong place?

FrigoEU, 2024-11-08:
Hey @ivmarkov,

Just checking if you've already looked at this. I don't have much experience making Rust bindings for C so I'm (for now at least) hoping you can point me in the right direction :P.

Simon

ivmarkov, 2024-11-08:
And in your `sdkconfig.defaults`:
```
CONFIG_ESP_NETIF_TCPIP_LWIP=y
CONFIG_ESP_NETIF_BRIDGE_EN=y

CONFIG_LWIP_NUM_NETIF_CLIENT_DATA=1
```

... as per the C example.

FrigoEU, 2025-01-28:
I've managed to get this working with the following code:

```rust
    let _bridge = match eth {
        Some::<BlockingEth<EspEth<SpiEth<&mut SpiDriver<'static>>>>>(mut eth) => {
            // Starting up Eth - WiFi bridge

            // In C code this is defined in https://github.com/espressif/esp-idf/blob/release/v5.2/examples/network/bridge/main/bridge_example_main.c
            // but I couldn't find this anywhere in Rust code so I just construct it here myself
            let BR0 = std::ffi::CString::new("BR0").unwrap();
            let br0 = std::ffi::CString::new("br0").unwrap();
            let mut bridgeif_config: bridgeif_config = bridgeif_config {
                max_fdb_dyn_entries: 10, // maximum number of address entries in dynamic forwarding database
                max_fdb_sta_entries: 2, // maximum number of address entries in static forwarding database
                max_ports: 2,           // maximum number of ports the bridge can consist of
            };
            let mut ESP_NETIF_INHERENT_DEFAULT_BR = esp_netif_inherent_config {
                flags: esp_netif_flags_ESP_NETIF_DHCP_CLIENT
                // esp_netif_flags_ESP_NETIF_DHCP_SERVER
                    // | esp_netif_flags_ESP_NETIF_FLAG_GARP
                    | esp_netif_flags_ESP_NETIF_FLAG_EVENT_IP_MODIFIED
                    | esp_netif_flags_ESP_NETIF_FLAG_IS_BRIDGE,
                // mac: Default::default(),
                mac, // same as eth mac
                // ip_info: unsafe {&_g_esp_netif_soft_ap_ip}, // if we are dhcp server I think?
                ip_info: std::ptr::null(),
                get_ip_event: ip_event_t_IP_EVENT_ETH_GOT_IP,
                lost_ip_event: ip_event_t_IP_EVENT_ETH_LOST_IP,
                if_key: BR0.as_ptr(),
                if_desc: br0.as_ptr(),
                route_prio: 70,
                bridge_info: &mut bridgeif_config,
            };

            let netif_br_cfg: esp_netif_config = esp_netif_config {
                driver: std::ptr::null(),
                base: &mut ESP_NETIF_INHERENT_DEFAULT_BR,
                stack: unsafe { _g_esp_netif_netstack_default_br },
            };

            let br_netif: *mut esp_netif_t =
                unsafe { esp_netif_new(&netif_br_cfg).as_mut() }.unwrap();

            // Create new esp netif bridge glue instance
            let netif_br_glue: esp_netif_br_glue_handle_t =
                unsafe { esp_idf_svc::sys::esp_netif_br_glue_new() };

            // Adding Ethernet to bridge
            esp!(unsafe { esp_netif_br_glue_add_port(netif_br_glue, eth.eth().netif().handle()) })
                .unwrap();

            // Adding WiFi to bridge
            esp!(unsafe {
                esp_netif_br_glue_add_wifi_port(
                    netif_br_glue,
                    wifi.lock().unwrap().wifi_mut().ap_netif().handle(),
                )
            })
            .unwrap();

            esp!(unsafe { esp_netif_attach(br_netif, netif_br_glue as *mut _) }).unwrap();

            let mut promiscuous: bool = true;
            let promiscuousp: *mut bool = &mut promiscuous;
            esp!(unsafe {
                esp_eth_ioctl(
                    eth.eth().driver().handle() as *mut _,
                    esp_eth_io_cmd_t_ETH_CMD_S_PROMISCUOUS,
                    promiscuousp as *mut _,
                )
            })
            .unwrap();

            info!("Starting ethernet");
            eth.start().expect("Failed to start eth");

            // -> "eth connected" = cable attached (I think?)
            info!("Waiting for eth to connect");
            eth.wait_connected().unwrap();
            let eth_connected = eth.is_connected().unwrap();
            info!("Eth connected: {:?}", eth_connected);

            // info!("Waiting for eth netif to be up");
            // eth.wait_netif_up().unwrap();
            // let ip_info = eth.eth().netif().get_ip_info().unwrap();
            // info!("Eth DHCP info: {:?}", ip_info);

            // -> network connection up
            // info!("Waiting for eth to be up");
            // eth.eth_wait_while(|| eth.is_up().map(|s| !s), None)
            //     .unwrap();
            let eth_up = eth.is_up().unwrap();
            let _ = eth_is_up.store(eth_up, Relaxed);
            info!("Eth up: {:?}", eth_up);

            info!("Starting wifi");
            wifi.lock().unwrap().start().unwrap();
            info!("Wifi started");

            Some((br_netif, netif_br_glue, eth))
        }
        None => {
            info!("No ethernet found");
            None
        }
    };

```