esp-rs/esp-idf-svc #515: the crate given by `ESP_IDF_SYS_ROOT_CRATE` does not exist in this workspace

vortex314, 2024-11-17:
When trying to include esp_idf_svc in my build for esp32, I always have the issue as mentioned below.
I try to include this crate as I need motor control and use the capture functionality which is not yet available in the esp_hal mcpwm no_std.
So I thought using the esp-idf-sys wrappers. 

I have no clue and there is no documentation available on what you need to specify for this env ESP_IDF_SYS_ROOT_CRATE variable.

Getting desperate to make this work after endless attempts to make this work. Anybody can help ? 

Sources : https://github.com/vortex314/zenoh-projects/tree/main/esp32-idf-dc-motor

```
   Compiling esp-idf-sys v0.35.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: the crate given by `ESP_IDF_SYS_ROOT_CRATE` does not exist in this workspace

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/release/build/esp-idf-sys-33e261a5dae3ad5b/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=the crate given by `ESP_IDF_SYS_ROOT_CRATE` does not exist in this workspace
  cargo:rerun-if-changed=/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/sdkconfig
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/lieven/esp/esp-idf" IDF_TARGET="esp32" PATH="/home/lieven/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/lieven/esp/esp-idf/components/espcoredump:/home/lieven/esp/esp-idf/components/partition_table:/home/lieven/esp/esp-idf/components/app_update:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/lieven/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/lieven/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/lieven/.local/bin:/home/lieven/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/lieven/.cargo/bin:/home/lieven/esp/esp-idf/components/espcoredump:/home/lieven/esp/esp-idf/components/partition_table:/home/lieven/esp/esp-idf/components/app_update:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/lieven/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/home/lieven/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/lieven/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/lieven/.espressif/python_env/idf5.1_py3.10_env/bin:/home/lieven/esp/esp-idf/tools:/home/lieven/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/go/bin:/home/lieven/.local/bin:/home/lieven/go/bin:/home/lieven/arm/bin:/usr/local/go/bin:/home/lieven/.local/bin:/home/lieven/go/bin:/home/lieven/arm/bin" PROJECT_DIR="/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor" SDKCONFIG_DEFAULTS="/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out/gen-sdkconfig.defaults;/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/sdkconfig" "cmake" "/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/lieven/esp/esp-idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/lieven/.espressif/python_env/idf5.1_py3.10_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address" "-DCMAKE_ASM_FLAGS= -mlongcalls"
  -- Building ESP-IDF components for target esp32
  -- Checking Python dependencies...
  Python requirements are satisfied.
  Constraint file: /home/lieven/.espressif/espidf.constraints.v5.1.txt
  Requirement files:
   - /home/lieven/esp/esp-idf/tools/requirements/requirements.core.txt
  Python being checked: /home/lieven/.espressif/python_env/idf5.1_py3.10_env/bin/python
  -- Project sdkconfig file /home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out/sdkconfig
  Loading defaults file /home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out/gen-sdkconfig.defaults...
  Loading defaults file /home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/sdkconfig...
  -- App "libespidf" version: e876ff0-dirty
  -- Adding linker script /home/lieven/workspace/zenoh-projects/esp32-idf-dc-motor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-428c047870959faf/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /home/lieven/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /home/lieven/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /home/lieven/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /home/lieven/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
[... cut ...]
               at ./build/native.rs:17:9
     7: build_script_build::main
               at ./build/build.rs:51:24
     8: core::ops::function::FnOnce::call_once
               at /home/rust/rust/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

 *  The terminal process "/usr/bin/bash '-c', '~/.cargo/bin/cargo run --release'" terminated with exit code: 101. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

ivmarkov, 2024-11-17:
Why don't you try to generate a small, clean project using the `esp-idf-template` first and see if that builds?

Please follow all instructions in its README carefully.

Also, do not mix `esp-hal` crates with `esp-idf-*` crates (esp-println). They don't mix and match yet.

ivmarkov, 2024-11-17:
Also, please don't try to download esp idf and activate it yourself. As per the README, this is supported bit NOT necessary, and unless you know really well what you are doing, this will certainly fail the build:
Using activated esp-idf v5.1.4 environment at '/home/lieven/esp/esp-idf'

closed 2024-11-25 by vortex314

ivmarkov, 2024-11-26:
> Started clean and added piece by piece. I created some actors based on esp-hal and tried to re-use them in esp-idf-svc context, which doen't work.

Mixing `esp-hal` with `esp-idf-*` currently does not work, and is unlikely if it would ever work. You have to pick one, or the other, and stick with it.

> Making an application in esp32 that uses embassy tasks , timers , channels and work async it's a complex undertaking which needs some perseverance. 

I can only agree. Add to this that Rust is not just a slightly enhanced version of C++, but a separate, very complex language and you need A LOT of perseverance. But it does pay off in the end. :-)

> The timer dependencies and critical sections across embassy and esp-idf-svc is a black box , any structured documentation ?

Yes, we are lacking a bit on the documentation front. 
We are trying to fill-in the gaps, but it happens slowly, because the `esp-idf-*` crates are a community effort, without any paid developer time from Espressif (I'll try to make this crystal clear in the crates' READMEs themselves, in the coming couple of weeks.)

We are open to PRs which fill-in the documentation gaps though. Specifically for timers and critical sections, it is actually not a rocket science, and - in fact - if you generate a new project using recent [esp-idf-template](https://github.com/esp-rs/esp-idf-template), everything w.r.t. timers and critical sections should be pre-set for you EXCEPT if you want to use `embassy-executor`, in which case you need to do a very small adjustment (depend on `critical-section/std` instead, and remove the `critical-section` feature from `esp-idf-svc`.)

> I hesitated going back to C++ , but again didn't want to loose all these nice Rust features. I noticed also that minimum image is +/- 1MB , is there a possibility to reduce this ?

`panic_immediate_abort` is something that if you use, will erase immediately 200[-300KB from your flash fatness, but you'll lose the panic error message.

Reality is, if you use lots of stuff from ESP-IDF (wifi, OTA, peripherals etc.) it would be > 1MB for sure.

I know with `esp-hal` the resulting firmware is a bit smaller, but I wouldn't say by a very wide margin?