esp-rs/esp-idf-svc #517: eth w5500 not work

hhy50, 2024-11-19:
code:
```rust
use esp_idf_svc::hal::prelude::FromValueType;
use esp_idf_svc::{
    eventloop::EspSystemEventLoop,
    hal::{prelude::Peripherals, spi},
    handle::RawHandle,
    nvs::EspDefaultNvsPartition,
    sys::{esp_netif_set_hostname, EspError},
};
use std::{thread, time::Duration};

fn main() -> Result<(), EspError> {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();
    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    // 初始化 ESP32 环境
    let peripherals: Peripherals = Peripherals::take()?;
    let sys_event_loop = EspSystemEventLoop::take()?;
    let nvs: esp_idf_svc::nvs::EspNvsPartition<esp_idf_svc::nvs::NvsDefault> =
        EspDefaultNvsPartition::take()?;

    let pins = peripherals.pins;
    let etg_driver = esp_idf_svc::eth::EthDriver::new_spi(
        spi::SpiDriver::new(
            peripherals.spi2,
            pins.gpio2,
            pins.gpio10,
            Some(pins.gpio3),
            &spi::SpiDriverConfig::new().dma(spi::Dma::Auto(4096)),
        )?,
        pins.gpio6,
        Some(pins.gpio7),
        Some(pins.gpio4),
        esp_idf_svc::eth::SpiEthChipset::W5500,
        5.MHz().into(),
        Some(&[0x02, 0x00, 0x00, 0xfc, 0x19, 0x01]),
        None,
        sys_event_loop,
    )?;
    log::info!("is_connected: {:?}", etg_driver.is_connected());
    // spi
    let mut esp_eth = esp_idf_svc::eth::EspEth::wrap(etg_driver)?;
    log::info!("start eth driver.......");
    esp_eth.start()?;
    // esp_e
    while esp_eth.is_started()? {
        thread::sleep(Duration::from_secs(1));
    }
    log::info!("eth driver started.....");
    Ok(())
}
```

log:
```txt
2024-11-19T08:56:26Z INFO ] Detected 2 serial ports
[2024-11-19T08:56:26Z INFO ] Ports which match a known common dev board are highlighted
[2024-11-19T08:56:26Z INFO ] Please select a port
[2024-11-19T08:56:27Z INFO ] Serial port: '/dev/cu.usbmodem142201'
[2024-11-19T08:56:27Z INFO ] Connecting...
[2024-11-19T08:56:27Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.4)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       e4:b0:63:00:4b:30
[... 224 lines cut ...]
I (319) cpu_start: Max chip rev:     v1.99 
I (324) cpu_start: Chip rev:         v0.4
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 3FC91A40 len 0002E5C0 (185 KiB): RAM
I (342) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (349) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (356) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (364) spi_flash: detected chip: generic
I (367) spi_flash: flash io: dio
W (371) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (381) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (390) sleep: Configure to isolate all GPIO pins in sleep state
I (397) sleep: Enable automatic switching of GPIO sleep configuration
I (404) coexist: coex firmware version: d96c1e51f
I (410) coexist: coexist rom version 9387209
I (415) main_task: Started on CPU0
I (415) main_task: Calling app_main()
E (525) w5500.mac: w5500_reset(305): reset timeout
E (525) w5500.mac: emac_w5500_init(823): reset w5500 failed
I (525) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (535) esp_eth: esp_eth_driver_install(228): init mac failed
I (535) esp_idf_svc::eventloop: System event loop dropped
E (545) spi_master: spi_master_deinit_driver(309): not all CSses freed
I (555) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrai
```

hhy50, 2024-11-19:
The problem is located here:

```rust
E (525) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (525) w5500.mac: emac_w5500_init(825): vefiry chip ID failed
I (525) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (535) esp_eth: esp_eth_driver_install(228): init mac failed
```

ivmarkov, 2024-11-26:
As discussed in the room, the problem is this line:
```
E (525) w5500.mac: emac_w5500_init(825): vefiry chip ID failed
```

... and this is unlikely to be a bug in the driver. Either the SPI pins are not connected correctly, or the external Ethernet module is not W5500 compatible. Closing.

closed 2024-11-26 by ivmarkov