esp-rs/esp-idf-svc #524: Shouldn't EspNetifDriver be `pub` ?

Luctins, 2024-12-05:
Shouldn't this line be `pub mod ...`? 
https://github.com/esp-rs/esp-idf-svc/blob/3f4a2c0ec893c5acaeb2cb4d8af320ab07e7928c/src/netif.rs#L960

I'm investigating how to use PPP over serial and wanted to use this, but it is not public outside the crate. 
Is the example code in the comment here: https://github.com/esp-rs/esp-idf-svc/blob/3f4a2c0ec893c5acaeb2cb4d8af320ab07e7928c/src/netif.rs#L1005-L1022
still valid?

ivmarkov, 2024-12-05:
[No it does not have to](https://github.com/esp-rs/esp-idf-svc/blob/3f4a2c0ec893c5acaeb2cb4d8af320ab07e7928c/src/netif.rs#L26).

To use PPP, you need to enable PPP support in your `sdkconfig.defaults`, with the relevant CONF setting. Google for "esp idf ppp enable sdkconfig"

ivmarkov, 2024-12-05:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-lwip-ppp-support

ivmarkov, 2024-12-05:
Oh. And you need `feature = "experimental"` on the `esp-idf-svc` dependency.

Luctins, 2024-12-05:
Oh! I really didn't saw that that `use driver::*` 🤦🏻 sorry and thanks for the help.
Also, why is this feature gated on `experimental` ?

ivmarkov, 2024-12-05:
Closing - still - let me know if you have troubles.

closed 2024-12-05 by ivmarkov