esp-rs/esp-idf-svc #527: with --cfg esp_idf_soc_sdmmc_use_gpio_matrix,  sd/mmc.rs has errors.

qianzhang5, 2024-12-11:
o   Compiling esp-idf-svc v0.49.1
error[E0560]: struct `esp_idf_hal::sys::sdmmc_slot_config_t` has no field named `clk`
   --> /home/qian/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/mmc.rs:170:17
    |
170 |                 clk: _clk.into_ref().deref().pin(),
    |                 ^^^ `esp_idf_hal::sys::sdmmc_slot_config_t` does not have this field
    |
    = note: all struct fields are already assigned

error[E0560]: struct `esp_idf_hal::sys::sdmmc_slot_config_t` has no field named `cmd`
   --> /home/qian/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/mmc.rs:172:17
    |
172 |                 cmd: _cmd.into_ref().deref().pin(),
    |                 ^^^ `esp_idf_hal::sys::sdmmc_slot_config_t` does not have this field
    |
    = note: all struct fields are already assigned

error[E0560]: struct `esp_idf_hal::sys::sdmmc_slot_config_t` has no field named `d0`
   --> /home/qian/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/mmc.rs:174:17
    |
174 |                 d0: _d0.into_ref().deref().pin(),
    |                 ^^ `esp_idf_hal::sys::sdmmc_slot_config_t` does not have this field
    |
    = note: all struct fields are already assigned

ivmarkov, 2024-12-11:
Without giving me info what chip and ESP IDF you've used, I can hardly do anything.

With that said, please patch `esp-idf-sys`, `esp-idf-hal` and `esp-idf-svc` to latest `master` (with `patch.crates-io`) if you plan to use the SDMMC code. And enable the `experimental` feature, as I think this functionality is still under the experimental flag.

ivmarkov, 2024-12-11:
Also. This:
`--cfg esp_idf_soc_sdmmc_use_gpio_matrix`

you should absolutely not do (i.e. custom manual `esp_idf_*` Rust cfgs. These are generated by `esp-idf-sys` itself, based on your `sdkconfig.defaults` settings.

closed 2024-12-11 by ivmarkov