esp-rs/esp-idf-svc #533: OTA example?

indexds, 2024-12-24:
Hello,

[Examples](https://github.com/esp-rs/esp-idf-svc/tree/master/examples) for this project don't include an OTA example. I'd like to do something like:
1. Boot normally,
2. Check if we're running the latest version of the firmware by getting a /latest API endpoint
3. Update to the latest firmware if we're not up to date then reboot
4. If something went wrong with the update process, revert to the previous version of the firmware.

I'm open to writing the example myself once I've figured it out, though a nudge in the right direction would be helpful.

(#258 related)

Thanks!

indexds, 2024-12-24:
https://github.com/faern/esp-ota

This seems to do what I want but I'm not sure whether the example is too complicated or not. It doesn't directly use esp-idf-svc either, but creates its own wrappers from esp-idf-sys.

ivmarkov, 2024-12-24:
The OTA API is one of the few which are quite well documented, including rough update steps in the `ota.rs` module, that you can locate and follow.

Here's a [blog](https://quan.hoabinh.vn/post/2024/03/programming-esp32-with-rust-ota-firmware-update) containing mote sample code how to do it. If you google the internet, you can find more.

closed 2024-12-24 by ivmarkov