esp-rs/esp-idf-svc #543: Errors compiling code generated with esp-rs/esp-idf-template

vbaggiol, 2025-01-01:
## Bug description
Compilation errors when trying to build the example project generated with 
`cargo generate esp-rs/esp-idf-template cargo`

- Would you like to work on a fix? I don't think I would be able to fix the problem efficiently :-(

## To Reproduce
Here's how I proceeded:

I re-installed rust dev environment from scratch:
```
rustup self uninstall
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh   # using option 1) "standard installation"
cargo install cargo-generate
cargo install esp-generate
```
Then I generated a project for `esp32c3` with name `prob-idf`:
[image]

```
cargo generate esp-rs/esp-idf-template cargo
cd prob-idf/
cargo build 
```

I have a [github repo](https://github.com/vbaggiol/prob-idf) to which I've committed the project and the build log.
It contains three branches:
- https://github.com/vbaggiol/prob-idf/tree/master - built with ESP_IDF_VERSION = v5.2.2 (stable) -> error log https://github.com/vbaggiol/prob-idf/blob/master/cargo_build.log
- https://github.com/vbaggiol/prob-idf/tree/esp_idf_version_521 - built with ESP_IDF_VERSION = v5.2.1 -> error log https://github.com/vbaggiol/prob-idf/blob/esp_idf_version_521/cargo_build_521.log 
- https://github.com/vbaggiol/prob-idf/tree/esp_idf_version_master - built with ESP_IDF_VERSION = master -> error log https://github.com/vbaggiol/prob-idf/blob/esp_idf_version_master/cargo_build_master.log

## Expected behavior
`cargo build` finishes successfully, without compilation errors.

## Environment

- Crate (`esp-idf-svc`) version: 0.49.1 
- ESP-IDF branch or tag: v5.2.2, v5.2.1, master
- Target device (MCU): esp32c3
- OS: Ubuntu 22.04

ivmarkov, 2025-01-01:
Tank you for the detailed bug report! Fortunately, it will disappear with the imminent upcoming new release of the esp idf crates. If you don't want to wait a couple of days, you can do any of these in the meantime:
* Build with rust nightly from beginning of December
* build with the esp toolchain. It works for riscv
* patch.crates-io all of embuild, esp-idf-sys, esp-idf-hal, esp-idf-svc to their git master versions, as the problem is already fixed there

closed 2025-01-01 by ivmarkov

vbaggiol, 2025-01-02:
Thanks a lot for the quick reaction! I'm happy to wait until the new release.