esp-rs/esp-idf-svc #549: esp32c6 doesn´t boot after flashing

thedevleon, 2025-01-08:
## Bug description

We have a custom PCB with an ESP32C6-WROOM-1-N8 module.
We already have firmware based on esp-idf-svc 0.49 and esp-idf v5.1.3 that flashes and runs just fine on this PCB.

However, we would like to update to the latest version of esp-idf-svc, and are running into a bootloop with even just a sightly adjusted sample from the esp-idf-template to match the increased flash size: https://github.com/thedevleon/esp-idf-issue-bootloop

Running 'cargo run --release' results in the following looping output only:

```
❯ cargo run --release
    Finished `release` profile [optimized] target(s) in 0.36s
     Running `espflash flash --partition-table partitions.csv -s 8mb --no-stub --monitor target/riscv32imac-esp-espidf/release/esp-idf-test`
[2025-01-08T17:01:40Z INFO ] Serial port: '/dev/ttyACM0'
[2025-01-08T17:01:40Z INFO ] Connecting...
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:25:ff:b8
Partition table:   partitions.csv
App/part. size:    371,568/4,063,232 bytes, 9.14%
[2025-01-08T17:01:41Z INFO ] Segment at address '0x0' has not changed, skipping write
[2025-01-08T17:01:41Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2025-01-08T17:01:41Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2025-01-08T17:01:41Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002f474
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
```

- Would you like to work on a fix? y

## To Reproduce
- clone the repo above
- connect a devkit with the WROOM-1-N8 module (via the native USB port, not UART)
- observe the looping output

## Expected behavior
I can see a hello world printed and the output of the bootloader.

## Environment

- Crate (`esp-idf-svc`) version: 0.50
- ESP-IDF branch or tag: v5.3.2
- Target device (MCU): esp32c6
- OS: Fedora Workstation 41

thedevleon, 2025-01-08:
If i first flash our existing firmware, and then the sample, I can see the output successfully:

```
❯ cargo run --release
   Compiling esp-idf-test v0.1.0 (/home/leon/Desktop/ivy/code/esp-idf-test)
    Finished `release` profile [optimized] target(s) in 0.97s
     Running `espflash flash --partition-table partitions.csv -s 8mb --no-stub --monitor target/riscv32imac-esp-espidf/release/esp-idf-test`
[2025-01-08T17:16:35Z INFO ] Serial port: '/dev/ttyACM0'
[2025-01-08T17:16:35Z INFO ] Connecting...
Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:25:ff:b8
Partition table:   partitions.csv
App/part. size:    371,568/4,063,232 bytes, 9.14%
[2025-01-08T17:16:35Z INFO ] Segment at address '0x0' has not changed, skipping write
[2025-01-08T17:16:35Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:01] [========================================]     201/201     0x10000                                                                                                                                                                               [2025-01-08T17:16:39Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x2d (SPI_FAST_FLASH_BOOT)
Saved PC:0x400228dc
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:02:08
I (24) boot: chip revision: v0.1
I (29) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 ota_0            OTA app          00 10 00010000 003e0000
I (88) boot:  4 ota_1            OTA app          00 11 003f0000 003e0000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=3dc3ch (252988) map
I (160) esp_image: segment 1: paddr=0004dc64 vaddr=40800000 size=023b4h (  9140) load
I (163) esp_image: segment 2: paddr=00050020 vaddr=42040020 size=11a18h ( 72216) map
I (180) esp_image: segment 3: paddr=00061a40 vaddr=408023b4 size=07388h ( 29576) load
I (188) esp_image: segment 4: paddr=00068dd0 vaddr=40809740 size=01d74h (  7540) load
I (193) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
I (209) cpu_start: Unicore app
I (218) cpu_start: Pro cpu start user code
I (218) cpu_start: cpu freq: 160000000 Hz
I (219) app_init: Application information:
I (221) app_init: Project name:     libespidf
I (226) app_init: App version:      1
I (230) app_init: Compile time:     Jan  8 2025 17:00:59
I (236) app_init: ELF file SHA256:  000000000...
I (242) app_init: ESP-IDF:          v5.3.2
I (246) efuse_init: Min chip rev:     v0.0
I (251) efuse_init: Max chip rev:     v0.99 
I (256) efuse_init: Chip rev:         v0.1
I (261) heap_init: Initializing. RAM available for dynamic allocation:
I (268) heap_init: At 4080C530 len 000700E0 (448 KiB): RAM
I (274) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (280) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (287) spi_flash: detected chip: generic
I (291) spi_flash: flash io: dio
W (296) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (304) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (315) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (324) sleep: Configure to isolate all GPIO pins in sleep state
I (330) sleep: Enable automatic switching of GPIO sleep configuration
I (337) coexist: coex firmware version: cbb41d7
I (343) coexist: coexist rom version 5b8dcfa
I (348) main_task: Started on CPU0
I (348) main_task: Calling app_main()
I (348) esp_idf_test: Hello, world!
```

Further notes:
- Putting the ESP into bootloader mode first, and then flashing just resuts in a bootloop with "waiting for download"
- Resetting via CTRL+R or via the EN pin doesn't help
- any attempt to monitor after the initial espflash --monitor from cargo run will result in a timeout
```
❯ espflash monitor
[2025-01-08T17:18:58Z INFO ] Serial port: '/dev/ttyACM0'
[2025-01-08T17:18:58Z INFO ] Connecting...
[2025-01-08T17:18:58Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command
  ```

ivmarkov, 2025-01-08:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002f474
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4086ed7a
SPIWP:0xee
```

How is that a problem with the new ESP IDF given that you can't even load and start the ESP IDF bootloader?

closed 2025-01-08 by ivmarkov

thedevleon, 2025-01-08:
> ```
> ESP-ROM:esp32c6-20220919
> Build:Sep 19 2022
> rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4002f474
> SPIWP:0xee
> mode:DIO, clock div:2
> load:0x4086c410,len:0xd48 [...]

I don't quite understand what you mean. Can you elaborate? I don't know why the bootloader doesn't load  either. All the configuration around the bootloader and partition layout is the same between our current firmware with the old esp-idf version and the new sample with the new esp-idf version. Why would one boot and not the other?

ivmarkov, 2025-01-08:
I have no idea because I have no detailed information for your configuration. What I'm saying however is that you are very likely looking at the wrong place. Your `esp-idf-test` (old or new) starts executing at microsecond 209 or so in your log. Everything before that is the ROM bootloader and then the 2nd stage bootloader flashed by espflash (as you are not flashing a custom one). So it sounds pretty much impossible the culprit to be the new esp idf 5.3, given that the esp idf 5.3 new code does not even get the chance to be loaded, let alone executed.

thedevleon, 2025-01-08:
I just generated a new project using cargo generate, but chose the v5.2 esp-idf sdk instead of the v5.3 and that boots an flashes fine... So at least from my view, that's the only difference and thus the culprit, wouldn´t you agree? 
So for now we'll stick with `ESP_IDF_VERSION = "v5.2.3"`.

ivmarkov, 2025-01-08:
- I just generated a new project using `cargo generate` (full logs below), flashed on the c3 and it works.
- I also tried to explain why I find the new ESP-IDF 5.3 support to be the least likely of all possible options (really, and again sounds theoretically impossible).

There are other more likely (IMO) possibilities:
- Brownout (though too early for that, usu. happens when Wifi kicks in) have you tried with more powerful USB power supply?
- Buggy board. Perhaps you should google why you get the `TG0_WDT_HPSYS` triggered for your board
- Old/buggy espflash

> So for now we'll stick with ESP_IDF_VERSION = "v5.2.3"

Sure, up to you.

=========================

```
ivan@m800:~/dev$ cargo generate esp-rs/esp-idf-template cargo
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
🤷   Project Name: test53
🔧   Destination: /home/ivan/dev/test53 ...
🔧   project-name: test53 ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c6
✔ 🤷   Configure advanced template options? · false
🔧   Moving generated files into: `/home/ivan/dev/test53`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /home/ivan/dev/test53
ivan@m800:~/dev$ cd test53
ivan@m800:~/dev/test53$ cargo build
    Updating crates.io index
     Locking 168 packages to latest compatible versions
  Downloaded autocfg v1.4.0
  Downloaded rustversion v1.0.19
  Downloaded toml_datetime v0.6.8
  Downloaded strum v0.24.1
  Downloaded shlex v1.3.0
  Downloaded bitflags v1.3.2
  Downloaded bindgen v0.71.1
  Downloaded globwalk v0.8.1
  Downloaded proc-macro-error2 v2.0.1
  Downloaded winnow v0.6.22
  Downloaded void v1.0.2
  Downloaded minimal-lexical v0.2.1
  Downloaded thiserror-impl v1.0.69
  Downloaded same-file v1.0.6
  Downloaded cfg-if v1.0.0
  Downloaded libc v0.2.169
  Downloaded unicode-ident v1.0.14
  Downloaded regex v1.11.1
  Downloaded strum v0.25.0
  Downloaded fastrand v2.3.0
  Downloaded unicode-xid v0.2.6
  Downloaded remove_dir_all v0.8.4
  Downloaded thiserror-impl v2.0.9
  Downloaded uncased v0.9.10
  Downloaded strum_macros v0.24.3
  Downloaded which v4.4.2
  Downloaded pin-utils v0.1.0
  Downloaded thiserror v2.0.9
  Downloaded walkdir v2.5.0
  Downloaded serde_derive v1.0.217
  Downloaded serde v1.0.217
  Downloaded toml_edit v0.22.22
  Downloaded darling_core v0.20.10
  Downloaded memchr v2.7.4
  Downloaded serde_json v1.0.135
  Downloaded nom v7.1.3
  Downloaded aho-corasick v1.1.3
  Downloaded futures-util v0.3.31
  Downloaded syn v1.0.109
  Downloaded itertools v0.13.0
[... 319 lines cut ...]
I (260) cpu_start: ESP-IDF:          v5.2.3
I (265) cpu_start: Min chip rev:     v0.0
I (270) cpu_start: Max chip rev:     v0.99 
I (274) cpu_start: Chip rev:         v0.0
I (279) heap_init: Initializing. RAM available for dynamic allocation:
I (286) heap_init: At 4080BE50 len 000707C0 (449 KiB): RAM
I (292) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (299) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (306) spi_flash: detected chip: generic
I (310) spi_flash: flash io: dio
W (314) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (327) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (335) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (346) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (355) sleep: Configure to isolate all GPIO pins in sleep state
I (362) sleep: Enable automatic switching of GPIO sleep configuration
I (369) coexist: coex firmware version: 27d8387
I (374) coexist: coexist rom version 5b8dcfa
I (379) main_task: Started on CPU0
I (379) main_task: Calling app_main()
I (379) test53: Hello, world!
I (389) main_task: Returned from app_main()
ivan@ivan-laptop:~$ 

```

ivmarkov, 2025-01-08:
In all fairness, the log from above ^^^ was for ESP IDF 5.2.3 as apparently this is still the default. Yet - I took the effort, re-generated a completely new project, went to advanced settings, made sure I select "5.3.2", built, flashed, ran. Same result.

If you want to get to the bottom of this, googling for `TG0_WDT_HPSYS` early during bootup might be key.