esp-rs/esp-idf-svc #552: esp-idf-svc 0.51 compilation error

haijd, 2025-01-21:
## Bug description

After I updated `esp-idf-svc` to the latest version 0.51.0, an error occurred when compiling the project.
The error is as follows:
```
error[E0599]: no method named `vtable` found for reference `&Waker` in the current scope
  --> /Users/test/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.7.0/src/raw/waker.rs:37:14
   |
37 |     if waker.vtable() as *const _ != &VTABLE as *const _ {
   |              ^^^^^^ method not found in `&Waker`

error[E0599]: no method named `data` found for reference `&Waker` in the current scope
  --> /Users/test/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.7.0/src/raw/waker.rs:41:38
   |
41 |     unsafe { TaskRef::from_ptr(waker.data() as *const TaskHeader) }
   |                                      ^^^^ method not found in `&Waker`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `embassy-executor` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
    Building [======================>  ] 224/235: esp-idf-sys(build)
```

- Would you like to work on a fix? [n]

## To Reproduce

The version of esp-idf-svc is 0.51.0, and the enabled features are: "critical-section", "embassy-time-driver", "embassy-sync", "experimental"

## Environment

- Crate (`esp-idf-svc`) version: [0.51.0]
- ESP-IDF branch or tag: [v5.2.3]
- Target device (MCU): [esp32s3]
- OS: [macOS]

ivmarkov, 2025-01-21:
Update to Rust 1.84.

closed 2025-01-21 by ivmarkov

ScottCUSA, 2025-02-16:
`cargo install espup`
`espup install`