esp-rs/esp-idf-svc #554: Set hostname for Ethernet router configuration?

indexds, 2025-01-26:
Hi, is there a way to set a hostname so that the computer the esp is connected to gets that hostname attributed on the wired connection? I saw this function

```Rust
use esp_idf_svc::sys::esp_netif_set_hostname;
```

But its wrapped version isn't public..

```Rust
fn set_hostname(&mut self, hostname: &str) -> Result<(), EspError>
```

And unless I'm mistaken, the `new_with_conf` function never sets a hostname if the configuration is in router mode. Why is that?

I also tried calling the function directly, just before starting the netif, but it didn't seem to work.

```Rust
esp!(unsafe { esp_netif_set_hostname(self.handle, hostname.as_ptr() as *const _) })?;
```

The program runs fine, but the name hasn't changed. I get a generic "Network 5" wired connection name on my computer.

ivmarkov, 2025-01-26:
If you want a nice name for your ESP hostname, you need to run a local DNS proxy (and deliver the DNS proxy IP via the DHCP conf to yuor PC) which resolves whatever name you select for your ESP as its IP, but **further** forwards all other DNS requests to the actual DNS server (1.1.1.1, 8.8.8.8 etc.). Difficult.

indexds, 2025-01-26:
Huh, I thought the dhcp hostname option was sent over when the handshake is happening. 

So you're saying mDNS is the solution here. Seems right to me. But iirc in #529 we established there was a problem when ethernet is in router mode..

I'm not too hung up over this though. If it doesn't work it's fine, as it doesn't impact functionality at all. It's purely cosmetic.

ivmarkov, 2025-01-26:
You can still fix the mDNS issue if you want though - comment here: https://github.com/esp-rs/esp-idf-svc/issues/529#issuecomment-2564072707

This you need to patch:
> Also since you are using a "router" eth, you might want to set the IF name of your netif to ETH_DEF, if you follow the conversation in that thread ^^^.

indexds, 2025-01-26:
Right yeah, that's what I have currently:

```rust
    let mut eth_netif = EspEth::wrap_all(
        eth_driver,
        EspNetif::new_with_conf(&NetifConfiguration {
            flags: 0,
            key: "ETH_DEF".try_into().unwrap(),
            description: "eth".try_into().unwrap(),
            route_priority: 10,
            ip_configuration: Some(Configuration::Router(RouterConfiguration {
                subnet: Subnet {
                    gateway: ETH_GATEWAY,
                    mask: Mask(30),
                },
                dhcp_enabled: true, // adds dhcp_server flag
                dns: None,
                secondary_dns: None,
            })),
            stack: NetifStack::Eth,
            custom_mac: None,
            got_ip_event_id: None,
            lost_ip_event_id: None,
        })?,
    )?;
```

I *think* I had tried at the time to do this and it didn't work so I gave up, but my memory's fuzzy on this. Might try again in a bit.

indexds, 2025-02-02:
Ok so I took the time to recompile everything properly. The ethernet code is the one pasted above, I added those lines afterward (mdns is kept in scope after this, never dropped.)

```rust 
let mut mdns = EspMdns::take()?;
mdns.set_hostname("charizhard")?;
```
I also added this line to kconfig `CONFIG_MDNS_PREDEF_NETIF_ETH=y`

Wireshark is not seeing any answers to my mdns requests.

My suspicion at this point is that this `CONFIG_MDNS_PREDEF_NETIF_ETH=y` parameter might be ignored at compile time.

The way i'm doing it is by simply adding the line at the end of the `sdkconfig` generated by cargo-pio. Is that not the way to go about this?