esp-rs/esp-idf-svc #556: bt_gatt_server example fails on esp32c6 with v5.2.3 and v5.3.2

untbu, 2025-01-27:
## Bug description

Calling `BtDriver::new(peripherals.modem, Some(nvs.clone()))?` on an esp32c6 with ESP-IDF version `v5.2.3` or `v5.3.2` results in the error `ESP_ERR_INVALID_VERSION`. On an esp32c3 or with version `v5.2.2` it works as expected.

<details>
<summary>Log</summary>

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x7f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4080053c
0x4080053c - bootloader_mspi_reset
    at /mnt/storage/code/esp-idf-svc/.embuild/espressif/esp-idf/v5.3.2/components/bootloader_support/bootloader_flash/src/bootloader_flash.c:807
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:02:08
I (24) boot: chip revision: v0.1
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00300000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=1756fch (1529596) map
I (404) esp_image: segment 1: paddr=00185724 vaddr=40800000 size=028f4h ( 10484) load
I (407) esp_image: segment 2: paddr=00188020 vaddr=42178020 size=42f28h (274216) map
I (466) esp_image: segment 3: paddr=001caf50 vaddr=408028f4 size=0fb54h ( 64340) load
I (481) esp_image: segment 4: paddr=001daaac vaddr=40812450 size=02d4ch ( 11596) load
I (489) boot: Loaded app from partition at offset 0x10000
I (490) boot: Disabling RNG early entropy source...
I (502) cpu_start: Unicore app
I (510) cpu_start: Pro cpu start user code
I (511) cpu_start: cpu freq: 160000000 Hz
I (511) app_init: Application information:
I (513) app_init: Project name:     libespidf
I (519) app_init: App version:      v0.51.0-1-gf266359a6-dirty
I (525) app_init: Compile time:     Jan 27 2025 16:41:58
I (531) app_init: ELF file SHA256:  000000000...
I (536) app_init: ESP-IDF:          v5.3.2
I (541) efuse_init: Min chip rev:     v0.0
I (546) efuse_init: Max chip rev:     v0.99 
I (551) efuse_init: Chip rev:         v0.1
I (556) heap_init: Initializing. RAM available for dynamic allocation:
I (563) heap_init: At 40816850 len 00065DC0 (407 KiB): RAM
I (569) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (575) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (582) spi_flash: detected chip: generic
I (586) spi_flash: flash io: dio
W (590) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (599) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (609) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (618) sleep: Configure to isolate all GPIO pins in sleep state
I (625) sleep: Enable automatic switching of GPIO sleep configuration
I (632) coexist: coex firmware version: cbb41d7
I (637) coexist: coexist rom version 5b8dcfa
I (643) main_task: Started on CPU0
I (643) main_task: Calling app_main()
I (653) esp_idf_svc::bt: Init bluetooth controller
I (663) BLE_INIT: Using main XTAL as clock source
INIT ERROR:Invalid version info:20231124, 20240422, 5a5aa5a5, 5a5aa5a5
W (663) BLE_INIT: r_ble_controller_init failed 266
I (673) esp_idf_svc::nvs: NvsDefault dropped
Error: ESP_ERR_INVALID_VERSION
I (673) main_task: Returned from app_main()
```
</details>

This issue could have something to do with another error present in `v5.2.3` and `v5.3.2` on an esp32c6, but not in `v5.2.2` and not on an esp32c3: `Image requires efuse blk rev >= v3.21, but chip is v0.3`.
As a workaround I split the compilation and flashing into two steps to avoid flashing the bootloader. Without flashing the bootloader, it works somehow. This error was also reported on [esp-idf-matter](https://github.com/ivmarkov/esp-idf-matter/issues/9).

<details>
<summary>Log</summary>

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0x1e (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x18ec
load:0x4086c410,len:0xe60
load:0x4086e610,len:0x2f2c
entry 0x4086c41a
I (22) boot: ESP-IDF v5.2.3 2nd stage bootloader
I (23) boot: compile time Jan 27 2025 21:04:28
I (23) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.3
I (30) boot.esp32c6: SPI Speed      : 80MHz
I (35) boot.esp32c6: SPI Mode       : DIO
I (39) boot.esp32c6: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00300000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=175538h (1529144) map
E (96) boot_comm: Image requires efuse blk rev >= v3.21, but chip is v0.3
E (103) boot: Factory app partition is not bootable
E (109) boot: No bootable app partitions in the partition table
```
</details>

- Would you like to work on a fix? n

## To Reproduce

1. clone `esp-idf-svc`
2. set `ESP_IDF_VERSION = "v5.2.3"` (or `v5.3.2`) in `.cargo/config.toml`
3. build: `MCU=esp32c6 cargo +nightly build --target riscv32imac-esp-espidf --example bt_gatt_server --features experimental`
4. flash: `espflash flash target/riscv32imac-esp-espidf/debug/examples/bt_gatt_server --monitor`

## Expected behavior

The example should run successfully, as it does with `v5.2.2` or on an esp32c3 with `v5.3.2`

## Environment

- Crate (`esp-idf-svc`) version: master
- ESP-IDF branch or tag: v5.2.3, v5.3.2
- Target device (MCU): esp32c6
- OS: Ubuntu 22.04, MacOS Sequoia 15.2

ivmarkov, 2025-01-28:
@untbu Sorry for the delay. I'll try to address both issues tomorrow. Was a bit too busy in the `rs-matter-embassy` sibling project these days...

ivmarkov, 2025-01-30:
OK so this:

> Calling BtDriver::new(peripherals.modem, Some(nvs.clone()))? on an esp32c6 with ESP-IDF version v5.2.3 or v5.3.2 results in the error ESP_ERR_INVALID_VERSION. On an esp32c3 or with version v5.2.2 it works as expected.

... is now fixed in `esp-idf-svc` `master` and I'll release a patch increment of `esp-idf-svc` these days as well.

This:

> This issue could have something to do with another error present in v5.2.3 and v5.3.2 on an esp32c6, but not in v5.2.2 and not on an esp32c3: Image requires efuse blk rev >= v3.21, but chip is v0.3.
As a workaround I split the compilation and flashing into two steps to avoid flashing the bootloader. Without flashing the bootloader, it works somehow. This error was also reported on https://github.com/ivmarkov/esp-idf-matter/issues/9.

... is a completely separate issue. It seems, for `esp32c6` specifically, the generated `.bin` file (you don't see this file when you pass to `espflash` the ELF one but the `.bin` is still generated internally and it is the `.bin` which is actually flashed)... so the generated `.bin` file is simply wrong (see below the issue that tracks this) in that one specific piece of information is not at the right place.

The reason why you only have problems when using `cargo espflash` is that this missing / reordered piece of information is checked at boot time **only** by the ESP-IDF 5.3.2+ (or 5.2.3+) bootloader. When you call `cargo espflash` you are in fact **also** flashing not just the ELF of your app image, but also a brand new ESP-IDF 5.3.2 bootloader which is generated as part of the build.

However when you call just `espflash flash` you are **only** flashing the ELF app image, and for a bootloader, `espflash` is flashing its own bootloader which is embedded in its contents, and it just so happens that this bootloader is still an older one (from ESP-IDF < 5.3.2) and as such, it does not check the missing /reordered piece of information at all, hence your boot succeeds.

This second problem is now tracked here, where it was originally reported: https://github.com/esp-rs/esp-idf-sys/issues/365

I'm now closing this issue, as the BT problem is fixed, and the second problem is tracked elsewhere (so that we don't end up with a ton of issues for the same problem everywhere) - sorry for making you jump from issue to issue!

ivmarkov, 2025-01-30:
@oddlama @untbu ^^^

closed 2025-01-30 by ivmarkov

oddlama, 2025-01-30:
> I'm now closing this issue, as the BT problem is fixed, and the second problem is tracked elsewhere (so that we don't end up with a ton of issues for the same problem everywhere) - sorry for making you jump from issue to issue!

No worries, thank you very much for the quick investigation and fix!

> Was a bit too busy in the `rs-matter-embassy` sibling project these days...

That sounds awesome, I need to try that!