esp-rs/esp-idf-svc #557: Modifying ethernet netif mtu?

indexds, 2025-01-29:
Hi,

I'm looking to make my ethernet netif's mtu match that of wireguard at 1420 to prevent fragmentation, but the default is 1500 and that doesn't seem changeable at all.
Since the rust wrappers are using the esp-netif api to create interfaces, the raw lwip interface is abstracted and we can't access the mtu. And even if we could (it's most likely possible through shenanigans) I'm pretty sure changing the mtu at runtime is gonna make everything explode.

Soooo... How do we create a netif with a custom mtu? Is it even possible?

From what I could see the mtu is hardcoded here,

```c
/**
 * In this function, the hardware should be initialized.
 * Invoked by ethernetif_init().
 * @param netif lwip network interface which has already been initialized
 */
static void ethernet_low_level_init(struct netif *netif)
{
    netif->mtu = 1500;
    ...
}
```
And this function is called by 
```c
/**
 * @brief   LWIP's network stack init function for Ethernet
 * @param netif LWIP's network interface handle
 * @return ERR_OK on success
 */
err_t ethernetif_init(struct netif *netif);
```
Which *should* be the init function passed when initializing an ethernet netif.

indexds, 2025-02-02:
My current workaround for this is to set the mtu of the ethernet netif on the computer to 1420 with

```fish
netsh interface ipv4 set subinterface "Ethernet" mtu=1420 store=persistent
```

This works, however it is not ideal.

ivmarkov, 2025-02-02:
I think at this point you know about this stuff more than me. If you find a way - let me know. It is also always an option to open a bug in the ESP-IDf repo requesting this functionality.

indexds, 2025-02-02:
Yeah I think that's what I'll do, this isn't really your problem at this point since you just use what's available from the lower layer. Thanks anyway!

indexds, 2025-02-05:
So we might be getting this [soon](https://github.com/espressif/esp-idf/issues/15319). However this would be merged into master for `esp-idf` right? So what's the relationship between the version of `esp-idf-svc` and the version of `esp-idf`?

I'm up for updating the wrappers here for them to match but from what I see we're still in 5.2.2?

Bit confused here.

indexds, 2025-02-05:
Alright, well we shall see how it goes.