esp-rs/esp-idf-svc #564: reading from TCP socket results in `No more processes (os error 11)`

br0kenpixel, 2025-03-08:
## Bug description

I have a weather station setup with my ESP32S3 that sends messages over TCP sockets. At some point, the device fails to read from the socket and gives me a `No more processes (os error 11)` error. The server sends all messages by calling `write_all()` and `flush()` (source [here](https://github.com/PixelWeatherProject/pwmp-server/blob/770e1b1147528c84d830b2675fbf090ef15f6569/src/server/client.rs#L61)). The error message on the server side is `Resource temporarily unavailable (os error 11`.

## Would you like to work on a fix? [y/n]
- Depends on the complexity of the fix.

## To Reproduce

1. Run the firmware.
2. Wait until the device fails to read from the socket.

The specific line at which the error occurs is [here](https://github.com/PixelWeatherProject/pwos/blob/9f61779b453ffd2269663c16d15db30bf39a1cf2/src/firmware.rs#L70).

## Expected behavior

The socket should read the incoming message.

## Environment

- Crate (`esp-idf-svc`) version: `0.51.0`
- ESP-IDF branch or tag: `5.2.2`
- Target device (MCU): `esp32s3`
- OS: Fedora 41

ivmarkov, 2025-03-08:
A failure to read from the socket might be cause by many things and might not necessarily be a bug in the ESP IDF (or `esp-idf-svc`) socket impl.

For example, the Wifi on the device might just be losing access. While I don't see any such logs from the Wifi, it seems you have somehow havily redacted the logger, as I actually don't see _any_ logs from the ESP IDF Wifi Blob.

Can you confirm or deny that the Wifi connection not dropped?

br0kenpixel, 2025-03-13:
I finally managed to fix this.

The issue was caused by the server kicking the client when it was not supposed to. I'm assuming that the ESP32 tried to send data to a broken stream, which was causing this odd error.

Sorry for the noise.

closed 2025-03-13 by br0kenpixel