esp-rs/esp-idf-svc #573: TLS Certificate Parse Error

indexds, 2025-04-05:
Hello, 

This might be a very stupid misconfig on my part but I feel like I've tried everything so here I am.

I am getting this error when trying to establish a tls connection with the following code:

```

```Rust
static CA_CERT: &str = include_str!("../certs/ca.pem");
static CLIENT_CERT: &str = include_str!("../certs/client.crt");
static CLIENT_KEY: &str = include_str!("../certs/client.key");

...

tls.connect(HOSTNAME, 443, &tls::Config {
    common_name: Some(HOSTNAME),
    ca_cert: Some(X509::pem(&CString::new(CA_CERT.as_bytes())?)),
    client_cert: Some(X509::pem(&CString::new(CLIENT_CERT.as_bytes())?)),
    client_key: Some(X509::pem(&CString::new(CLIENT_KEY.as_bytes())?)),
    alpn_protos: Some(&["http/1.1"]),
    non_block: false,
    ..Default::default()
})?;
```

indexds, 2025-04-05:
They are, decoding them on the internet works just fine, decoding them with openssl works just fine, using them to curl the endpoint with mtls works just fine, but mbedtls just won't agree

ivmarkov, 2025-04-05:
These other tools might just recognize automatically dem vs per, that's why I'm asking. As in, if you open them it is all base64 with the === at the beginning?

If yes, no idea.

indexds, 2025-04-05:
Yes it is proper base64 with the ----BEGIN CERTIFICATE---- / ---END CERTIFICATE--- delimiters.

indexds, 2025-04-05:
Do you happen to have (or know of) test certificates and an endpoint I can use to check whether it's me doing something wrong or the certificates i'm using that are problematic?

indexds, 2025-04-05:
I tried with other cert/privkey pairs and the same problem occurs. I'm drawing a blank here

indexds, 2025-04-05:
Okay so increasing log level to verbose gives this:

```E (00:01:18.928) esp-tls-mbedtls: mbedtls_x509_crt_parse of public cert returned -0x262E
I (00:01:18.929) esp-tls-mbedtls: (FFFFD9D2): X509 - Signature algorithm (oid) is unsupported : OID - OID is not found
E (00:01:18.938) esp-tls-mbedtls: Failed to set client pki contextE (00:01:18.944) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
E (00:01:18.957) esp-tls: create_ssl_handle failed
E (00:01:18.962) esp-tls: Failed to open new connection
```

It doesn't seem to be able to find the sign algorithm which does not make sense because it is very much present. Is ED25519 unsupported? I have this enabled already, is that not enough? `CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y`

indexds, 2025-04-05:
So, redid my whole pki with rsa instead of ed25519 which is apparently not supported, and it works. So that's fixed. Apparently support for ed25519 is not there yet.

https://github.com/Mbed-TLS/mbedtls/issues/2452

closed 2025-04-05 by indexds