esp-rs/esp-idf-svc #575: could not find `bt` in `esp_idf_svc`

cyygs, 2025-04-09:
## Bug description

- Would you like to work on a fix? [y/n]

## To Reproduce

1. ...
2. ...

## Expected behavior

## Environment

- Crate (`esp-idf-svc`) version: [e.g. 0.49.1; type master if you use the crate from the GIT master branch]
- ESP-IDF branch or tag: [e.g. release/v5.2, v5.2.1, etc.] 
- Target device (MCU): [e.g. esp32s3] 
- OS: [e.g. Ubuntu 20.04]

Vollbrecht, 2025-04-09:
The current esp-idf-svc version is 0.51. With that said in 99% its not an issue when you compile your crate but just rust-analyzer for example beeing confused. Are you able to compile your program or is it just rust-analyzer not understanding what you are doing?

For bt to be available you need to enable Bluetooth via the sdkconfigs described in the example and have the "experimental" cargo feature enabled. You can see that directly[ here at the code point](https://github.com/esp-rs/esp-idf-svc/blob/master/src/lib.rs#L36)

ivmarkov, 2025-04-09:
@Vollbrecht For the next release, I think it is safe to promote everything from "experimental" to "regular" except openthread, which might have a bit more development.

Vollbrecht, 2025-04-09:
Yeah, though i think the bigger culprit that many people are hitting, is actually not the "experimental" missing, but a problem in rust-analyzer that i cannot pin down exactly. The sdkconfig created cfg flag "esp_idf_bt_bluedroid_enabled" is always there when i check it through rustc, but rust-analyzer specifically has problems with it, despite not stumbling over similar from us generated flags via embuild. 

So they are in the state where they can compile it always but can never make it work in RA. Down the line if i get the sdkconfig parsing done one day we may upfront can generate all cfg's for all combinations possible upfront, so we don't need the ignore unknown cfg's thing. That might fix that specific issue. 

I am ok with removing the experimental on bt, though i just have a PR in the pipe, let me send it later.

cyygs, 2025-04-12:
Enabling VNet BT-BLUEDROID_ENABLED=y can solve the problem of finding bt, but if you want to use ESP32 nimble, there will be a problem. There will be conflicts when using both nimble and bluesoid for ESP32. How to solve the coordination problem between the ESP32 nimble library and the ESP idf svc library?

cyygs, 2025-04-12:
.cargo/registry/src/rsproxy.cn-e3de039b2554c837/esp-idf-svc-0.51.0/src/lib.rs:30:1
 #[cfg(all(
   esp_idf_bt_enabled,
    esp_idf_bt_bluedroid_enabled,
    feature = "alloc",
    feature = "experimental"
 ))]

cyygs, 2025-04-12:
It is mandatory here that esp_idf_bt_bluedrid_enabled must be y

Vollbrecht, 2025-04-12:
The wrapper in the bt module are for using the bluedroid stack, if you wanting to use nimble, like with the rust esp32-nimble crate, than you can ignore this module. It does not make sense to run the nimble controller and the bluedroid controller at the same time. 

If you have general question feel free to ask in matrix chat linked in the README if you have problems using stuff. 

Closing here as there is no issue itself with importing the bt module.

closed 2025-04-12 by Vollbrecht