esp-rs/esp-idf-svc #603: question about MountedFatfs

joyhope, 2025-08-21:
In the sdmmc demo code:

The mounted_fatfs should be kept. 
```
    // Keep it around or else it will be dropped and unmounted
    let _mounted_fatfs = MountedFatfs::mount(Fatfs::new_sdcard(0, sd_card_driver)?, "/sdcard", 4)?;
```

I am using Rust std environment. My plan is to make the mounted_fatfs object global, so any other multi-thread could use the mounted fs. 

static MOUNTED_FATFS: OnceCell<MountedFatfs<SdCardDriver<SdMmcHostDriver>>> = OnceCell::new();

i tried to use this oncecell, but the compiler said 

```
 static MOUNTED_SD_CARD: OnceCell<MountedFatfs<SdCardDriver<SdMmcHostDriver>>> = OnceCell::new();
      |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `*mut sd_pwr_ctrl_drv_t` cannot be sent between threads safely
```

How could I keep the Mounted Fatfs in my whole application life time?

joyhope, 2025-08-21:
The function returned mounted_fs

```
pub fn sdmmc_init() -> anyhow::Result<MountedFatfs<Fatfs<SdCardDriver<SdMmcHostDriver<'static>>>>> {

    let peripherals = Peripherals::take()?;
    let pins = peripherals.pins;

    let sd_card_driver = SdCardDriver::new_mmc(
        // => Data width = 4 bits
        SdMmcHostDriver::new_4bits(
            peripherals.sdmmc1,
            pins.gpio42,
            pins.gpio41,
            pins.gpio40,
            pins.gpio39,
            pins.gpio1,
            pins.gpio2,
            None::<gpio::AnyIOPin>,
            None::<gpio::AnyIOPin>,
            &SdMmcHostConfiguration::new(),
        )?,
        // => Data width = 1 bit
        // Comment out the above configuration and uncomment this block
        // if you have connected only the d0 pin
        // SdMmcHostDriver::new_slot1_1bit(
        //     peripherals.sdmmc1,
        //     pins.gpio15,
        //     pins.gpio14,
        //     pins.gpio2,
        //     None::<gpio::AnyIOPin>,
        //     None::<gpio::AnyIOPin>,
        //     &SdMmcHostConfiguration::new(),
        // )?,
        &SdCardConfiguration::new(),
    )?;

    // Keep it around or else it will be dropped and unmounted
    let mounted_fatfs = MountedFatfs::mount(Fatfs::new_sdcard(0, sd_card_driver)?, "/sdcard", 4)?;

    Ok(mounted_fatfs)
}

```

In the main.rs
```
fn main() -> anyhow::Result<()> {
    esp_idf_svc::sys::link_patches();
    EspLogger::initialize_default();

    let mounted_fatfs=sdmmc_init()?;

    loop {
        // your app logic
        std::thread::sleep(std::time::Duration::from_secs(1));
    }
    Ok(())
}
...

But in other thread,  I still unable to use standard file i/o.  Is there anything I made wrong?

joyhope, 2025-08-21:
Sorry,  my code problem.  It is returned in the main, so the fatfs is still dropped.  

It is clear that if mounted_fatfs is kept,  the fs is available. 

I hope there is some global fatfs solution.

ivmarkov, 2025-08-21:
> Sorry, my code problem. It is returned in the main, so the fatfs is still dropped.
> 
> It is clear that if mounted_fatfs is kept, the fs is available.
> 
> I hope there is some global fatfs solution.

You are opening bugs for what is actually _general Rust knowledge_. Please join [The Matrix room](https://matrix.to/#/#esp-rs:matrix.org) and ask questions there, and also google a bit for Rust training / topics.

There are many way to solve your problem. In particular:
- `core::mem::forget()` `mounted_fatfs`
- Put `mounted_fatfs` in a static context using the `lazy_static!` macro of Rust plus `std::sync::Mutex<Option<MountedFatFs<'static>>>`.

closed 2025-08-21 by ivmarkov

joyhope, 2025-08-21:
In fact, I tried many ways, butwas  unable to solve. 

```
std::sync::Mutex<Option<MountedFatFs<'static>>>
```
Not work, because MountedFatFs needs T.