esp-rs/esp-idf-svc #607: Panic when using Thread

l0nax, 2025-09-03:
## Bug description

When using the experimental Thread module, the ESP32-C6 sometimes crashes.
Previously I wasn't able to connect to the WiFi, but after using ESP-IDF v5.2.x and creating an "IoT" network with PMF disabled the ESP now connects successfully.

Sometimes (by resetting the MCU over and over again) the OpenThread integration works as expected. But I only managed to get it into this state once.

- Would you like to work on a fix? -> yes

## To Reproduce

The related code & config can be found here: https://gitlab.com/l0nax/esp32c6-openthread-panic-bug-repro#

Please note that when using the `master` branch of this repository I receive a bunch of compilation errors which I did not manage to resolve:
```
error[E0432]: unresolved imports `crate::sys::otDnsTxtEntry`, `crate::sys::otSrpClientAddService`, `crate::sys::otSrpClientClearHostAndServices`, `crate::sys::otSrpClientClearService`, `crate::sys::otSrpClientEnableAutoStartMode`, `crate::sys::otSrpClientGetHostInfo`, `crate::sys::otSrpClientGetServerAddress`, `crate::sys::otSrpClientGetServices`, `crate::sys::otSrpClientHostInfo`, `crate::sys::otSrpClientIsAutoStartModeEnabled`, `crate::sys::otSrpClientIsRunning`, `crate::sys::otSrpClientItemState`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_ADDING`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REFRESHING`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REGISTERED`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REMOVED`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REMOVING`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_ADD`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_REFRESH`, `crate::sys::otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_REMOVE`, `crate::sys::otSrpClientRemoveHostAndServices`, `crate::sys::otSrpClientRemoveService`, `crate::sys::otSrpClientService`, `crate::sys::otSrpClientSetHostAddresses`, `crate::sys::otSrpClientSetHostName`, `crate::sys::otSrpClientStart`, `crate::sys::otSrpClientStop`
  --> /home/eb/.cargo/git/checkouts/esp-idf-svc-3846902cb7f9c731/3c7ce53/src/thread/srp.rs:10:39
   |
10 |     esp, esp_openthread_get_instance, otDnsTxtEntry, otError, otError_OT_ERROR_INVALID_ARGS,
   |                                       ^^^^^^^^^^^^^ no `otDnsTxtEntry` in `sys`
11 |     otError_OT_ERROR_NO_BUFS, otIp6Address, otIp6Address__bindgen_ty_1, otSrpClientAddService,
   |                                                                         ^^^^^^^^^^^^^^^^^^^^^ no `otSrpClientAddService` in `sys`
12 |     otSrpClientClearHostAndServices, otSrpClientClearService, otSrpClientEnableAutoStartMode,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `otSrpClientEnableAutoStartMode` in `sys`
   |     |                                |
   |     |                                no `otSrpClientClearService` in `sys`
   |     no `otSrpClientClearHostAndServices` in `sys`
13 |     otSrpClientGetHostInfo, otSrpClientGetServerAddress, otSrpClientGetServices,
   |     ^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^ no `otSrpClientGetServices` in `sys`
   |     |                       |
   |     |                       no `otSrpClientGetServerAddress` in `sys`
   |     no `otSrpClientGetHostInfo` in `sys`
14 |     otSrpClientHostInfo, otSrpClientIsAutoStartModeEnabled, otSrpClientIsRunning,
   |     ^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^
   |     |                    |
   |     |                    no `otSrpClientIsAutoStartModeEnabled` in `sys`
   |     no `otSrpClientHostInfo` in `sys`
15 |     otSrpClientItemState, otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_ADDING,
   |     ^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
16 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REFRESHING,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REGISTERED,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REMOVED,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_REMOVING,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
20 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_ADD,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
21 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_REFRESH,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
22 |     otSrpClientItemState_OT_SRP_CLIENT_ITEM_STATE_TO_REMOVE, otSrpClientRemoveHostAndServices,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23 |     otSrpClientRemoveService, otSrpClientService, otSrpClientSetHostAddresses,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
24 |     otSrpClientSetHostName, otSrpClientStart, otSrpClientStop, EspError, ESP_ERR_INVALID_STATE,
   |     ^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^

error[E0432]: unresolved imports `crate::sys::otSrpClientEnableAutoHostAddress`, `crate::sys::otSrpClientGetKeyLeaseInterval`, `crate::sys::otSrpClientGetLeaseInterval`, `crate::sys::otSrpClientGetTtl`, `crate::sys::otSrpClientSetKeyLeaseInterval`, `crate::sys::otSrpClientSetLeaseInterval`, `crate::sys::otSrpClientSetTtl`
  --> /home/eb/.cargo/git/checkouts/esp-idf-svc-3846902cb7f9c731/3c7ce53/src/thread/srp.rs:29:5
   |
29 |     otSrpClientEnableAutoHostAddress, otSrpClientGetKeyLeaseInterval, otSrpClientGetLeaseInterval,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `otSrpClientGetLeaseInterval` in `sys`
   |     |                                 |
   |     |                                 no `otSrpClientGetKeyLeaseInterval` in `sys`
   |     no `otSrpClientEnableAutoHostAddress` in `sys`
30 |     otSrpClientGetTtl, otSrpClientSetKeyLeaseInterval, otSrpClientSetLeaseInterval,
   |     ^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `otSrpClientSetLeaseInterval` in `sys`
   |     |                  |
   |     |                  no `otSrpClientSetKeyLeaseInterval` in `sys`
[... 95 lines cut ...]
1879 |         netif: EspNetif,
     |         ---------------
1880 |         mut mode: T,
     |         -----------
help: provide the arguments
     |
1726 -         Self::internal_init(Self {
1727 -             driver,
1728 -             inner: UnsafeCell::new(EspThreadInner::new(netif, BorderRouter(backbone_netif))),
1729 -         })
1726 +         Self::internal_init(/* ThreadDriver<'_, Host> */, /* netif::EspNetif */, /* BorderRouter<N> */)
     |
```

## Expected behavior

* That the WiFi connects without running into the `ESP_ERR_TIMEOUT` problem over and over again, i.e. it can be used with a "normal" WiFi network.
* The OpenThread integration can be used "stable", i.e., not crashing almost every time.

## Environment

- Crate (`esp-idf-svc`) version: v0.51.0
- ESP-IDF branch or tag: v5.2.2, v5.2.5
- Target device (MCU): esp32c6 (on the  'esp32-c6-devkitc-1' board)
- OS: Arch Linux

l0nax, 2025-09-05:
Thanks for the fast help :)

Here is one of the crashes with the obscure panic:
```
[...]
I (5469) esp32c6_ble_wifi_credentials: Wi-Fi connected! IP: 192.168.1.24
I (5469) esp32c6_ble_wifi_credentials: Initializing OpenThread stack...
I (5479) OPENTHREAD: Host connection mode none
I (5479) phy: libbtbb version: f97b181, May  8 2024, 10:42:29
I(5489) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I (5499) OPENTHREAD: OpenThread attached to netif
I (5499) esp32c6_ble_wifi_credentials: [OT-Sub] Got: Started
I(5509) OPENTHREAD:[N] RoutingManager: BR ULA prefix: fd5e:49c4:61b::/48 (loaded)
I(5509) OPENTHREAD:[N] RoutingManager: Local on-link prefix: fdde:ad00:beef:cafe::/64
I (5519) esp32c6_ble_wifi_credentials: [OT-Sub] Got: MulticastJoined
I (5519) esp32c6_ble_wifi_credentials: [OT-Sub] Got: MulticastJoined
I (5529) esp32c6_ble_wifi_credentials: [OT-Sub] Got: MulticastJoined
I (5529) esp32c6_ble_wifi_credentials: [OT-Sub] Got: GotIpv6
I (5539) OPENTHREAD: Platform UDP bound to port 49153
I(5539) OPENTHREAD:[N] Mle-----------: Role disabled -> detached
I (5549) esp32c6_ble_wifi_credentials: [OT-Sub] Got: GotIpv6
I (5549) esp32c6_ble_wifi_credentials: [OT-Sub] Got: MulticastJoined
I (5559) esp32c6_ble_wifi_credentials: [OT-Sub] Got: MulticastJoined
I (5569) wifi:Set ps type: 2, coexist: 0

I (5579) esp32c6_ble_wifi_credentials: Thread Border Router initialized, now running...
I (5589) esp32c6_ble_wifi_credentials: [OT-Sub] Got: RoleChanged { current_role: Detached, previous_role: Disabled }
I (5589) OT_STATE: netif up
I (5589) esp32c6_ble_wifi_credentials: [OT-Sub] Got: IfUp
I (5599) OPENTHREAD: NAT64 ready
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40810e56  RA      : 0x40811410  SP      : 0x40843aa0  GP      : 0x40822fd4
TP      : 0x407ec4dc  T0      : 0x00000000  T1      : 0x40803652  T2      : 0x000002ff
S0/FP   : 0x4084e46e  S1      : 0x4084e368  A0      : 0x4084c210  A1      : 0x40833338
A2      : 0x4084e368  A3      : 0x00000018  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x4084e46e  A7      : 0x0062b82b  S2      : 0x00000000  S3      : 0x40833000
S4      : 0x0000007c  S5      : 0x4084c210  S6      : 0x4084d28c  S7      : 0x00000018
S8      : 0x00000001  S9      : 0x4083250c  S10     : 0x4084e454  S11     : 0x4084e368
T3      : 0x80000000  T4      : 0x0205003a  T5      : 0x02000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000004
MHARTID : 0x00000000

Stack memory:
40843aa0: 0x36818c58 0xcd5018dd 0x4084d28c 0x40833000 0x00000000 0x40833338 0x00000000 0x40811410
40843ac0: 0x40848090 0x40824274 0x408349b0 0x000000f5 0x00000000 0xffffffc4 0x000000a0 0x4084e45e
40843ae0: 0x4084e368 0x421b4000 0x4084e3f8 0x00000000 0x00000000 0x00000000 0x40880000 0x00002000
40843b00: 0x000000c6 0x421b4000 0x40880000 0x000000f5 0x40880000 0x4084e3f8 0x4084e368 0x40820b12
40843b20: 0x00000000 0x00000000 0x40880000 0x40880000 0x40880000 0x40880000 0x00000000 0x40880000
40843b40: 0x408257c0 0x421b23ac 0x40880000 0x4081b506 0x00000000 0x00000000 0x00000011 0x00000000
40843b60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843b80: 0x00000000 0x00000000 0x00000000 0x4080a898 0x00000000 0x00000000 0x00000000 0x00000000
40843ba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40843bc0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x40843a70 0x00000157 0x40826d1c
40843be0: 0x40826d1c 0x40843bd4 0x40826d14 0x00000002 0x40841b0c 0x40841b0c 0x40843bd4 0x00000000
40843c00: 0x00000017 0x408421d0 0x69666977 0x00000000 0x00000000 0x00000000 0x40843bc0 0x00000017
40843c20: 0x00000000 0x00000000 0x00000000 0x00000000 0x40836bd0 0x40836c38 0x40836ca0 0x00000000
40843c40: 0x00000000 0x00000001 0x00000000 0x00000000 0x4084c138 0x42061dd2 0x00000000 0x00000000
40843c60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843c80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843ca0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843cc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843ce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843d00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40843d20: 0x00000000 0x00000054 0x40843d28 0x40843d28 0x40843d28 0x40843d28 0x00000000 0x40843d40
40843d40: 0xffffffff 0x40843d40 0x40843d40 0x00000000 0x40843d54 0xffffffff 0x40843d54 0x40843d54
40843d60: 0x00000000 0x00000001 0x00000000 0x4f00ffff 0x00000000 0x005f9cbb 0x40843d20 0x00000bfc
40843d80: 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d
40843da0: 0x00010000 0x00000000 0x00000004 0x00000001 0x40832a14 0x2e617473 0x64697373 0x00000000
[... 86 lines cut ...]
                found struct `EspThread<'d, BorderRouter<N>>`
note: associated function defined here
    --> /home/eb/.cargo/git/checkouts/esp-idf-svc-3846902cb7f9c731/a37ce9d/src/thread.rs:1877:8
     |
1877 |     fn internal_init(
     |        ^^^^^^^^^^^^^
1878 |         driver: ThreadDriver<'d, Host>,
     |         ------------------------------
1879 |         netif: EspNetif,
     |         ---------------
1880 |         mut mode: T,
[... cut ...]
1726 -         Self::internal_init(Self {
1727 -             driver,
1728 -             inner: UnsafeCell::new(EspThreadInner::new(netif, BorderRouter(backbone_netif))),
1729 -         })
1726 +         Self::internal_init(/* ThreadDriver<'_, Host> */, /* netif::EspNetif */, /* BorderRouter<N> */)
     |

Some errors have detailed explanations: E0061, E0433, E0560.
For more information about an error, try `rustc --explain E0061`.
error: could not compile `esp-idf-svc` (lib) due to 5 previous errors
```

l0nax, 2025-09-05:
I've already set `CONFIG_ESP_MAIN_TASK_STACK_SIZE=20000`.

When compiling now, most of the errors are resolved. But now I cannot call `init_coex`:
```
error[E0599]: no method named `init_coex` found for struct `EspThread<'d, T>` in the current scope
  --> src/main.rs:92:12
   |
92 |     thread.init_coex();
   |            ^^^^^^^^^ method not found in `EspThread<'_, BorderRouter<&EspNetif>>`
```

When compiling ESP-IDF v5.4.2 (and locally applying the patch from #608) I was unable to produce a crash, but the WiFi connection seemed to cut of:
<details>

<summary>Logs</summary>

```
I (638) esp32c6_ble_wifi_credentials: Initializing WiFi...
I (638) pp: pp rom version: 5b8dcfa
I (638) net80211: net80211 rom version: 5b8dcfa
I (648) wifi:wifi driver task: 4084d5d4, prio:23, stack:6656, core=0
I (658) wifi:wifi firmware version: bea31f3
I (658) wifi:wifi certification version: v7.0
I (658) wifi:config NVS flash: enabled
I (658) wifi:config nano formatting: disabled
I (668) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N, band mode:0x1
I (668) wifi:Init data frame dynamic rx buffer num: 32
I (678) wifi:Init static rx mgmt buffer num: 5
I (678) wifi:Init management short buffer num: 32
I (688) wifi:Init dynamic tx buffer num: 32
I (688) wifi:Init static tx FG buffer num: 2
I (688) wifi:Init static rx buffer size: 1700 (rxctrl:92, csi:512)
I (698) wifi:Init static rx buffer num: 10
I (698) wifi:Init dynamic rx buffer num: 32
I (708) wifi_init: rx ba win: 6
I (708) wifi_init: accept mbox: 6
I (708) wifi_init: tcpip mbox: 32
I (718) wifi_init: udp mbox: 6
I (718) wifi_init: tcp mbox: 6
I (718) wifi_init: tcp tx win: 5760
I (728) wifi_init: tcp rx win: 5760
I (728) wifi_init: tcp mss: 1440
I (728) wifi_init: WiFi IRAM OP enabled
I (728) wifi_init: WiFi RX IRAM OP enabled
I (738) wifi_init: WiFi SLP IRAM OP enabled
I (748) esp32c6_ble_wifi_credentials: Allowing to stabilize...
I (858) esp32c6_ble_wifi_credentials: Connecting to WiFi
I (858) esp32c6_ble_wifi_credentials: Starting Wi-Fi...
I (858) wifi:Set ps type: 0, coexist: 0

I (858) phy_init: phy_version 331,5b89037,Mar  3 2025,16:01:12
W (948) wifi:ACK_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (948) wifi:CTS_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (958) wifi:(agc)0x600a7128:0xd2008a06, min.avgNF:0xce->0xd2(dB), RCalCount:0x8, min.RRssi:0xa06(-95.62)
I (968) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

I (978) wifi:mode : sta (58:8c:81:36:dd:18)
I (978) wifi:enable tsf
I (978) esp32c6_ble_wifi_credentials: Connecting to Wi-Fi...
I (988) wifi:(he)max.RxAMPDULenExponentExtension:3, max.RxAMPDUpre-EOFpaddingBytes:8388607(0x7fffff)
I (998) wifi:(mac)omc_ul_mu_data_disable_rx:0
I (998) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:0
I (1008) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:1, nss:0)
I (1008) wifi:(phy)rx_mcs_map:0xfffa, tx_mcs_map:0xfffa, stbc_tx:0, bfmer(su:1, mu:0), ldpc:1, max_rate:172
I (1018) wifi:(phy)nsts:2, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (1028) wifi:(phy)threshold_bits:24, nsts_num:2, ru_num:2, he_cap->len:26, ppe_threshold_len:4(6,11,4)
I (1038) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
I (1038) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (1058) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:0
I (1068) wifi:(opr)len:7, BSS Color:23, disabled:0, Partial BSS Color:0
[... 77 lines cut ...]
I (4988) OPENTHREAD: NAT64 ready
I (4998) esp_idf_svc::http::server: Started Httpd server with config Configuration { http_port: 80, ctrl_port: 32768, https_port: 443, max_sessions: 16, session_timeout: 1200s, stack_size: 10240, max_open_sockets: 4, max_uri_handlers: 32, max_resp_headers: 8, lru_purge_enable: true, uri_match_wildcard: false }
I (5018) esp_idf_svc::http::server: Registered Httpd server handler Get for URI "/"
I (5028) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=true is_up=true
I (7048) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=true is_up=true
I (9058) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=true is_up=true
I (11068) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=true is_up=true
I (11338) wifi:bcn_timeout,ap_probe_send_start
I (13078) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=true is_up=true
I (13838) wifi:ap_probe_send over, reset wifi status to disassoc
I (13838) wifi:state: run -> init (0xc800)
I (13848) wifi:pm stop, total sleep time: 3753276 us / 10198090 us

I (13848) wifi:<ba-del>idx:0, tid:0
I (13848) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (15088) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (17098) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (19108) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (21118) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (23128) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (25138) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (27148) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
```

</details>

l0nax, 2025-09-05:
I tried to access it, but `EspThread::driver_mut` is private. 

I manually applied – temporarily – it because of the build errors on `master` to test whether it would work with newer ESP-IDF versions.

With the latest `master` version and ESP-IDF v5.5.1 it does not crash anymore but the WiFi coex mode does not seem to work.

<details>

<summary>Logs</summary>

```
I (859) wifi:Set ps type: 0, coexist: 0

I (859) phy_init: phy_version 341,a6c8fe6,Jul 30 2025,18:34:11
W (949) wifi:ACK_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (949) wifi:CTS_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (959) wifi:(agc)0x600a7128:0xd2061a0a, min.avgNF:0xce->0xd2(dB), RCalCount:0x61, min.RRssi:0xa0a(-95.38)
I (969) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

I (979) wifi:mode : sta (58:8c:81:36:dd:18)
I (979) wifi:enable tsf
I (979) esp32c6_ble_wifi_credentials: Connecting to Wi-Fi...
I (989) wifi:(he)max.RxAMPDULenExponentExtension:3, max.RxAMPDUpre-EOFpaddingBytes:8388607(0x7fffff)
I (999) wifi:(mac)omc_ul_mu_data_disable_rx:0
I (999) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:0
I (1009) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:1, nss:0)
I (1009) wifi:(phy)rx_mcs_map:0xfffa, tx_mcs_map:0xfffa, stbc_tx:0, bfmer(su:1, mu:0), ldpc:1, max_rate:172
I (1019) wifi:(phy)nsts:2, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (1029) wifi:(phy)threshold_bits:24, nsts_num:2, ru_num:2, he_cap->len:26, ppe_threshold_len:4(6,11,4)
I (1039) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
I (1049) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (1059) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:0
I (1069) wifi:(opr)len:7, BSS Color:23, disabled:0, Partial BSS Color:0
I (1069) wifi:(spr)len:2, ctrl:0x3, PSR Disallowed:1, Non-SRG OBSS PD SR Disallowed:1
I (1079) wifi:(spr)len:2, ctrl:0x3, Non-SRG Offset Present:0, SRG Info Present:0
I (3389) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (3389) wifi:(connect)dot11_authmode:0x3, pairwise_cipher:0x3, group_cipher:0x3
I (3399) wifi:state: init -> auth (0xb0)
I (3429) wifi:state: auth -> assoc (0x0)
I (3449) wifi:(assoc)RESP, Extended Capabilities length:10, operating_mode_notification:0
I (3449) wifi:(assoc)RESP, Extended Capabilities, MBSSID:0, TWT Responder:0, OBSS Narrow Bandwidth RU In OFDMA Tolerance:0
I (3459) wifi:Extended Capabilities length:10, operating_mode_notification:1
I (3469) wifi:(he)max.RxAMPDULenExponentExtension:3, max.RxAMPDUpre-EOFpaddingBytes:8388607(0x7fffff)
I (3479) wifi:(mac)omc_ul_mu_data_disable_rx:0
I (3479) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:0
I (3489) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:1, nss:0)
I (3499) wifi:(phy)rx_mcs_map:0xfffa, tx_mcs_map:0xfffa, stbc_tx:0, bfmer(su:1, mu:0), ldpc:1, max_rate:172
I (3499) wifi:(phy)nsts:2, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (3509) wifi:(phy)threshold_bits:24, nsts_num:2, ru_num:2, he_cap->len:26, ppe_threshold_len:4(6,11,4)
I (3519) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
I (3529) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (3539) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:0
I (3549) wifi:(opr)len:7, BSS Color:23, disabled:0, Partial BSS Color:0
I (3559) wifi:state: assoc -> run (0x10)
I (3559) wifi:(he)ppe_thresholds_present:1, nominal_packet_padding(rx:0, cfg:2)
I (3569) wifi:(trc)phytype:CBW20-SGI, snr:43, maxRate:172, highestRateIdx:0
W (3569) wifi:(trc)band:2G, phymode:3, highestRateIdx:0, lowestRateIdx:11, dataSchedTableSize:14
I (3579) wifi:(trc)band:2G, rate(S-MCS7, rateIdx:0), ampdu(rate:S-MCS7, schedIdx(0, stop:8)), snr:43, ampduState:wait operational
I (3589) wifi:ifidx:0, rssi:-53, nf:-96, phytype(0x3, CBW20-SGI), phymode(0x7, 11ax), max_rate:172, he:1, vht:0, ht:1
I (3599) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:0(no restriction)
I (3639) wifi:(extcap)mbssid:0, enhanced_mbssid_advertise:0, complete_nontxbssid_profiles:0, twt_responder: 1
I (3639) wifi:connected with Elysnet IoT, aid = 1, channel 1, BW20, bssid = ee:38:83:43:f7:46
I (3649) wifi:security: WPA2-PSK, phy:ax, rssi:-53, cipher(pairwise:0x3, group:0x3), pmf:0,
I (3659) wifi:pm start, type: 0, twt_start:0

I (3659) wifi:pm start, type:0, aid:0x1, trans-BSSID:ee:38:83:43:f7:46, BSSID[5]:0x46, mbssid(max-indicator:0, index:0), he:1
I (3669) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3679) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (3689) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (3699) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x30)
[... 74 lines cut ...]
I (35209) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (37219) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (39229) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
[... cut ...]
I (43249) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (45259) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (47269) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (49279) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
I (51289) esp32c6_ble_wifi_credentials: Awaiking from the main thread... is_connected=false is_up=false
```

</details>

ivmarkov, 2025-09-05:
> I tried to access it, but `EspThread::driver_mut` is private.

Ok, I'll fix that too. There is no longer a reason why it should be private.

> With the latest `master` version and ESP-IDF v5.5.1 it does not crash anymore but the WiFi coex mode does not seem to work.

I wonder why you think coex does not work though? Wifi gets a DHCP address and then the BR elects itself as a Leader?

ivmarkov, 2025-09-07:
Not sure if that matters, but if you have coex problems, maybe you could try to call [init_coex](https://gitlab.com/l0nax/esp32c6-openthread-panic-bug-repro/-/blob/main/src/main.rs?ref_type=heads#L88) _before_ the Wifi is connected?

ivmarkov, 2025-09-09:
> I already call `init_coex` – but this does not change the outcome.

Are you calling it _before_ the Wifi is connected?

> Once the OpenThread stack is fully initialized, I can no longer communicate with the ESP32, and `ping 'ing also results in no communication. Additionally, `is_connected`and`is_up`for WiFi return`false` after the OpenThread initialization.

Anyway, I'm back. If you publish somewhere a version of your code which is against latest `master` `esp-idf-svc`, I can try running it on my end.

l0nax, 2025-09-09:
> Are you calling it before the Wifi is connected?

No, I'm calling it in a loop every two seconds.

> Anyway, I'm back. If you publish somewhere a version of your code which is against latest master esp-idf-svc, I can try running it on my end.

Sure, I've uploaded the latest changes to the PoC repo (https://gitlab.com/l0nax/esp32c6-openthread-panic-bug-repro) on `master`

ivmarkov, 2025-09-09:
> > Are you calling it before the Wifi is connected?
> 
> No, I'm calling it in a loop every two seconds.

You don't understand. Lines 83 to 92 in your most recent code need to go **above** line 76. That's what I'm suggesting at least.

> > Anyway, I'm back. If you publish somewhere a version of your code which is against latest master esp-idf-svc, I can try running it on my end.
> 
> Sure, I've uploaded the latest changes to the PoC repo (https://gitlab.com/l0nax/esp32c6-openthread-panic-bug-repro) on `master`

Thanks. give me a day or so to get to it.

ivmarkov, 2025-09-09:
> You don't understand. Lines 83 to 92 in your most recent code need to go **above** line 76. That's what I'm suggesting at least.

Argh, that might not work because of reasons. You could also try commenting out the call to `init_coex` and place this raw call above line 76:

```rust
esp!(unsafe { esp_idf_svc::sys::esp_coex_wifi_i154_enable() })?;
```