esp-rs/esp-idf-svc #609: Perhaps WifiDriver::get_ap_info() shouldn't require a mut reference?

darconeous, 2025-09-04:
I noticed that `WifiDriver::get_ap_info()` requires a `mut` reference:

```rust
pub fn get_ap_info(&mut self) -> Result<AccessPointInfo, EspError> {
    let mut ap_info_raw: wifi_ap_record_t = wifi_ap_record_t::default();
    // If Sta not connected throws EspError(12303)
    esp!(unsafe { esp_wifi_sta_get_ap_info(&mut ap_info_raw) })?;
    let ap_info: AccessPointInfo = Newtype(&ap_info_raw).try_into().unwrap();

    ::log::debug!("AP Info: {:?}", ap_info);
    Ok(ap_info)
}
```

This seems to be unnecessary because `esp_wifi_sta_get_ap_info` is thread safe and it just writes into `ap_info_raw` which we have on the stack.

Could this method be changed to not require a `mut` reference? This would greatly help improve the ergonomics of the method.

ivmarkov, 2025-09-05:
Thanks, addressed in https://github.com/esp-rs/esp-idf-svc/commit/2b74a683cfcf1b72c26d7cedbaf488d3cc00f829

closed 2025-09-05 by ivmarkov

darconeous, 2026-02-02:
Any idea when this might make it into a release?